lvgl/lvgl #3787: roller with more than 60 options will be messed up

cuiqingwei, 2022-11-08:
This is the result of the simulation on pc
[image]

The same problem is created by code to run directly on the target board
`lv_obj_t *roller = lv_roller_create(obj);`

cuiqingwei, 2022-11-08:
LVGL version is ： 8.2.0
Supplement: 39 are normal, 40 or more are problematic

kisvegabor, 2022-11-09:
It's already fix in v8.3.3 but as a workaround you can enable `LV_USE_LARGE_COORD 1` in `lv_conf.h`.

JeremiahGillis, 2022-11-18:
I still have this issue on c4c400716e80a279e7b3d43b8992915fe94441eb when `LV_USE_LARGE_COORD 0`. However, I do have five Rollers on the screen, with the right two set to `LV_ROLLER_MODE_INFINITE`. I did try just the right Roller set to infinite and I still saw the same issue. My `LV_ROLLER_INF_PAGES` is 7.

I may have too many rollers for your initial optimization. The number of elements in each is shown in red.

[image]

All of my rollers will work in infinite mode when `LV_USE_LARGE_COORD 1`. It does require `LV_MEM_SIZE` set to at least 64KB. The downside for me is that the screen has some draw hesitation when first switching to it vs `LV_ROLLER_MODE_NORMAL`.

kisvegabor, 2022-11-19:
What is the result for this?
```c
    lv_obj_update_layout(roller1);
    lv_obj_t * label = lv_obj_get_child(roller1, 0);
    printf("%d, %d, %d\n", ((lv_roller_t *)roller1)->inf_page_cnt, label->coords.y1, label->coords.y2);
```

JeremiahGillis, 2022-11-21:
> ```c
> inf_page_cnt
> ```

This doesn't exist in release/v8.3. I do see it in the master branch, but I don't have that currently checked out.

kisvegabor, 2022-11-21:
Oh, sorry, I remember incorrectly. I thought it wad added to v8.3 too.

Setting `LV_ROLLER_INF_PAGES` in `lv_conf.h` to `3` or `5` probably will solve the issue.

Anyway if you are interested in adding the changes manually take a look at [this commit](https://github.com/lvgl/lvgl/commit/878e3d333#diff-19bd6e734029ac822446b417025df0803c97fd2a89af39b90f2104c7c3c35494).

JeremiahGillis, 2022-11-21:
No problem. Changing `LV_ROLLER_INF_PAGES` to `3` solved it. I only tried larger values before. I didn't expect less to help.

kisvegabor, 2022-11-21:
Great!

It sets how many times LVGL adds the options to have space for infinite rolling. With >10 options usually `LV_ROLLER_INF_PAGES = 3` is fine (1 page above, 1 the actual page, 1 page below) as probably you won't scroll more than 10 options with one gesture.

closed 2022-11-21 by kisvegabor