lvgl/lvgl #3788: Russian can not be shown in V8.2

yin2021, 2022-11-08:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Russian can be shown in V7.  However, it can not be shown in V8.2.

### To Reproduce

void lv_example_label_1(lv_obj_t *par)
{
    lv_obj_t * bg = lv_obj_create(lv_scr_act());
    lv_obj_set_size(bg, 300, 300);
    lv_obj_align(bg, LV_ALIGN_TOP_LEFT, 0, 0);

    lv_obj_t * label1 = lv_label_create(bg);
    lv_label_set_long_mode(label1, LV_LABEL_LONG_CLIP);
    lv_label_set_text(label1, "first line");
    lv_obj_set_width(label1, 88);
    lv_obj_align(label1, LV_ALIGN_TOP_LEFT, 0, 10);

    lv_obj_t * label2 = lv_label_create(bg);
    lv_label_set_long_mode(label2, LV_LABEL_LONG_CLIP);
    // lv_label_set_text(label2, "second line");
    lv_label_set_text(label2, "ХОРОШО");
    lv_obj_set_size(label2, 88, 20);
    lv_obj_align(label2, LV_ALIGN_TOP_LEFT, 0, 40);

    lv_obj_t * label3 = lv_label_create(bg);
    lv_label_set_long_mode(label3, LV_LABEL_LONG_CLIP);
    lv_label_set_text(label3, "thrid line");
    lv_obj_set_size(label3, 88, 20);
    lv_obj_align(label3, LV_ALIGN_TOP_LEFT, 0, 70);
}

### Expected behavior

Russian can be shown in V8.2

### Screenshots or video

[image]

[image]

yin2021, 2022-11-09:
> Your font should contain the Russian characters. You can easily create a new font with the online font converter: https://lvgl.io/tools/fontconverter

Thank you for your reply. When we create a label and do not specify any style, the problem will happen.

embeddedt, 2022-11-09:
I could be wrong but I don’t think the default fonts contain Russian
characters either.

pete-pjb, 2022-11-09:
Hi @yin2021 ,

I would suggest setting the default font in ```lv_conf.c``` to your own font:

```
/*Always set a default font*/
#define LV_FONT_DEFAULT &lv_font_my_russian_14
```

I hope that helps.

Kind Regards,

Pete

yin2021, 2022-11-10:
> Hi @yin2021 ,
> 
> I would suggest setting the default font in `lv_conf.h` to your own font:
> 
> ```
> /*Always set a default font*/
> #define LV_FONT_DEFAULT &lv_font_my_russian_14
> ```
> 
> I hope that helps.
> 
> Kind Regards,
> 
> Pete

Thank you！That's a good technique.

yin2021, 2022-11-10:
> I could be wrong but I don’t think the default fonts contain Russian characters either.
> […](#)

Thanks. In fact，we usually specify the style for label. So, it is no longer a problem for me.

kisvegabor, 2022-11-10:
You can also do this `lv_obj_set_style_text_font(screen1, &font, 0)`.

As the issue seems to be resolved I close it.

closed 2022-11-10 by kisvegabor