lvgl/lvgl #3789: LVGL Meter with multiple arcs More Property Adjustments

arilink, 2022-11-08:
I would like to make a similar control by referring to the example of Meter with multiple arcs, but I can only modify the background color of meter but cannot modify the background color of internal arc. Could you please tell me whether it is possible to modify the background color of internal arc and adjust the value of arc to achieve better effects? (Currently you cannot adjust the arc background color. You can only use arc start end to adjust the color of the area to be displayed.)

kisvegabor, 2022-11-09:
I added a simple change to the `meter` to handle the bg img too: d274523be

The `meter`'s API will be updated in lvgl v9, so I haven't added a API for it, but it can be used easily like this:
```c
    indic->type_data.arc.src = &image;
```

arilink, 2022-11-10:
Ok, I'll try and get back to you later. Thank you very much

closed 2022-12-03 by lvgl-bot