lvgl/lvgl #3790: some error when define LV_USE_FFMPEG =1

kay54068, 2022-11-08:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

i got some error when  define  LV_USE_FFMPEG.
```
[Error]	(5.201, +0)	 decoder_info: ffmpeg can't get image header 	(in lv_ffmpeg.c line #266)
[Error]	(5.231, +30)	 ffmpeg_get_img_header: Could not open source file A:/home/psx/dev/panelapp/src/resources/main_status_usb.png 	(in lv_ffmpeg.c line #584)
```
but the screen is can show image. 

### To Reproduce

```c
 lv_obj_t * img = lv_img_create(lv_scr_act());
 lv_img_set_src(img, "A:/home/psx/dev/panelapp/src/resources/main_status_usb.png");
```

### Expected behavior

### Screenshots or video

kisvegabor, 2022-11-09:
>  lv_img_set_src(img, "A:/home/psx/dev/panelapp/src/resources/main_status_usb.png");

`ffmpeg` [doesn't use LVGL file system API](https://docs.lvgl.io/master/libs/ffmpeg.html#usage). So the path should be:
`lv_img_set_src(img, "/home/psx/dev/panelapp/src/resources/main_status_usb.png");`

kay54068, 2022-11-09:
It's work fine for me.
thanks. 

add a question:
the  LVGL file system API is support  [lv_web_emscripten ](https://github.com/lvgl/lv_web_emscripten)  solution ?

kisvegabor, 2022-11-10:
>  the LVGL file system API is support [lv_web_emscripten ](https://github.com/lvgl/lv_web_emscripten)solution ?

It's a little bit complicated as you cannot simply access external files from a browser. First you need to add the files via a JS API to emscripten in order to make them visible for LVGL. 

We did it in a project but was pure pain. :slightly_smiling_face:  So if possible I suggest trying to avoid it.

kay54068, 2022-11-10:
I am interested in accessing external files from a browser on emscripten solution. the project is open source ?

how to to avoid it. ?   use C array if i want call  lv_img_set_src to show image on [lv_web_emscripten ](https://github.com/lvgl/lv_web_emscripten)solution ?

I thought it was as simple as just setting -preload-file,  sorry I was wrong ><
https://emscripten.org/docs/porting/files/packaging_files.html

kisvegabor, 2022-11-11:
> I am interested in accessing external files from a browser on emscripten solution. the project is open source ?

It's not open source unfortunately. Finally we have used the [NODEFS file system](https://emscripten.org/docs/api_reference/Filesystem-API.html#filesystem-api-nodefs). 

> how to to avoid it. ? use C array if i want call lv_img_set_src to show image on [lv_web_emscripten ](https://github.com/lvgl/lv_web_emscripten)solution ?

Yes, for images in C the simplest is to use arrays.

closed 2022-12-04 by lvgl-bot