lvgl/lvgl #3792: suggestion for LV_LABEL_LONG_WRAP and LV_LABEL_LONG_BREAK mode

yin2021, 2022-11-09:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I met the same proble #2150. And I set #define LV_TXT_LINE_BREAK_LONG_LEN 1, which can meet my need. However,
When a label been set LV_LABEL_LONG_WRAP and LV_LABEL_LONG_BREAK mode, it can break in the middle of a word without any connector, for example "-".
[image]

### To Reproduce

void lv_example_label_1(lv_obj_t *par)
{
    lv_obj_t * bg = lv_obj_create(lv_scr_act());
    lv_obj_set_size(bg, 200, 200);
    lv_obj_align(bg, LV_ALIGN_TOP_LEFT, 0, 0);

    //LV_LABEL_LONG_WRAP
    lv_obj_t * label3 = lv_label_create(bg);
    lv_label_set_long_mode(label3, LV_LABEL_LONG_WRAP);
    lv_label_set_text(label3, "sweaterdres");
    lv_obj_set_width(label3, 90);
    lv_obj_align(label3, LV_ALIGN_TOP_LEFT, 0, 80);

    label3 = lv_label_create(bg);
    lv_label_set_long_mode(label3, LV_LABEL_LONG_WRAP);
    lv_label_set_text(label3, "-ses");
    lv_obj_set_width(label3, 90);
    lv_obj_align(label3, LV_ALIGN_TOP_LEFT, 0, 95);
}

### Expected behavior

[image]

### Screenshots or video

[image]

kisvegabor, 2022-11-10:
To correctly do this we needed to have a hyphenation engine. As it should support multiple language it should be modular, but as it affects the typesetting of letters the text layout engine should be updated too. Long story short, it seems very complicated to do it correctly. 

However in Libre office write also this is the default behavior:

[image]

yin2021, 2022-11-10:
> To correctly do this we needed to have a hyphenation engine. As it should support multiple language it should be modular, but as it affects the typesetting of letters the text layout engine should [...]

You're right. Thank you!

kisvegabor, 2022-11-13:
I think we can do much here with a reasonable effort so I close this issue. :slightly_frowning_face:

closed 2022-11-13 by kisvegabor