lvgl/lvgl #3799: Online C code editor

kisvegabor, 2022-11-14:
### Introduce the problem

We have already discussed that it'd be great to have an online C editor with a builtin compiler for LVGL. 

### Examples and cases

It'd make it easy 
- to try out LVGL (lower barrier for entry)
- to report bugs from a known LVGL version and environment (have easy to reproduce issues)

### Suggested solution

GitHub Codespaces is exactly this. Here Is what I know so far:
- It's free for individuals in 60 hours/month (should be enough)
- Has VSCode editor
- We can build there too
- People need to fork our repo to start Codespaces on it

So we could have `lv_port_codespace` repo based on Emscripten but it'd be better to make people fork LVGL directly (just to have better stats on the main repo).

So the steps the user need to do could be:
1. Fork LVGL
2. Open Codespaces on it
3. run `scripts/codespace_init.sh` which clones `emsdk`, initializes it, creates an `lv_conf.h`, a `main.c` and the emscripten drivers. The image and font converter can be installed too and images and fonts can be downloaded with `wget` (or maybe there are better solutions too). There should be a `ui.c` file too which can be shared to report bugs. 
4. Have a `codespace_build.sh` script which builds and runs (i.e. opens `index.html`) the projects.

What do you think?

cc @embeddedt

We could do the same for the bindings too so I cc  
@amirgon (I know MP already has a similar solution)
@pikasTech 
@Sxs7513
@s-hadinger
(@GorgonMeducer)

pikasTech, 2022-11-14:
Maybe pikascript binding can be merged to the lv_port_codespace? Just enable a MACRO `LV_USING_PIKASCRIPT` to launch it.

kisvegabor, 2022-11-14:
I still don't know how to manage the bindings. My approach is that the binding (either in source code or pre-compiled form) should be available with drivers to display the UI. 

Actually it's not 100% that we need GitHub Codespaces for that. We already have an online simulator for [MicroPypthon](https://sim.lvgl.io/v8.3/micropython/ports/javascript/index.html). Maybe we could do the same with PikaScript too. 

I'm really not sure, just thinking aloud.  The goal would be to make the bindings (and LVGL in general) closer to the users, i.e. make it easier to get started and make experiments.

s-hadinger, 2022-11-14:
I love the idea. Berry binding should be easier since you don't actually need to compile anything. It could be similar to the Python emulator.

Berry already runs with Emscripten, it just needs to add a REPL and a virtual screen like in the Python emulator.

amirgon, 2022-11-14:
> @amirgon (I know MP already has a similar solution)

For LVGL+Micropython we already have two different online simulators today:
- The [javascript simulator](https://sim.lvgl.io/v8.3/micropython/ports/javascript/index.html) you mentioned. Very quick to start and easy to use. I use it a lot to reproduce simple things and share online code snippets.
- [GitPod simulator](https://gitpod.io/#https://github.com/lvgl/lv_micropython). GitPod is very much like GitHub CodeSpaces ([comparison](https://www.gitpod.io/vs/github-codespaces)) in the sense that you get an online VSCode instance where you can build and run the LVGL C code (with Micropython).  
One difference is that in GitPod the user doesn't need to go through all the steps you mentioned (fork, open codespace, run some scripts etc). The user just clicks a button and waits (a little too long actually) while VSCode opens in his browser with a built lv_micropython and a running demo application GUI without the user needing to to anything. At that point the user can either play with the GUI, edit the code, build, etc.  

Ideally, I think it would be nice to have it all in the browser hosted from a static site (such as GH-pages) as we do it in the Micropython js simulator.  
Emscripten [can actually be self hosted](https://github.com/emscripten-core/emscripten/issues/6432) so theoretically you could build LVGL in the browser and then run the built LVGL in the browser, without a backend server do the work for you. I'm not sure how mature/stable/fast it is, or how much work is involved, but it sure sounds like an interesting challenge :smile:

embeddedt, 2022-11-15:
I think self-hosted Emscripten is an impressive and promising concept, but it is too slow for our purposes if it takes 15-30 seconds (at least for me) to compile a hello world application.

> Is this long wait time related to MicroPython? Could it be faster with a plain LVGL?

When we looked into it before I believe the issue was that it had to pull the container environment internally every time and their system was not very fast at that.

kisvegabor, 2022-11-15:
It means  Codespaces seems the most viable option. I don't think the steps are too difficult:
1. Fork LVGL (Click a [link](https://github.com/lvgl/lvgl/fork) to get to the fork site)
2. Run Codespaces (Click a [link](https://github.com/codespaces/new) to get to repo select page)
3. ~Run `python scripts/codspace_setup.py`~ Use customization via [`.devcontainer` file](https://docs.github.com/en/codespaces/setting-up-your-project-for-codespaces/introduction-to-dev-containers) `ui.c` can be opened automatically.
4. Run `python scripts/codspace_build.py` to build an open `index.html`. Actually I couldn't open a HTML file in Codespaces yet... But I hope it's manageable somehow.

kisvegabor, 2022-11-30:
> Actually I couldn't open a HTML file in Codespaces yet... But I hope it's manageable somehow.

Seemingly it's more complicated than I anticipated. I couldn't open a HTML file yet. :slightly_frowning_face:

kisvegabor, 2023-03-28:
Good news! I've just made it work.

You just need to click the add button here:
[image]

Wait 1-2 minutes while environment is setting up and call `./build_all.sh`. 

All users have 120 free core-hours per month (I configured CPU 4 cores by default, so 30 hours per month).

I will add a ui.c file too, where people can add platform independent LVGL code and use it report and reproduce issues too.

Please try it out and let me know what you think.

kisvegabor, 2023-04-12:
It still needs to be documented.

closed 2023-05-05 by lvgl-bot

AndryPol, 2024-08-17:
- [ ] .. Рдщдббдщ8ж б... 5ио9щщ
. Юдлдбьрг68 9ш6г