lvgl/lvgl #3802: lv_canvas_set_px_color() causes flickering

johado, 2022-11-15:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

When writing a simple page that acts as a simple paint program to verify the touch performance on prototype hardware I see strange things.
I'm using a canvas and draw with lv_canvas_set_px_color() in the event callback.
This works fine in the -natrive build when running on a Linux PC that uses SDL (simulator),
but I get flickering when running it on target hardware.
Same problem for both FBDEV and DRM backend.

Problem goes away if I also draw a rect using lv_canvas_draw_rect()
Found using 8.3.0, but same problem in 8.3.3

### To Reproduce

### Expected behavior

Smooth drawing experience without flickering - same behaviour on both sdl, fbdev and drm backend.

### Screenshots or video

kisvegabor, 2022-11-19:
How does this flickering exactly look like? Could you send a video?

johado, 2022-11-23:
https://user-images.githubusercontent.com/25119818/203560574-c54646c3-b328-4682-a189-d451471e09f1.mp4

kisvegabor, 2022-11-24:
I've just tried out and it works well for me in the simulator. 

I've simplified your code a little bit. Does it make any difference?
```c
lv_obj_t * canvas;

static void
canvas_event_cb(lv_event_t *e)
{
  lv_event_code_t code = lv_event_get_code(e);
  if (code == LV_EVENT_PRESSING) {
      static lv_draw_rect_dsc_t rect_dsc;
      static int rect_init = 0;
      if (!rect_init) {
          lv_draw_rect_dsc_init(&rect_dsc);
          rect_dsc.radius = 10;
          rect_dsc.bg_opa = LV_OPA_COVER;
          rect_dsc.bg_grad.dir = LV_GRAD_DIR_HOR;
          rect_dsc.bg_grad.stops[0].color = lv_palette_main(LV_PALETTE_RED);
          rect_dsc.bg_grad.stops[1].color = lv_palette_main(LV_PALETTE_BLUE);
          rect_dsc.border_width = 2;
          rect_dsc.border_opa = LV_OPA_90;
          rect_dsc.border_color = lv_color_white();
          rect_dsc.shadow_width = 5;
          rect_dsc.shadow_ofs_x = 5;
          rect_dsc.shadow_ofs_y = 5;
          rect_init = 1;
      }

      lv_point_t p;
      lv_indev_t *indev = lv_indev_get_act();
      lv_indev_get_point(indev, &p);
      printf("canvas: %i,%i\n", (int)p.x, (int)p.y);
      lv_canvas_set_px_color(canvas, p.x, p.y, lv_color_hex(0xFFFFFF));
      /* !!!!!!!!!!!!!!! Problem goes away when uncommenting below !!!!!! */
      //        lv_canvas_draw_rect(canvas, p.x, p.y, 3, 3, &rect_dsc);
  }
}

static uint8_t cbuf[LV_CANVAS_BUF_SIZE_TRUE_COLOR(CANVAS_WIDTH, CANVAS_HEIGHT)];

lv_obj_t * draw_canvas(lv_obj_t *parentin)
{
    canvas = lv_canvas_create(parentin);
    lv_canvas_set_buffer(canvas, cbuf, CANVAS_WIDTH, CANVAS_HEIGHT, LV_IMG_CF_TRUE_COLOR);
    lv_canvas_fill_bg(canvas, lv_color_hex(0x000000), LV_OPA_COVER);

    lv_color_t framecolor = lv_color_hex(0x10FF10);
    lv_coord_t x = 0;
    for (x=0; x < CANVAS_WIDTH; x++) {
      lv_canvas_set_px_color(canvas, x, 0, framecolor);
      lv_canvas_set_px_color(canvas, x, CANVAS_HEIGHT-1, framecolor);
    }

    lv_coord_t y = 0;
    for (y=0; y < CANVAS_HEIGHT; y++) {
      lv_canvas_set_px_color(canvas, 0, y, framecolor);
      lv_canvas_set_px_color(canvas, CANVAS_WIDTH - 1, y, framecolor);
    }

    lv_obj_add_flag(canvas, LV_OBJ_FLAG_CLICKABLE);
    lv_obj_add_event_cb(canvas, canvas_event_cb, LV_EVENT_ALL, canvas);
    return canvas;
}

```

A few questions:
1. How did you allocate `cbuf`?
2. What if you call `lv_obj_invalidate(canvas)` after `lv_canvas_set_px_color`?

johado, 2022-11-24:
It works fine for me as well in the simulatror with SDL.
1. cbuf is allocated/initialised as a static variable:
static uint8_t cbuf[LV_CANVAS_BUF_SIZE_TRUE_COLOR(CANVAS_WIDTH, CANVAS_HEIGHT)];

2. No change (lv_canvas_set_px_color() does that already.)
The other draw functions uses the "fake_disp" and disp_refreshing functions - e.g. drawRect that works smoothly.

kisvegabor, 2022-11-25:
I have no idea why `draw_rect` solves the issue. Although its more complicated, but in he end it just draws something to the buffer similarly to `set_px`.

Could you check after `set_px` the given `x` and `y` point of the buffer is really set?

Can it be a caching issue? Did you need to invalidate D cache somewhere?

closed 2022-12-18 by lvgl-bot