lvgl/lvgl #3806: free png_data resources

glory-man, 2022-11-17:
lodepng_load_file() could exit with error code
https://github.com/lvgl/lvgl/blob/d219960de0ca00122f131e8dbdf55b153eb8a383/src/libs/png/lv_png.c#L163
but problem may be inside of
https://github.com/lvgl/lvgl/blob/d219960de0ca00122f131e8dbdf55b153eb8a383/src/libs/png/lodepng.c#L380
In this case memory allocated in lodepng_load_file()
https://github.com/lvgl/lvgl/blob/d219960de0ca00122f131e8dbdf55b153eb8a383/src/libs/png/lodepng.c#L402
not freed, what can cause a leakage in case of problems with open/seek/read files

kisvegabor, 2022-11-20:
Could you send a Pull request with a fix?

glory-man, 2022-12-05:
PR https://github.com/lvgl/lvgl/pull/3853 created

kisvegabor, 2022-12-21:
The PR is merged, so closing this issue.

closed 2022-12-21 by glory-man