lvgl/lvgl #3812: High CPU usage will occur after a single gif stop

qingliyu, 2022-11-19:


qingliyu, 2022-11-19:
After the last bug(https://github.com/lvgl/lvgl/issues/3760) was repaired, a recent test found that the CPU usage was high after the git stopped playing.

qingliyu, 2022-11-19:
[image]

qingliyu, 2022-11-19:
Code after I fix it
~~~c
void lv_gif_restart(lv_obj_t * obj)
{
    lv_gif_t * gifobj = (lv_gif_t *) obj;
    gd_rewind(gifobj->gif);
    lv_timer_resume(gifobj->timer);
    lv_timer_reset(gifobj->timer);
}
static void next_frame_task_cb(lv_timer_t * t)
{
    lv_obj_t * obj = t->user_data;
    lv_gif_t * gifobj = (lv_gif_t *) obj;
    uint32_t elaps = lv_tick_elaps(gifobj->last_call);
    if(elaps < gifobj->gif->gce.delay * 10) return;

    gifobj->last_call = lv_tick_get();

    int has_next = gd_get_frame(gifobj->gif);
    if(has_next == 0) {
        /*It was the last repeat*/
        if(gifobj->gif->loop_count <= 1) {
            lv_res_t res = lv_event_send(obj, LV_EVENT_READY, NULL);
            lv_timer_pause(t);
            if(res != LV_FS_RES_OK) return;
        }
        else {
            if(gifobj->gif->loop_count > 1)  gifobj->gif->loop_count--;
            gd_rewind(gifobj->gif);
        }
    }

    gd_render_frame(gifobj->gif, (uint8_t *)gifobj->imgdsc.data);

    lv_img_cache_invalidate_src(lv_img_get_src(obj));
    lv_obj_invalidate(obj);
}
~~~

kisvegabor, 2022-11-20:
Could you send a Pull request?

commit 51046911 (2022-12-01): fix(gif): stop the timer on looping end to avoid high CPU usage (#3841)

fixes  #3812
closed 2022-12-01 by kisvegabor