lvgl/lvgl #3813: lv_tabview can not scroll when use Cmake build

haidongpan, 2022-11-20:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce
1. add the demo of lv_demo_widgets in main.c
2. build it by CMakeList
### Expected behavior
Expected the lv_tabview can scroll and switch normally

### Screenshots or video

https://user-images.githubusercontent.com/107170273/202888507-453f7796-b253-4f37-889b-96bc2617d864.mp4

embeddedt, 2022-11-20:
We will need some more details. What platform are you using? Which LVGL commit/version?

haidongpan, 2022-11-21:
The version of lvgl i used is v8.3, and i portting it on linux-5.10

haidongpan, 2022-11-22:
i also download the latest source code, which is master branch, lv_tabview also can not scroll, and can not change tab 

https://user-images.githubusercontent.com/107170273/203196676-f84931fe-d8de-46b9-a726-f30e9c4693c4.mp4

haidongpan, 2022-11-23:
does this problem have have confirmed ?

kisvegabor, 2022-11-24:
It's very strange. I wonder what could be the difference between make and Cmake. They both `glob` all source files so the only difference could be the build settings. I think mainly the optimization level can affect these strange behavior.

Is there any difference you change the optimization level?

You can also enable `TRACE` level logging in `lv_conf.h` and check the difference.

haidongpan, 2022-11-24:
I have not change the optimization level, just download source code from github, does your local project have the same phenomenon as my project

haidongpan, 2022-11-24:
all source code build in execute file by Makefile,  and source code build in execute file and  liblvgl.so liblvgl_demos.a liblvgl_samples.a dynamic library, i do not know why

haidongpan, 2022-11-25:
[make.log](https://github.com/lvgl/lvgl/files/10089594/make.log)
[cmake.log](https://github.com/lvgl/lvgl/files/10089595/cmake.log)
1. attachment is the lv_log when info debug is enable, i found no diffrence in each other
2. i make sure that liblvgl, liblvgl_demos and liblvgl_examples will be generate when build by cmake, you can check is in custom.cmake file

haidongpan, 2022-11-25:
[image]
attachment is the place that liblvgl, liblvgl_demos and liblvgl_examples define

haidongpan, 2022-11-25:
the execute file will link the dynamic library when it is in runtime

haidongpan, 2022-11-28:
did you have any more suggestion ?

kisvegabor, 2022-11-28:
In the `flush_cb` please `printf` the coordinates of the flushed area. Do you see large screens sized areas when the screen transition should happen?

haidongpan, 2022-12-05:
You are write, yes, I found out the reason that due this case, the optimize level is O2 in my project. i do not known which part is optimized.

kisvegabor, 2022-12-07:
Normally optimization shouldn't matter. I usually use O3 in production and never found an issue like this. 

Anyway, at least now we know that not build system causing the issue.

haidongpan, 2022-12-09:
thank you very much

kisvegabor, 2022-12-11:
Anyway, I'd be curious which part gets broken with optimization. Do you see any warnings?

haidongpan, 2022-12-12:
if i have time i will check these details

closed 2023-01-03 by lvgl-bot