lvgl/lvgl #3824: Include lvgl.mk in Makefile only works when placing lvgl folder in workspace root directory.

lclancey, 2022-11-23:
### Introduce the problem

I'm not familiar with makefile so I'm not sure if I did something wrong.

This new lvgl.mk code in e42f01a38ad43140cdeb99799c737e9f5e668103  seems only effective when lvgl located in workspace root directory (`${workspaceFolder}/lvgl`).
### Examples and cases

I've tried putting `lvgl` folder in a place like `${workspaceFolder}/thirdpartylib/lvgl`, and add 
```c
include thirdpartylib/lvgl/lvgl.mk
C_SOURCES += \
CSRCS 
```
in `${workspaceFolder}/Makefile`,  `make` command will not correctly resolve  C_SOURCES . 

### Suggested solution

I think the old way that user pass some parameter like `LVGL_PATH` is necessary.

kisvegabor, 2022-11-24:
The `?=` in [`LVGL_PATH ?= ${shell pwd}/lvgl`](https://github.com/lvgl/lvgl/blob/master/lvgl.mk#L1) means: set `LVGL_PATH` to some default value if it's not set yet. So in your Makefile you can do something like `LVGL_PATH =./thirdpartylib/lvgl/`

lclancey, 2022-11-24:
Got it.
Sounds like a elegant solution.

closed 2022-11-24 by lclancey