lvgl/lvgl #3828: Abnormal screen

chenyinen, 2022-11-24:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

version：lvgl-release-v8.3
framework：arm 32
The mouse cannot move and the animation cannot be displayed. The animation will be generated only when the mouse is clicked and moved, but the screen will be blurred.
### To Reproduce

Pseudocode:
void * tick_thread (void *args)
{
    while(1) {
    usleep(10*1000); /*Sleep for 10 millisecond*/
    lv_tick_inc(10); /*Tell LVGL that 5 milliseconds were elapsed*/
    }
}
int main(){
    xxxx
   lv_init();
  static lv_disp_draw_buf_t disp_buf;
    lv_disp_draw_buf_init(
        &disp_buf,
       fb_addr,// fb address
        NULL,
        WIDTH * HEIGHT);
static lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);
    disp_drv.hor_res = WIDTH;
    disp_drv.ver_res = HEIGHT;
    disp_drv.flush_cb = dummy_flush_cb;
    disp_drv.draw_buf = &disp_buf;
    disp_drv.rounder_cb = NULL;
    lv_disp_drv_register(&disp_drv);

   lv_indev_drv_t indev_drv;
     lv_indev_t* mouse_dev;
    lv_indev_drv_init(&indev_drv);
    indev_drv.type = LV_INDEV_TYPE_POINTER;
    indev_drv.read_cb = lv_test_mouse_read_cb;
    mouse_dev = lv_indev_drv_register(&indev_drv);
   lv_obj_t * mouse_cursor = lv_img_create(lv_scr_act());
    lv_img_set_src(mouse_cursor, LV_SYMBOL_HOME);
    lv_indev_set_cursor(mouse_dev, mouse_cursor);
   lv_demo_widgets();
    while(1) {
        /* Periodically call the lv_task handler.
         * It could be done in a timer interrupt or an OS task too.*/
        lv_task_handler();
        usleep(10000);       /*Just to let the system breath
} 
### Expected behavior

lv_demo_widgets() normal work！
### Screenshots or video

[image]
The reason for the color difference on the display screen is that my display format is 1555, while the lvgl format is 565. The problem with my relationship is that painting is abnormal

kisvegabor, 2022-11-25:
How does your `flush_cb` look like?

chenyinen, 2022-11-25:
do nothing
static lv_disp_draw_buf_t disp_buf;
lv_disp_draw_buf_init(
&disp_buf,
fb_addr,// fb address
NULL,
WIDTH * HEIGHT);
fb_addr is linux fb。
in flush_cb i just call  lv_disp_flush_ready(disp_drv);

kisvegabor, 2022-11-25:
Ah, I see now. So you are rendering directly into the frame buffer. 

You should set `disp_drv.direct_mode=1` in this case.

chenyinen, 2022-11-25:
Thank you. I'll try your suggestion next week！
 I think this is a good gui framework！

closed 2022-11-26 by chenyinen