lvgl/lvgl #3829: Can't rotate when enable direct-mode

Lzw655, 2022-11-24:
Setup: STM32H7 (no RTOS) with a 320x480px display (8bit parallel interface on FMC) which is mounted rotated cw by 90°, so my GUI is 480x320px.

Without direct mode it looks like this, which is how it's supposed to look:
[image]

But with direct mode it looks like this:
[image]

Code:
```c
#ifndef DIRECTMODE
static lv_color_t buf1[320 * 480 / 10];
#endif

void GUI_Init(void) {
	lv_init();
#ifndef DIRECTMODE
	lv_disp_draw_buf_init(&draw_buf, buf1, NULL, 320 * 480 / 10);
#else
	lv_disp_draw_buf_init(&draw_buf, (uint32_t *)0x24035000, NULL, 320 * 480);  
// 0x24035000 is the start of my sram display buffer
#endif
	static lv_disp_drv_t disp_drv = {}; 
	lv_disp_drv_init(&disp_drv); 
	disp_drv.flush_cb = my_disp_flush; 
	disp_drv.draw_buf = &draw_buf;  
	disp_drv.hor_res = 320;
	disp_drv.ver_res = 480; 
	disp_drv.rotated = LV_DISP_ROT_90;
	disp_drv.sw_rotate = 1;
#ifdef DIRECTMODE
	disp_drv.direct_mode = 1;
#endif
	lv_disp_drv_register(&disp_drv);
//...
}

void my_disp_wait(lv_disp_drv_t * disp) {
	lv_disp_flush_ready(disp);
}

void my_disp_flush(lv_disp_drv_t * disp, const lv_area_t * area, lv_color_t * color_p) {
#ifndef DIRECTMODE
	for(y = area->y1; y <= area->y2; y++) {
		for(x = area->x1; x <= area->x2; x++) {
			LCD_DrawDot(x, y, (*color_p).full);
			color_p++;
		}
	}
#endif
	lv_disp_flush_ready(disp);
}
```

In normal mode, it renders the rotated data in the 1/10 size buffer and that gets flushed per software into my full sized buffer (320x480px). (The posted images are the content of this buffer, just rotated into landscape orientation.) The microcontroller continuously writes this buffer per DMA at 10 Hz to the displays GRAM.

Trying different settings just results in other stange renderings. Here i swapped `disp_drv.hor_res` and `disp_drv.ver_res` in direct mode.
[image]

Ultimately, what I want to achieve is to skip the step where the software has to copy the smaller rendering buffer to an area in my full sized buffer. The idea is to get a little bit more fps (right now i get only about 5fps).

_Originally posted by @fraengers in https://github.com/lvgl/lvgl/issues/2460#issuecomment-928236641_

Lzw655, 2022-11-24:
I got the same problem as the above comment. I'm using [ESP32-S3-LCD-EV-BOARD ](https://github.com/espressif/esp-dev-kits/tree/master/esp32-s3-lcd-ev-board) with 800x480 RGB LCD. Normally I have a good UI display with dircet-mode enabled. But when I rotate the screen by software(its code is like below), my UI becomes strange. It looks like it's jumping back and forth
```
disp_drv.sw_rotate = 1;
disp_drv.rotated = LV_DISP_ROT_180;
```

Lzw655, 2022-11-24:
No rotation:

https://user-images.githubusercontent.com/109257001/203758564-80a0b3d2-8688-45c4-be11-13c3f136d2e2.mp4

After rotation:

https://user-images.githubusercontent.com/109257001/203758106-15c95864-25ae-4483-b1d1-f2ce7e94983f.mp4

embeddedt, 2022-11-24:
The approach I used for software rotation makes the assumption that there is an intermediate buffer (as the rotation is done within that buffer) so I am pretty sure it won't work at all in direct mode. However, this should be documented or produce an error to prevent confusion.

Lzw655, 2022-11-25:
What about full refresh mode? Doesn't it work with rotation? I'm also having some issues combining these two operations.

embeddedt, 2022-11-25:
Full refresh is also not supported unfortunately. :( There is a warning output in the log if you try to use this combination.

Lzw655, 2022-11-25:
Got it. Thanks for your answer. Hope it'll be supported someday...

closed 2022-11-25 by Lzw655

weilian1977, 2023-01-27:
关注一下