lvgl/lvgl #3835: MCU crash after calling lv_obj_del on other thread

jiabuda, 2022-11-26:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
mcu reset after a normal call lv_obj_del() when using lvgl with nimble lib

### To Reproduce
Hi, I am integrating Nimble lib and LVGL 8.3.3 on ESP32 Dev Board, I try to call a screen switch after BLE connection established callback, the MCU reset and report a error on calling lv_obj_del, the struct lv_task is no longer exist on LVGL8, so if it is safe to call lvgl function on other thread like ble or wifi?

the code is like this
```
void displayModeChooseScreen()
{
    modeChooseScreen = lv_obj_create(NULL);
    levelButton = lv_btn_create(modeChooseScreen);
    battleButton = lv_btn_create(modeChooseScreen);
    levelLabel = lv_label_create(levelButton);
    battleLabel = lv_label_create(battleButton);
    lv_label_set_text(levelLabel, "Level Game");
    lv_label_set_text(battleLabel, "Bluetooth Battle");
    lv_obj_align(levelButton, LV_ALIGN_TOP_MID, 0, 40);
    lv_obj_align_to(battleButton, levelButton, LV_ALIGN_OUT_BOTTOM_MID, 0, 40);
    lv_disp_load_scr(modeChooseScreen);
    deleteAllScreenExcept(modeChooseScreen);
}

void deleteAllScreenExcept(lv_obj_t *remain)
{
    if (pairScreen != NULL && pairScreen != remain)
    {
        lv_obj_clean(pairScreen);
        lv_obj_del(pairScreen);
        pairScreen = NULL;
    }

    if (modeChooseScreen != NULL && modeChooseScreen != remain)
    {
        lv_obj_clean(modeChooseScreen);
        lv_obj_del(modeChooseScreen);
        modeChooseScreen = NULL;
    }

    if (battleQuestionScreen != NULL && battleQuestionScreen != remain)
    {
        lv_obj_clean(battleQuestionScreen);
        lv_obj_del(battleQuestionScreen);
        battleQuestionScreen = NULL;
    }
}
```

here is the stack trace
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400fb2ca  PS      : 0x00060730  A0      : 0x800fc2bd  A1      : 0x3ffdad40
0x400fb2ca: _lv_event_mark_deleted at D:/git/giiker/esp32_nimble/components/lvgl/src/core/lv_event.c:156

A2      : 0x3ffc5d20  A3      : 0x3ffc58f8  A4      : 0x00000020  A5      : 0x3ffc5d20
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000075  A9      : 0x800d1300
A10     : 0x00000001  A11     : 0x3ffc58f8  A12     : 0x3ffc591c  A13     : 0x00000075
A14     : 0x0000009b  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000079  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff0

Backtrace:0x400fb2c7:0x3ffdad400x400fc2ba:0x3ffdad60 0x4014ff9b:0x3ffdad80 0x4014ffa9:0x3ffdada0 0x40100dc8:0x3ffdadc0 0x40100ef4:0x3ffdade0 0x400fab9e:0x3ffdae00 0x400fa9ae:0x3ffdae20 0x400d5f56:0x3ffdae50 0x400d6515:0x3ffdae70 0x4014f122:0x3ffdaea0 0x400d6c5e:0x3ffdaec0 0x400d6cad:0x3ffdaee0 0x400d8987:0x3ffdaf00 0x400d89bf:0x3ffdaf20 0x400d9cf5:0x3ffdaf40 0x400da2f5:0x3ffdaf60 0x400d858d:0x3ffdaf80 0x400d844e:0x3ffdafb0 0x400d7b45:0x3ffdafe0 0x400d7b57:0x3ffdb000 0x400e06da:0x3ffdb020 0x400d63a6:0x3ffdb040 0x40093cc1:0x3ffdb060
0x400fb2c7: _lv_event_mark_deleted at D:/git/giiker/esp32_nimble/components/lvgl/src/core/lv_event.c:155

0x400fc2ba: lv_obj_destructor at D:/git/giiker/esp32_nimble/components/lvgl/src/core/lv_obj.c:435

0x4014ff9b: _lv_obj_destruct at D:/git/giiker/esp32_nimble/components/lvgl/src/core/lv_obj_class.c:136 (discriminator 1)

0x4014ffa9: _lv_obj_destruct at D:/git/giiker/esp32_nimble/components/lvgl/src/core/lv_obj_class.c:143

0x40100dc8: obj_del_core at D:/git/giiker/esp32_nimble/components/lvgl/src/core/lv_obj_tree.c:389

0x40100ef4: lv_obj_clean at D:/git/giiker/esp32_nimble/components/lvgl/src/core/lv_obj_tree.c:94

0x400fab9e: deleteAllScreenExcept at D:/git/giiker/esp32_nimble/components/screen/screenTotal.c:34

0x400fa9ae: displayBattleQuestionScreen at D:/git/giiker/esp32_nimble/components/screen/battleQuestionScreen.c:101

0x400d5f56: centralConnectCallback at D:/git/giiker/esp32_nimble/components/klotski/game.c:168

0x400d6515: blecent_on_disc_complete at D:/git/giiker/esp32_nimble/components/ble_gatt/central.c:163 (discriminator 13)

0x4014f122: peer_disc_complete at D:/git/giiker/esp32_nimble/components/ble_gatt/peer.c:82

0x400d6c5e: peer_disc_dscs at D:/git/giiker/esp32_nimble/components/ble_gatt/peer.c:211

0x400d6cad: peer_dsc_disced at D:/git/giiker/esp32_nimble/components/ble_gatt/peer.c:235

0x400d8987: ble_gattc_disc_all_dscs_cb at C:/Users/Endy/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:2577

0x400d89bf: ble_gattc_disc_all_dscs_err at C:/Users/Endy/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:2646

0x400d9cf5: ble_gattc_rx_err at C:/Users/Endy/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:4387

0x400da2f5: ble_att_clt_rx_error at C:/Users/Endy/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_clt.c:46

0x400d858d: ble_att_rx at C:/Users/Endy/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att.c:503

0x400d844e: ble_hs_hci_evt_acl_process at C:/Users/Endy/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:927

0x400d7b45: ble_hs_process_rx_data_queue at C:/Users/Endy/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:238

0x400d7b57: ble_hs_event_rx_data at C:/Users/Endy/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:532

0x400e06da: ble_npl_event_run at C:/Users/Endy/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:125
 (inlined by) nimble_port_run at C:/Users/Endy/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:78

0x400d63a6: blecent_host_task at D:/git/giiker/esp32_nimble/components/ble_gatt/central.c:377

0x40093cc1: vPortTaskWrapper at C:/Users/Endy/esp/esp-idf/components/freertos/port/xtensa/port.c:131
```

embeddedt, 2022-11-26:
No, you still have to use a mutex as outlined in the [documentation](http://docs.lvgl.io/8/porting/os.html#tasks-and-threads).

jiabuda, 2022-11-27:
> No, you still have to use a mutex as outlined in the [documentation](http://docs.lvgl.io/8/porting/os.html#tasks-and-threads).

Hi embeddebt, i try to take and give the semaphore around my ui update, it never crash again. thanks

closed 2022-11-27 by jiabuda