lvgl/lvgl #3838: Has the non antialiased drawing disappeared?

alepi, 2022-11-28:
Hi,

I have a possibly unusual use case where I have a big display ( a TV, it could be any size up to tens of inches ) and low resolution ( 320 x 240 ).

Antialiasing does not work well for me in that situation.

There used to be a LV_ANTIALIAS constant to set to 0 in lv_conf.h but it seems to have disappeared.

I tried messing around a bit with the latest master branch (8.3) code and I managed to get rid of antialias on text and images but not on rounded corners of rectangles.

Is there any chance we can get the LV_ANTIALIAS constant back?

FASTSHIFT, 2022-11-28:
Antialiasing parameters are located at: https://github.com/lvgl/lvgl/blob/master/src/hal/lv_hal_disp.h#L100
It is turned on by default: https://github.com/lvgl/lvgl/blob/master/src/hal/lv_hal_disp.c#L89

alepi, 2022-11-28:
Hi, thank you for your response.

That is indeed one of the changes I made when I talked about "messing around" with the code. I just grep'd for alias in the whole LVGL directory and changed that setting and the analogous one for images. Also, I set the default font to unscii 8.

I believe that rounded corners are drawn with some form of antialias regardless, that's why I opened this issue.

FASTSHIFT, 2022-11-28:
The anti-aliasing algorithm can only alleviate the aliasing problem, but it cannot completely solve it. After all, the accuracy of physical display is limited.

FASTSHIFT, 2022-11-29:
Sorry, I think I misunderstood you, it's true that lvgl doesn't seem to handle drawing without anti-aliasing.

kisvegabor, 2022-11-29:
It was moved to `disp_drv.antialiasing = 0;`.

FASTSHIFT, 2022-11-29:
> It was moved to `disp_drv.antialiasing = 0;`.

I tried turning that off, but the non-anti-aliased drawing doesn't seem to be taking effect.
I searched the code, it seems that lvgl only accesses this parameter when `layer_type != LV_LAYER_TYPE_NONE` (https://github.com/lvgl/lvgl/blob/master/src/core/lv_refr.c#L936), but later The software drawing does not continue to read this parameter.
Has non-anti-aliased rendering been removed?

alepi, 2022-11-29:
yep, like I said I had a look at the code too and it seems that text and images can be drawn without antialiasing, but rounded corners and possibly other drawing primitives still draw with antialiasing on

kisvegabor, 2022-12-01:
Do you use `master`? If so, please use the `release/v8.3`.

alepi, 2022-12-01:
I was using master indeed. I switched to 8.3. Still no joy.
I set
      disp_drv.antialiasing = 0;
after initialising it and still the rounded corners are not sharp and don't look good on a large display.
Also tried changing the code in lv_img.c and lv_hal_disp.c to initialise without anti aliasing but I still don't get sharp rounded corners.
Version 7 still does not antialias sharp corners.

kisvegabor, 2022-12-03:
Uhh, you are right. I was sure it's working fine in v8 but it really wasn't. 

I've pushed a fix.

commit 2c17b28a (2022-12-03): fix(darw): add back the disappeared antialising=0 support

fixes #3838
closed 2022-12-05 by alepi