lvgl/lvgl #3839: Delete obj or unsubscribe during msg notify cb

wrgallo, 2022-11-28:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

The `notify` msg implementation is as follow:
```C
static void notify(lv_msg_t * m)
{
    sub_dsc_t * s;
    _LV_LL_READ(&LV_GC_ROOT(_subs_ll), s) {
        if(s->msg_id == m->id && s->callback) {
            m->user_data = s->user_data;
            m->_priv_data = s->_priv_data;
            s->callback(m);
        }
    }
}
```

If the variable `_subs_ll` is modified during the `s->callback` function, (i.e. the object was deleted or unsubscribed), `s` will become a dangling pointer and there is a high chance a crash will happen here. 

**Is it intended to be like this?**

### To Reproduce
```c
static void slider_event_cb(lv_event_t* e) {
    lv_obj_t* slider = lv_event_get_target(e);
    int32_t v = lv_slider_get_value(slider);
    lv_msg_send(MSG_NEW_TEMPERATURE, &v);
}

static void label_event_cb(lv_event_t* e) {
    lv_obj_t* label     = lv_event_get_target(e);
    lv_msg_t* m         = lv_event_get_msg(e);
    const int32_t* v    = (int32_t*)lv_msg_get_payload(m);
    lv_label_set_text_fmt(label, "%d °C", *v);
    lv_obj_del(label); // this line crashes the application
    lv_msg_unsubscribe(label->user_data); // This would crash it in the same way
}

void lv_example_msg_1_modified(void)
{
    // Create a slider in the center of the display
    lv_obj_t * slider = lv_slider_create(lv_scr_act());
    lv_obj_center(slider);
    lv_obj_add_event_cb(slider, slider_event_cb, LV_EVENT_VALUE_CHANGED, NULL);

    // Create a label below the slider
    lv_obj_t * label = lv_label_create(lv_scr_act());
    lv_obj_add_event_cb(label, label_event_cb, LV_EVENT_MSG_RECEIVED, NULL);
    lv_label_set_text(label, "0%");
    lv_obj_align(label, LV_ALIGN_CENTER, 0, 30);

    // Subscribe the label to a message.
    label->user_data = lv_msg_subsribe_obj(MSG_NEW_TEMPERATURE, label, NULL);
}
```

### Expected behavior

It is possible to call `lv_timer_del` during a `lv_timer_cb_t`, or `lv_obj_remove_event_cb` during a `lv_event_cb_t`, so it is not intuitive that it is prohibited to unsubscribe during a msg cb, or delete the object which triggers an internal unsubscribe event callback (`obj_delete_event_cb`) which is the same.

wrgallo, 2022-11-28:
Maybe implementing the `notify` function as follow would fix this issue:

```C
static void notify(lv_msg_t * m) {
    sub_dsc_t* s = _lv_ll_get_head(&LV_GC_ROOT(_subs_ll));
    while (s) {
        sub_dsc_t* s_next = _lv_ll_get_next(&LV_GC_ROOT(_subs_ll), s);
        if (s->msg_id == m->id && s->callback) {
            m->user_data = s->user_data;
            m->_priv_data = s->_priv_data;
            s->callback(s, m);
        }
        s = s_next;
    }
}
```

kisvegabor, 2022-11-29:
Nice catch, I really wasn't think of that case. 

Your solution looks good. Can you send a [Pull request](https://docs.lvgl.io/master/CONTRIBUTING.html#pull-request)?

wrgallo, 2022-11-29:
> Nice catch, I really wasn't think of that case.
> 
> Your solution looks good. Can you send a [Pull request](https://docs.lvgl.io/master/CONTRIBUTING.html#pull-request)?

Actually this solutions still needs a little more work, it does not consider multiple unsubscribe or newer subscription. But I think it won't be hard to copy the same concept used inside `lv_timer_handler` to trigger all `lv_timer_t` callbacks properly.

I will figure a new approach before the pull request.

wrgallo, 2022-11-29:
### Observations

It should be made clear in documentation that currently it is not possible to call `lv_msg_send` during a `lv_event_cb_t` with code `LV_EVENT_MSG_RECEIVED`, which would cause undefined behavior, as the `notify` function would enter recursion, for that to be possible `lv_msg_t` would need to be buffered.

### Implementation to fix the current issue

- `sub_dsc_t` was changed to prevent multiple notifications
```C
typedef struct {
    uint32_t msg_id;
    lv_msg_subscribe_cb_t callback;
    void * user_data;
    void * _priv_data;      /*Internal: used only store 'obj' in lv_obj_subscribe*/
    uint8_t _checked;       /*Internal: used to prevent multiple notifications*/
} sub_dsc_t;
```
- two static variables serve as flags for changes in `_subs_ll`
```C
/**********************
 *  STATIC VARIABLES
 **********************/
static bool sub_dsc_deleted;
static bool sub_dsc_created;
```
- `sub_dsc_deleted` is set inside `lv_msg_unsubscribe`. And `sub_dsc_created` inside `lv_msg_subsribe`
- `_checked` is also set during `lv_msg_subsribe`
- A macro for trace could be added
```C
/**********************
 *      MACROS
 **********************/
#if LV_LOG_TRACE_MSG // TODO Add this to LV_CONF.h
#define MSG_TRACE(...) LV_LOG_TRACE(__VA_ARGS__)
#else
#define MSG_TRACE(...)
#endif
```
- Finally the `nofity` function is written as follows
```C
static void notify(lv_msg_t * m) {
    // First clear all sub_dsc_t checked flags
    sub_dsc_t* s;
    _LV_LL_READ(&LV_GC_ROOT(_subs_ll), s) {
        s->_checked = 0;
    }
    
    //Run all sub_dsc_t from the list
    do {
        sub_dsc_deleted = false;
        sub_dsc_created = false;
        s = _lv_ll_get_head(&LV_GC_ROOT(_subs_ll));
        while (s) {
            // get next element while current is surely valid
            sub_dsc_t* next = _lv_ll_get_next(&LV_GC_ROOT(_subs_ll), s);

            // Notify only once
            if (!s->_checked) {
                s->_checked = 1;

                // Check if this sub_dsc_t is about this msg_id
                if (s->msg_id == m->id && s->callback) {
                    // Notify
                    m->user_data    = s->user_data;
                    m->_priv_data   = s->_priv_data;
                    s->callback(s, m);
                }
            }

            // Load next or restart
            if (sub_dsc_created || sub_dsc_deleted) {
                MSG_TRACE("Start again because at least one sub_dsc_t was created or deleted");
                break;
            }
            s = next;
        }
    } while (s);
}
```

kisvegabor, 2022-12-01:
It sounds good. We are using a similar technique in [`lv_timer`](https://github.com/lvgl/lvgl/blob/51046911e48baf64015624541d905ce0ba37ad41/src/misc/lv_timer.c#L36-L37) and [`lv_anim`](https://github.com/lvgl/lvgl/blob/master/src/misc/lv_anim.c#L39).

PR #3847 (2022-12-11): fix(msg): able to create/delete msgs during notify
Creating or deleting subscription is now possible during a message notification.

Fixes: #3839
closed 2022-12-11 by kisvegabor