lvgl/lvgl #3842: Load screen after digital write

yaawl, 2022-11-29:
##Description

Hi i have a problem. I would need a load screen and after blink with led. But now when i click on the button led blink and only after that screen load. Where is a problem please ? Thank you very much.

##What MCU/Processor/Board and compiler are you using?

ESP32
3.5" inch TFT

##What LVGL version are you using?

8.3.3

##Code to reproduce

Add a code snippet which can run in the simulator. It should contain only the relevant code that compiles without errors when separated from your main code base.

```
void ta_event_show(lv_event_t *e) {
  lv_event_code_t code = lv_event_get_code(e);
  lv_obj_t *ebutton = lv_event_get_target(e);
  lv_obj_t *error = lv_event_get_target(e);
  lv_obj_t *entryinput = (lv_obj_t *)lv_event_get_user_data(e);
  String entrycode;

  if (code == LV_EVENT_CLICKED) {
    entrycode = String(lv_textarea_get_text(entryinput));
    if (entrycode == String(1234)) {
      lv_scr_load_anim(ui_Screen3, LV_SCR_LOAD_ANIM_NONE, 0, 0, true);
      digitalWrite(33, HIGH);  // turn the LED on
      delay(10000);
      digitalWrite(33, LOW);  // turn the LED on
    } else {
      lv_scr_load_anim(ui_Screen2, LV_SCR_LOAD_ANIM_NONE, 0, 0, true);
    }
  }
}
```

embeddedt, 2022-11-29:
If you call `delay` inside an LVGL event handler the UI will freeze until it finishes.

yaawl, 2022-11-29:
> 

@embeddedt But delay is between the led. How i solve this problem ? Thank you very much.