lvgl/lvgl #3844: Wrong FPS when running benchmark

HaoboGu, 2022-11-30:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version - I'm using LVGL v8.3
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I'm using LVGL with azure threadx together. 
When I ran the lvgl benchmark, I got 33341 FPS:
[image]

The timer tick is already set to 1ms:
[image]

### To Reproduce

### Expected behavior

The benchmark should give me correct fps

### Screenshots or video

kisvegabor, 2022-12-02:
I have 2 questions:
- approximately what is the correct FPS?
- why are these artifacts on the screen?

HaoboGu, 2022-12-05:
hello @kisvegabor
 
1. the correct FPS should be 100 ~ 120 when I ran lvgl on the bare metal chip.
2. The timing I used for the screen is not perfect, results in those black rectangles. You can just ignore it.

attachment: the correct displayed content:
[image]

kisvegabor, 2022-12-06:
How fast the MCU is and what is the resolution of the display? Maybe the 30k FPS is correct.

30,000 FPS means 0.03 ms = 30 us rendering time. It looks like a ~128x48 display -> ~6k pixels -> 30us/6k = 5 ns -> 200 MHz. So 1 instruction per pixel with a 200 MHz MCU. Does't seem unrealistic.

HaoboGu, 2022-12-06:
My MCU is running at 280MHZ, but the screen is 310 * 100. It strange because the fps monitor at the bottom-right shows 33FPS, which is not same as benchmark shows

HaoboGu, 2022-12-11:
I suspect the weird FPS is caused by some configurations of Azure RTOS. When I ran benchmark on bare mental, the FPS seems to be right(about 120FPS).

kisvegabor, 2022-12-13:
It's great that we can narrow down the issue.

I can imagine only one thing that can mess up the measurement: if  `lv_tick_inc` is not called from an interrupt to always tell the precise "time".

HaoboGu, 2022-12-13:
`lv_tick_inc` is called from a timer thread which has higher priority than main thread:

```c
// It's timer thread
void lvgl_tick_thread_entry(ULONG thread_input)
{
  while (1)
  {
    lv_tick_inc(1);
    // sleep for 1 tick(1ms)
    tx_thread_sleep(1);
  }
}
```

The main thread is:
```c
void main_thread_entry(ULONG thread_input)
{
  while (1)
  {
    tx_thread_sleep(5);
    lv_timer_handler_run_in_period(5);
  }
}
```

I fount that when I changed the time period in MAIN thread to 50, the FPS becomes 20000:
```c
// Results in 20000FPS
void main_thread_entry(ULONG thread_input)
{
  while (1)
  {
    tx_thread_sleep(50);
    lv_timer_handler_run_in_period(50);
  }
}
```
It seems that when calculating FPS, lvgl regards the ticks in us rather than ms somewhere?

kisvegabor, 2022-12-13:
Ar you sure the OS is preemptive? I.e. a higher priority task can interrupt an other task at any time?

Is there `get_milliseconds()` or similar function in the OS? If so you can use it instead of `lv_tick_inc()`. See [here](https://github.com/lvgl/lvgl/blob/bcffe08b36e9dd9fcd91aaf598f37f78486e28cc/lv_conf_template.h#L79-L85).

HaoboGu, 2022-12-14:
I tried `LV_TICK_CUSTOM`, the FPS is correct then. Maybe the timer thread was not running as I imagined. Anyway, the debug is fixed, thanks @kisvegabor

HaoboGu, 2022-12-14:
I'm still trying to figure out why using a timer thread results in the incorrect FPS

closed 2022-12-14 by HaoboGu