lvgl/lvgl #3851: FatFS doesn't work

shur1k, 2022-12-04:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

kisvegabor, 2022-12-06:
Not compiling, freezing, or does nothing?

How did you try to use it?

shur1k, 2022-12-06:
Hey @kisvegabor . I'm so sorry, I was stupid enough to forget to remove the comments mark up in the template. I've updated the issue, removing the comments :-).  It's not compiling. I tried to use it in Arduino IDE with an ESP32S3 chip configured. Everything worked fine when I tried to use an image like c array. But it uses to much space in MCU flash, so I decided to try the external image on the SD card. I changed just the image source set in my sketch `lv_img_set_src(backgroung, "S:mountain.bin");` and changed the configuration in the `lv_conf.h` to use FatFS:
```
#define LV_USE_FS_FATFS  1
#if LV_USE_FS_FATFS
    #define LV_FS_FATFS_LETTER 'S'     /*Set an upper cased letter on which the drive will accessible (e.g. 'A')*/
    #define LV_FS_FATFS_CACHE_SIZE 0    /*>0 to cache this number of bytes in lv_fs_read()*/
#endif
```
And compilation started to fail with the following error:
`/.../Arduino/libraries/lvgl/src/extra/libs/fsdrv/lv_fs_fatfs.c:228:5: error: unknown type name 'DIR';`
I've spotted, the `ff.h` include in the `.../Lvgl/src/extra/libs/fsdrv/lv_fs_fatfs.c` file. But I can't find such a header file. 
I'm using the latest stable version of lvgl.

shur1k, 2022-12-08:
Update:
I used the latest library versions for lvgl and FatFS. But it looks like lvgl doesn't see the FatFs lib. Probably, some additional configuration needed. But I can't find what else needed.

kisvegabor, 2022-12-11:
So it seems FatFS is not compiled and/or linked? Please open an FatFS file and add `#error "Build FatFS"`.

Do you see the error?

shur1k, 2022-12-11:
Nope, it's because `lv_fs_fatfs.c` doesn't see the `ff.h` header file. I've changed the include in `lv_fs_fatfs.c` from  `#include "ff.h"` to `#include "../../../../../FatFs/src/ff.h"`and compilation went fine. I still can't see the image on the screen, but at least compilation went fine.

kisvegabor, 2022-12-13:
Instead of modifying the include path (if your IDE/platform supports it) you can also add `FatFs/src` as an include path.

Was the screen working without FatFS and File system in general?

shur1k, 2022-12-13:
I'm using Arduino IDE and  `FatFs` was installed like an Arduino library, so 'FatFs/src' should be added to the path. I've also added `#include <ff.h>` in my main sketch file and there were no problems with include itself, but the mentioned error still persist until I changed the path in `lv_fs_fatfs.c `.

kisvegabor, 2022-12-15:
Oh got it, probably in this case the simplest is to modify the include pats manually.

What about the black screen issue?
> 
Instead of modifying the include path you can also add FatFs/src as an include path (if your IDE/platform supports it).
> Was the screen working without FatFS and File system in general?

shur1k, 2022-12-15:
I still can't get the image rendered on the screen. But compilation goes fine and all the other things (like lvgl widgets) work as expected.

kisvegabor, 2022-12-15:
I mean have you ever seen an image on the screen or is it related to the File system?

kisvegabor, 2022-12-18:
Okay, so the next task is to figure out where/why it crashes. What is the ESP log in restart?
Please also enable [LVGL logging](https://docs.lvgl.io/master/porting/log.html).

shur1k, 2022-12-18:
It not crashes actually, it just doesn't render the image. All the other widgets are rendered and work normally,
I tried to initialise it as following:
```c
f_mount(&fatfs, "S:", 1);
    lv_fs_fatfs_init();
    LV_IMG_DECLARE(mountain);
    lv_obj_t * backgroung = lv_img_create(lv_scr_act());
    lv_img_set_src(backgroung, "S:mountain.bin");
    lv_obj_align(backgroung, LV_ALIGN_CENTER, 0, -20);
    lv_obj_set_size(backgroung, 800, 520);
```
compilation went fine. And devices started normally, just no image was rendered.

kisvegabor, 2022-12-19:
What is the result of this?
```c
  lv_fs_file_t f;
  lv_fs_res_t res = lv_fs_open(&f, "S:mountain.bin", LV_FS_MODE_RD);
  printf("res: %d\n", res);
```

RGarrett93, 2022-12-20:
Without gate crashing this issue, I can raise a separate one but believe this is the same issue to @shur1k with LVGL FatFS, ie the image is not rendering.
It doesn't show the data element of the image but you can tell it is reading part of the file as it renders the pixel size correctly but displays a white box with `No Data`.

[image]

I've posted about it in the LVGL forum [LVGL is not loading image file with FatFS - no data](https://forum.lvgl.io/t/lvgl-is-not-loading-image-file-with-fatfs-no-data/10466) and found a slight workaround but it is not perfect: I'm wondering whether it could be a bug with how the built-in File System utilises FatFs as I've proven the file is being read via LVGL manually and loads:
`res = lv_fs_open(&f, src, LV_FS_MODE_RD);`
`res = lv_fs_seek(&f, 0, LV_FS_SEEK_END);`
`res = lv_fs_tell(&f, &fSize);`
`res = lv_fs_read(&f, buf, fSize, &rd);`

[image]

But it doesnt render the image if I call `lv_img_set_src(img, "A:image.bin")`, I have to instead call my custom function  to pre-load `lv_img_set_src(img, lv_img_dsc_load_src("A:image.bin"));`

lv_img_dsc_load_src
```
lv_img_dsc_t* lv_img_dsc_load_src(const char* src)
{
    lv_fs_res_t res = LV_FS_RES_OK;

    lv_fs_file_t f;
    uint32_t fSize = 0;
    res = lv_fs_open(&f, src, LV_FS_MODE_RD);
    if (res != LV_FS_RES_OK)
    {
        LV_LOG_ERROR("[Preload] (%s) FS open failed.", src);
        return NULL;
    }
    res = lv_fs_seek(&f, 0, LV_FS_SEEK_END);
    if (res != LV_FS_RES_OK)
    {
        LV_LOG_ERROR("[Preload] (%s) Seek end failed.", src);
        lv_fs_close(&f);
        return NULL;
    }
    res = lv_fs_tell(&f, &fSize);
    if (res != LV_FS_RES_OK)
    {
        LV_LOG_ERROR("[Preload] (%s) Get size failed.", src);
        lv_fs_close(&f);
        return NULL;
    }
    res = lv_fs_seek(&f, 0, LV_FS_SEEK_SET);
    if (res != LV_FS_RES_OK)
    {
        LV_LOG_ERROR("[Preload] (%s) Seek start failed.", src);
        lv_fs_close(&f);
        return NULL;
    }
    uint32_t rd = 0;
    uint8_t* buf = (uint8_t*)lv_mem_alloc(fSize);
    res = lv_fs_read(&f, buf, fSize, &rd);
    if (res != LV_FS_RES_OK || rd != fSize)
    {
        LV_LOG_ERROR("[Preload] (%s) Read failed.", src);
        lv_fs_close(&f);
        return NULL;
    }

    lv_img_dsc_t* img_dsc = (lv_img_dsc_t*)buf;
    img_dsc->data_size = lv_img_buf_get_img_size(img_dsc->header.w, img_dsc->header.h, img_dsc->header.cf);
    img_dsc->data = (uint8_t*)buf + sizeof(lv_img_header_t);

    lv_fs_close(&f);

    return img_dsc;
}
```

RGarrett93, 2022-12-30:
@kisvegabor any advice on whether I'm doing something with the file structure or things I could try out to try and get this to work?

kisvegabor, 2023-01-05:
Hi Garrett,

Sorry for the late reply. 

As you can see [here](https://github.com/lvgl/lvgl/blob/master/src/libs/fsdrv/lv_fs_fatfs.c#L154) LVGL's FatFS wrapper is very simple. I can't really imagine how it work in `lv_img_dsc_load_src` but not in LVGL image draw routines. 

Please enable [LVGL's logging](https://docs.lvgl.io/master/porting/log.html). Hopefully it show something useful.

RGarrett93, 2023-01-05:
@kisvegabor 

No problem, I really appreciate the reply and help.

I have already enabled [LVGL's logging](https://docs.lvgl.io/master/porting/log.html) (sorry forgot to mention that) and this is what appears with [Warn] trace set:

```flash chip initialised
f_mount successful
I was able to open 'favicon.bin' for reading!

[Warn]  (0.276, +276)    _lv_img_cache_open: Image draw cannot open the image resource  (in lv_img_cache.c line #125)
[Warn]  (0.286, +10)     lv_draw_img: Image draw error  (in lv_draw_img.c line #81)
[Warn]  (0.295, +9)      _lv_img_cache_open: Image draw cannot open the image resource  (in lv_img_cache.c line #125)
[Warn]  (0.305, +10)     lv_draw_img: Image draw error  (in lv_draw_img.c line #81)
[Warn]  (0.314, +9)      _lv_img_cache_open: Image draw cannot open the image resource  (in lv_img_cache.c line #125)
[Warn]  (0.324, +10)     lv_draw_img: Image draw error  (in lv_draw_img.c line #81)
[Warn]  (0.333, +9)      _lv_img_cache_open: Image draw cannot open the image resource  (in lv_img_cache.c line #125)
[Warn]  (0.343, +10)     lv_draw_img: Image draw error  (in lv_draw_img.c line #81)
[Warn]  (0.352, +9)      _lv_img_cache_open: Image draw cannot open the image resource  (in lv_img_cache.c line #125)
[Warn]  (0.362, +10)     lv_draw_img: Image draw error  (in lv_draw_img.c line #81)
[Warn]  (0.371, +9)      _lv_img_cache_open: Image draw cannot open the image resource  (in lv_img_cache.c line #125)
[Warn]  (0.381, +10)     lv_draw_img: Image draw error  (in lv_draw_img.c line #81)
[Warn]  (0.390, +9)      _lv_img_cache_open: Image draw cannot open the image resource  (in lv_img_cache.c line #125)
[Warn]  (0.400, +10)     lv_draw_img: Image draw error  (in lv_draw_img.c line #81)
[Warn]  (0.409, +9)      _lv_img_cache_open: Image draw cannot open the image resource  (in lv_img_cache.c line #125)
[Warn]  (0.419, +10)     lv_draw_img: Image draw error  (in lv_draw_img.c line #81)
[Warn]  (0.428, +9)      _lv_img_cache_open: Image draw cannot open the image resource  (in lv_img_cache.c line #125)
[Warn]  (0.438, +10)     lv_draw_img: Image draw error  (in lv_draw_img.c line #81)
[Warn]  (0.447, +9)      _lv_img_cache_open: Image draw cannot open the image resource  (in lv_img_cache.c line #125)
[Warn]  (0.457, +10)     lv_draw_img: Image draw error  (in lv_draw_img.c line #81)
[Warn]  (0.466, +9)      _lv_img_cache_open: Image draw cannot open the image resource  (in lv_img_cache.c line #125)
[Warn]  (0.476, +10)     lv_draw_img: Image draw error  (in lv_draw_img.c line #81)
[Warn]  (0.485, +9)      _lv_img_cache_open: Image draw cannot open the image resource  (in lv_img_cache.c line #125)
[Warn]  (0.495, +10)     lv_draw_img: Image draw error  (in lv_draw_img.c line #81)
[Warn]  (0.504, +9)      _lv_img_cache_open: Image draw cannot open the image resource  (in lv_img_cache.c line #125)
[Warn]  (0.514, +10)     lv_draw_img: Image draw error  (in lv_draw_img.c line #81)
[Warn]  (0.523, +9)      _lv_img_cache_open: Image draw cannot open the image resource  (in lv_img_cache.c line #125)
[Warn]  (0.533, +10)     lv_draw_img: Image draw error  (in lv_draw_img.c line #81)
[Warn]  (0.542, +9)      _lv_img_cache_open: Image draw cannot open the image resource  (in lv_img_cache.c line #125)
[Warn]  (0.552, +10)     lv_draw_img: Image draw error  (in lv_draw_img.c line #81)
[Warn]  (0.561, +9)      _lv_img_cache_open: Image draw cannot open the image resource  (in lv_img_cache.c line #125)
[Warn]  (0.571, +10)     lv_draw_img: Image draw error  (in lv_draw_img.c line #81)
```

I had a look at the FatFS wrapper which like you said is fairly straight forward and even tried my own custom driver but to no avail, I was looking at  [lv_img](https://github.com/lvgl/lvgl/blob/master/src/widgets/img/lv_img.c) but couldn't really see anything obvious in comparison to me doing it manually with `lv_img_dsc_load_src`

RGarrett93, 2023-01-07:
If I set the log level to `LV_LOG_LEVEL_TRACE` it crashes on me just before it loads the image using the standard method as per the docs for loading a .bin image, below is part of the full logs captured up to the point it crashes.

[image]

```
[Trace] (1.870, +9)      lv_mem_realloc: allocated at 0x20005430        (in lv_mem.c line #215)
[Trace] (1.877, +7)      lv_mem_realloc: reallocating 0x20005430 with 24 size   (in lv_mem.c line #196)
[Trace] (1.886, +9)      lv_mem_realloc: allocated at 0x20005430        (in lv_mem.c line #215)
[Trace] (1.893, +7)      lv_mem_realloc: reallocating 0x20005430 with 30 size   (in lv_mem.c line #196)
[Trace] (1.902, +9)      lv_mem_realloc: allocated at 0x20005430        (in lv_mem.c line #215)
[Trace] (1.910, +8)      lv_mem_alloc: allocating 12 bytes      (in lv_mem.c line #127)
[Trace] (1.917, +7)      lv_mem_alloc: allocated at 0x20005454  (in lv_mem.c line #160)
[Trace] (1.924, +7)      lv_mem_realloc: reallocating 0x20005454 with 18 size   (in lv_mem.c line #196)
[Trace] (1.933, +9)      lv_mem_realloc: allocated at 0x20005454        (in lv_mem.c line #215)
[Trace] (1.940, +7)      lv_mem_realloc: reallocating 0x20005454 with 24 size   (in lv_mem.c line #196)
[Trace] (1.949, +9)      lv_mem_realloc: allocated at 0x20005454        (in lv_mem.c line #215)
[Trace] (1.956, +7)      lv_mem_realloc: reallocating 0x20005454 with 30 size   (in lv_mem.c line #196)
[Trace] (1.965, +9)      lv_mem_realloc: allocated at 0x20005454        (in lv_mem.c line #215)
[Trace] (1.972, +7)      lv_mem_alloc: allocating 12 bytes      (in lv_mem.c line #127)
[Trace] (1.979, +7)      lv_mem_alloc: allocated at 0x20005478  (in lv_mem.c line #160)
[Trace] (1.987, +8)      lv_mem_realloc: reallocating 0x20005478 with 18 size   (in lv_mem.c line #196)
[Trace] (1.995, +8)      lv_mem_realloc: allocated at 0x20005478        (in lv_mem.c line #215)
[Trace] (2.003, +8)      lv_mem_realloc: reallocating 0x20005478 with 24 size   (in lv_mem.c line #196)
[Trace] (2.011, +8)      lv_mem_realloc: allocated at 0x20005478        (in lv_mem.c line #215)
[Trace] (2.019, +8)      lv_mem_realloc: reallocating 0x20005478 with 30 size   (in lv_mem.c line #196)
[Trace] (2.028, +9)      lv_mem_realloc: allocated at 0x20005478        (in lv_mem.c line #215)
[Trace] (2.035, +7)      lv_mem_realloc: reallocating 0x20005478 with 36 size   (in lv_mem.c line #196)
[Trace] (2.044, +9)      lv_mem_realloc: allocated at 0x20005478        (in lv_mem.c line #215)
[Trace] (2.051, +7)      lv_mem_realloc: reallocating 0x20005478 with 42 size   (in lv_mem.c line #196)
[Trace] (2.060, +9)      lv_mem_realloc: allocated at 0x20005478        (in lv_mem.c line #215)
[Trace] (2.067, +7)      lv_mem_alloc: allocating 12 bytes      (in lv_mem.c line #127)
[Trace] (2.074, +7)      lv_mem_alloc: allocated at 0x200054a8  (in lv_mem.c line #160)
[Trace] (2.082, +8)      lv_mem_realloc: reallocating 0x200054a8 with 18 size   (in lv_mem.c line #196)
[Trace] (2.090, +8)      lv_mem_realloc: allocated at 0x200054a8        (in lv_mem.c line #215)
[Trace] (2.098, +8)      lv_mem_realloc: reallocating 0x200054a8 with 24 size   (in lv_mem.c line #196)
[Trace] (2.106, +8)      lv_mem_realloc: allocated at 0x200054a8        (in lv_mem.c line #215)
[Trace] (2.114, +8)      lv_mem_realloc: reallocating 0x200054a8 with 30 size   (in lv_mem.c line #196)
[Trace] (2.123, +9)      lv_mem_realloc: allocated at 0x200054a8        (in lv_mem.c line #215)
[Trace] (2.130, +7)      lv_mem_realloc: reallocating 0x200054a8 with 36 size   (in lv_mem.c line #196)
[Trace] (2.139, +9)      lv_mem_realloc: allocated at 0x200054a8        (in lv_mem.c line #215)
[Trace] (2.146, +7)      lv_mem_realloc: reallocating 0x200054a8 with 42 size   (in lv_mem.c line #196)
[Trace] (2.155, +9)      lv_mem_realloc: allocated at 0x200054a8        (in lv_mem.c line #215)
[Trace] (2.162, +7)      lv_mem_realloc: reallocating 0x200054a8 with 48 size   (in lv_mem.c line #196)
[Trace] (2.171, +9)      lv_mem_realloc: allocated at 0x200054a8        (in lv_mem.c line #215)
[Trace] (2.178, +7)      lv_mem_realloc: reallocating 0x200054a8 with 54 size   (in lv_mem.c line #196)
[Trace] (2.187, +9)      lv_mem_realloc: allocated at 0x200054a8        (in lv_mem.c line #215)
[Trace] (2.195, +8)      lv_mem_alloc: allocating 12 bytes      (in lv_mem.c line #127)
[Trace] (2.202, +7)      lv_mem_alloc: allocated at 0x200054e4  (in lv_mem.c line #160)
[Trace] (2.209, +7)      lv_mem_realloc: reallocating 0x200054e4 with 18 size   (in lv_mem.c line #196)
[Trace] (2.218, +9)      lv_mem_realloc: allocated at 0x200054e4        (in lv_mem.c line #215)
[Trace] (2.225, +7)      lv_mem_realloc: reallocating 0x200054e4 with 24 size   (in lv_mem.c line #196)
[Trace] (2.234, +9)      lv_mem_realloc: allocated at 0x200054e4        (in lv_mem.c line #215)
[Trace] (2.241, +7)      lv_mem_realloc: reallocating 0x200054e4 with 30 size   (in lv_mem.c line #196)
[Trace] (2.250, +9)      lv_mem_realloc: allocated at 0x200054e4        (in lv_mem.c line #215)
[Trace] (2.257, +7)      lv_mem_alloc: allocating 12 bytes      (in lv_mem.c line #127)
[Trace] (2.264, +7)      lv_mem_alloc: allocated at 0x20005508  (in lv_mem.c line #160)
[Trace] (2.272, +8)      lv_mem_realloc: reallocating 0x20005508 with 18 size   (in lv_mem.c line #196)
[Trace] (2.280, +8)      lv_mem_realloc: allocated at 0x20005508        (in lv_mem.c line #215)
[... cut ...]
I do run out of allocated memory to buffer the image if my .bin file is over 40kB which I could increase `LV_MEM_SIZE` but not sure if that is the most efficient way of doing it compared to letting LVGL handling it all?

```
[Error] (0.287, +287)    lv_mem_realloc: couldn't allocate memory       (in lv_mem.c line #211)
[Error] (0.295, +8)      lv_mem_buf_get: Asserted at expression: buf != NULL (Out of memory, can't allocate a new buffer (increase your LV_MEM_SIZE/heap size))         (in lv_mem.c line #312)
```

kisvegabor, 2023-01-10:
It's not entirely clear from the log but it seems the the built in image decoder couldn't "accept" the image. Can you debug but happens [here](https://github.com/lvgl/lvgl/blob/release/v8.3/src/draw/lv_img_decoder.c#L331)?

RGarrett93, 2023-01-30:
I've not had a chance to debug but will hopefully look in to it this week.

closed 2023-02-22 by lvgl-bot