lvgl/lvgl #3855: lodepng_zlib_decompressv failure

Duolabs, 2022-12-06:
### Perform all steps below and tick them with [x]
- [X] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [] Update lvgl to the latest version
- [X] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
**lodepng_zlib_decompressv** returns error 83 generated by function **inflatev** when loading the a .png file from drive. 
By digging the nested code it fails in function lodepng_zlib_decompressv with error 83 while running in inflatev.
Afterward the in function zlib_decompress the *out is NULL and fails in  **zlib_decompress** and so on. 
The final result is a "no data" on the screen.
I am using version 8.3.0 of LVGL in ESP32 managed component.
The same .png image if converted in .bin (with online converter) works beautifully.

### To Reproduce
```c
lv_img_set_src(ui.mage_1, "S:/mypicture.png");
```

### Expected behavior
If lodepng is correctly loaded and initialized the png image should be loaded from external file .png

### Screenshots or video

glory-man, 2022-12-06:
https://github.com/lvgl/lvgl/blob/32948428c396edfb74bcb3189e9c7f02319899ed/src/libs/png/lodepng.c#L6749

> case 83:
            return "memory allocation failed";

You have no enough ram to decompress image

Duolabs, 2022-12-08:
It is correct. It was a RAM issue with the ESP module I had to activate the external SPI RAM to get it to work.
Issue closed.

closed 2022-12-08 by Duolabs