lvgl/lvgl #3857: Refactor CMSIS pack

bbkarim, 2022-12-07:
### Introduce the problem
We need to make the LVGL CMSIS pack compliant with Open-CMSIS-Pack rules (https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/cp_Packs.html#pack_Cclass)

- LVGL is a Variant (Bundle) of the Graphics Class but not a class.
- Graphics Classes should be implemented thru Bundles.
- All files inside the pack should be added to the PDSC file, including header files and documentation.
- Conditions need to reviewed.
- GPU support should be done via variants, and they should be exclusive.
- Port files need to be implemented as templates.
- lv_conf_cmsis.h need to reviewed for better include of the RTE_Components.h.
- Examples have to be properly implemented also inside the PDSC.
- All component's versions need to reviewed and defined.

### Examples and cases
The goal is to be able to enable the LVGL Graphics software pack and build its examples.

### Suggested solution
Initial draft is available from https://github.com/lvgl/lvgl/pull/3820

kisvegabor, 2022-12-08:
cc @GorgonMeducer

GorgonMeducer, 2022-12-10:
> LVGL is a Variant (Bundle) of the Graphics Class but not a class.
> Graphics Classes should be implemented thru Bundles.

Those are recommendations, not mandates. As LVGL cms is-pack has already been added to KEIL.com, the current structure is already approved by Arm. For backward compatibility and better identification. I think no need to change existing solutions.

GorgonMeducer, 2022-12-10:
> All files inside the pack should be added to the PDSC file, including header files and documentation.
That's one think we could improve.

GorgonMeducer, 2022-12-10:
> Conditions need to review.

We need a dedicated issue for this.

GorgonMeducer, 2022-12-10:
> GPU support should be done via variants, and they should be exclusive.

Please create a dedicated PR for just this.

GorgonMeducer, 2022-12-10:
> Port files need to be implemented as templates.

Please create a dedicated PR for just this.

GorgonMeducer, 2022-12-10:
> lv_conf_cmsis.h need to reviewed for better include of the RTE_Components.h.

We need dedicated issues to discuss this, not here. Please identify the problem you observed. For each problem, please create a dedicated issue.

GorgonMeducer, 2022-12-10:
> Examples have to be properly implemented also inside the PDSC.

Good idea. Please create a dedicated PR for just this.

GorgonMeducer, 2022-12-10:
> All component's versions need to reviewed and defined.

Sound like a big project, I am just not sure where should we identify those versions.

bbkarim, 2022-12-11:
@GorgonMeducer thank you very much for your inputs. My understanding is that LVGL is not class to add it at the same level with other real classes like Graphics, Network and Device. It was just confusing for me, but if you say that it is something intended to be like that and you agreed on it, then it is fine unless a user will select another Graphics variant in addition to LVGL. How do you plan to manage such situation? Also other classes would need to know maybe that a 'Graphics' class is enabled in the solution so they can add some extras... Do they need to check for LVGL but not Graphics ?

GorgonMeducer, 2022-12-11:
>How do you plan to manage such situation? 

It allows people to add more than one Graphic library. I see no problem at all. 

> Do they need to check for LVGL but not Graphics ?
Usually checking the existence of the `Graphics` class has no practice meaning. Can you name any situation in that people have to check the `Graphics` and do something accordingly?

> Do they need to check for LVGL
Checking the LVGL is more meaningful than checking the Graphics.

kisvegabor, 2022-12-11:
>  For LVGL's method, I think most of the header files don't need to be added.

Only `lvgl.h` needs to be included.

> As far as I know, LVGL is open and friendly, it provides an option for customers to choose from. LVGL doesn't have the attitude that "IF YOU CHOOSE LVGL, YOU CANNOT USE OTHER GUI".

LVGL is a library, not a framework. E.g. as can use 2 FFT libraries at the same, you can use LVGL with other GUI libs too. And actually it can happen in the practice too. E.g. LVGL has a [GTK driver](https://github.com/lvgl/lv_drivers/tree/master/gtkdrv), so GTK and LVGL needs to live together.  To give a more embedded-like example: If your system has 2 displays you can use an other lib for a small monochrome display, and LVGL for a TFT.

>> My understanding is that LVGL is not class

> As you can see, different people have different understandings. My understanding is that class can be anything, and pack creators have the freedom to introduce their classes.

These are my thoughts:
- If there is no strong reason to break backward compatibility, then don't break backward-compatibility.
- If ARM and Keil say it's ok if LVGL is class, then it should be fine
- LVGL is more then a simple lib (like an FFT lib) but less than a framework (e.g. QT). I can imagine more modules,  plugins, or bindings related to LVGL.  Then we can add these in the class and consider the class as an umbrella for the "LVGL ecosystem".

GorgonMeducer, 2022-12-12:
I can give you another example of a non-standard class:

[image]

Here TencentOS Tiny is an RTOS from [Tecent](https://www.tencent.com/en-us/)

`TencentOS Tiny` is used as a class, the same as LVGL. 

PikaScript is an ultra-small Python VM. (There is no such class for script language or VM)

kisvegabor, 2022-12-12:
@bbkarim 
Just to see things clearer: am I correct to assume that the root of the "class-issue" is that you are planning to add UI libs supported by ST as CMSIS-Packs (I don't know just guessing), and you would like to see all libs in one common place? Or your view is purely technical?

bbkarim, 2022-12-27:
> @bbkarim Just to see things clearer: am I correct to assume that the root of the "class-issue" is that you are planning to add UI libs supported by ST as CMSIS-Packs (I don't know just guessing), and you would like to see all libs in one common place? Or your view is purely technical?

@kisvegabor 
My view is purely technical. Wasn't expecting a variant of 'well known class' like Graphics to be exposed as a new class by itself. But thanks to you and @GorgonMeducer I can understand more now your vision. Even if I think that the provided GTK example is not enough because it is more for another build environment for linux but not cmsis-pack (AFAIK) but the problem of having two Graphical stacks in the same application is still ambiguous for such class definition.
So I agree, hopefully you have green light to do it, so let's continue that.
Originally, I had two goals with this thread:
1. Run LVGL on STM32 NUCLEO boards equipped with basic display boards, using the X-NUCLEO-GFX01M2 and X-NUCLEO-GFX01Z1 expansion boards.
2. Achieve that using:
  - STM32CubeMX with LVGL CMSIS pack + X-CUBE-DISPLAY expansion software pack
  - Keil with LVGL + X-CUBE-DISPLAY CMSIS packs

kisvegabor, 2023-01-02:
Great that we are on the same page now :relaxed: 

Finally what issues left?

bbkarim, 2023-01-05:
> Great that we are on the same page now ☺️
> 
> Finally what issues left?

GorgonMeducer, 2023-01-06:
@bbkarim please generate PRs one by one. Thank you very much.

In my opinion, the top priority should be the support for STM32U5.

GorgonMeducer, 2023-01-06:
> arch(cmsis-pack): refactoring cmsis-pack
> * Reworked components to get more granulated sub-groups
> * Renamed LVGL-Essential into LVGL-CORE Clean-up its files
> * Add missing hearder files
> * Add more widgets and demos

For this part, please keep the original LVGL::lvgl structure and add an additional one called: LVGL::lvgl9.

This LVGL::lvgl9 will employ the new structure you proposed, and the existing users can use the old LVGL::lvgl for backward compatibility - we will mark its version as deprecated.

bbkarim, 2023-01-06:
I'm also implementing a draft script to check content of the PDSC files/defines in regards of repo files and lv_conf_cmsis.h content.
The script will of course support ignored files/defines to allow exceptions.
The proposed script will be as I said a first draft, but up to you to keep it or enhance it.

bbkarim, 2023-01-08:
> For this part, please keep the original LVGL::lvgl structure and add an additional one called: LVGL::lvgl9.
>
> This LVGL::lvgl9 will employ the new structure you proposed, and the existing users can [...]

Hello @GorgonMeducer,
This mean a complete new cmsis pack no more related to LVGL.lvgl !
I had tried this and found that all dependencies will need to be reworked to target only LVGL.lvgl9 but not LVGL.lvgl.
Also, releases section will be clean-up and we will start again from v1.0.0 ...
Can't we found a better way to ensure backward compatibility and so keep the same LVGL.lvgl cmsis pack ?
Did you tried to open an old project generated with 1.12.0 and migrate it to 1.13.0-alpha ? Did you expect a lot of broken things ?
Sorry I don't know how to test that... Should I use Keil ? which project can I try with ??

Finally I think moving from **LVGL.lvgl** to **LVGL.lvgl9** is a big decision that need more analysis...

Thanks in advance,
Karim

GorgonMeducer, 2023-01-13:
> Changes are now available from #3920
> 
> 
> 
> @GorgonMeducer, it will be nice if you explain more what I've missed from your last message about the LVGL::lvgl structure and how do you plan to fork the two pack version.
> 
> 
> 
> Thanks in advance,
> 
> Karim
> 
> 

As I mentioned before, please separate those changes into dedicated PR. 

What you have done will block this Big PR for really a long time because a lot of changes need to be discussed first. 

Many of your changes has no dependency, hence separate them is the good manners.

GorgonMeducer, 2023-01-13:
> @GorgonMeducer, it will be nice if you explain more what I've missed from your last message about the LVGL::lvgl structure and how do you plan to fork the two pack version.
> 

[image]

As shown above, in current version of pdsc file, we have a bundle called 'LVGL::lvgl'

What I suggested was that in this pdsc file: 

a. Keep the existing bundle (we will put a 'deprecated' in its description. 
b. Add a new bundle called LVGL::lvgl9

With this method, we can have backwards compatibility for existing users (they can still the old bundle, we might maintain the old bundle for half year or so. This buys people time to move to the new bundle). 

New users can use the new LVGL::lvgl9 bundle. We can have a version number starting from v1.1.0 after the corresponding PR is accepted.

GorgonMeducer, 2023-01-29:
We can close this as there are dedicated PR for topics discussed here.

closed 2023-01-31 by kisvegabor