lvgl/lvgl #3859: The transition between part_main and selected of the roller is unnatural

snoopy3377, 2022-12-09:
I used two different size fonts in roller

[image]

how can i make it more natural.Just like this.

Uploading normal video.mp4…

snoopy3377, 2022-12-09:
https://user-images.githubusercontent.com/120169591/206665513-b723c501-fe71-4ee9-b359-803db753a0ee.mp4

kisvegabor, 2022-12-11:
Please send a code snippet to reproduce the issue.

It should work as it's shown on your image.

snoopy3377, 2022-12-12:
char option[(180 - 1) * 5] = "";
    char buffer[10];
    for (uint16_t i = 1; i < 180; i++)
    {
        memset(buffer, 0, sizeof(buffer));
        sprintf(buffer, "%d\n", i);
        strcat(option, buffer);
    }
    memset(buffer, 0, sizeof(buffer));
    sprintf(buffer, "%d", 160);
    strcat(option, buffer);

    lv_obj_t *roller = lv_roller_create(lv_scr_act());
    lv_roller_set_options(roller, option, 0);
    lv_obj_set_height(roller,200);
    lv_obj_set_style_text_line_space(roller, 55, LV_PART_MAIN);
    lv_obj_set_style_text_color(roller, lv_color_black(), LV_PART_SELECTED);
    lv_obj_set_style_text_color(roller, lv_color_black(), LV_PART_MAIN);
    lv_obj_set_style_text_font(roller, &fzltc_60, LV_PART_SELECTED);
    lv_obj_set_style_text_font(roller, &fzltx_44, LV_PART_MAIN);

Yes，it does work.But LV_PART_SELECTED has bigger font sizes.When font through the edge of LV_PART_SELECTED and LV_PART_MAIN，the font will be half large and half small.I think it is unnatural.How can I solve this problem？

kisvegabor, 2022-12-13:
I think it's usually handled by zooming into the middle option. 
Do you have any other ideas?

closed 2023-01-04 by lvgl-bot