lvgl/lvgl #3860: Click on disabled lv object triggers a click event on the lv object behind

sparkles43, 2022-12-09:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version (used: 8.3.3 - not master)
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Hi, 

currently I am working on a little lvgl-based UI project. (I have to really praise lvgl for being a great UI framework, thanks so much!) In my project I want to use disabled buttons. Unfortunately, I have experienced a somewhat quirky behavior: clicks on disabled lv objects will travel through them and cause a click event in the object below.

What I did specifically is:
- Create a lv_obj screen
- Make the lv_obj clickable and implement an event handler for the screen's click event
- Add a child button to the screen
- Implement an event handler for the child's click event
- Add state LV_STATE_DISABLED to the button
- Run the simulator
- Click the button
- Event handler of the screen is executed

### To Reproduce

I have updated the button counter example to illustrate the behavior I am observing:
```python
##### startup script #####

#!/opt/bin/lv_micropython -i

import lvgl as lv
import display_driver

##### main script #####

class CounterBtn():
    def __init__(self):
        self.cnt = 0
        self.cnt2 = 0
        #
        # Create a button with a label and react on click event.
        #

        bg_obj = lv.obj(lv.scr_act())
        bg_obj.add_event_cb(self.bg_event_cb, lv.EVENT.ALL, None)  # Assign a callback to the button
        bg_lbl = lv.label(bg_obj)                                    # Add a label to the button

        btn = lv.btn(bg_obj)                               # Add a button the current screen
        btn.set_pos(10, 10)                                      # Set its position
        btn.set_size(120, 50)                                    # Set its size
        btn.align(lv.ALIGN.CENTER,0,0)
        btn.add_event_cb(self.btn_event_cb, lv.EVENT.ALL, None)  # Assign a callback to the button

        label = lv.label(btn)                                    # Add a label to the button
        label.set_text("Button")                                 # Set the labels text
        label.center()

        btn.add_state(lv.STATE.DISABLED)
        btn.has_state(lv.STATE.DISABLED)
        self.btn = btn

    def btn_event_cb(self,evt):
        code = evt.get_code()
        btn = evt.get_target()
        if code == lv.EVENT.CLICKED:
            self.cnt += 1

        # Get the first child of the button which is the label and change its text
        label = btn.get_child(0)
        label.set_text("Clicked Button: " + str(self.cnt))

    def bg_event_cb(self,evt):
        code = evt.get_code()
        btn = evt.get_target()
        if code == lv.EVENT.CLICKED:
            self.cnt2 += 1

        # Get the first child of the button which is the label and change its text
        label = btn.get_child(0)
        label.set_text("Clicked BG: " + str(self.cnt2))

counterBtn = CounterBtn()
```

I debugged my application and found that the problems stems from the "object hit testing":
- `lv_indev_search_obj(lv_disp_get_layer_top(disp), &proc->types.pointer.act_point);` (I used objects on the top layer)
  -  https://github.com/lvgl/lvgl/blob/v8.3.3/src/core/lv_indev.c#L830
- `bool hit_test_ok = lv_obj_hit_test(obj, &p_trans);`
  - https://github.com/lvgl/lvgl/blob/v8.3.3/src/core/lv_indev.c#L310
- `if(lv_obj_has_state(obj, LV_STATE_DISABLED)) return false;`
  - https://github.com/lvgl/lvgl/blob/v8.3.3/src/core/lv_obj_pos.c#L956

### Expected behavior

The high-level behavior I am expecting is that clicking a disabled button dismisses the click event and does not the background.
On the implementation level I expect the lv_indev_search_obj and the lv_obj_hit_test function to regard disabled objects and return without sending a click event.

### Screenshots or video

n/a

kisvegabor, 2022-12-11:
I agree. It'd be more intuitive if the disabled objects were absorb the clicks without firing events instead of clicking through.   

I can't fix it v8 as it might change the behavior of existing UIs. However we can update it in v9.

kisvegabor, 2022-12-11:
As a workaround your can change [this line](https://github.com/lvgl/lvgl/blob/v8.3.3/src/core/lv_indev.c#L413) to:
```c
bool dis = false;
``` 

And in button event handlers check if `lv_obj_has_state(btn, LV_STATE_DISABLED)`

sparkles43, 2022-12-12:
> I agree. It'd be more intuitive if the disabled objects were absorb the clicks without firing events instead of clicking through.
> 
> I can't fix it v8 as it might change the behavior of existing UIs. However we can update it in v9.

I figured that it will come only in v9 for exactly that reason. I am just happy its accepted for v9 😃 

> As a workaround your can change [this line](https://github.com/lvgl/lvgl/blob/v8.3.3/src/core/lv_indev.c#L413) to:
> 
> ```c
> bool dis = false;
> ```
> 
> And in button event handlers check if `lv_obj_has_state(btn, LV_STATE_DISABLED)`

Thank you for the workaround. 😄

kisvegabor, 2022-12-13:
I added it to the [RADMAP](https://github.com/lvgl/lvgl/blob/master/docs/ROADMAP.md)

C47D, 2023-07-27:
Hi @kisvegabor ,

Is there anything else apart from the workaround to be done?

kisvegabor, 2023-07-27:
I think that's it. Make the disabled widgets simple not send click, press, etc events.

C47D, 2023-08-02:
I've updated the code for v9

```c
void bg_event_cb(lv_event_t * e)
{
	lv_event_code_t code = lv_event_get_code(e);
	lv_obj_t * btn = lv_event_get_target(e);

	if (LV_EVENT_CLICKED == code) {
		cnt2 += 1U;
	}

	// Get the first child of the button which is the label and change it text
	lv_obj_t * label = lv_obj_get_child(btn, 0);
	lv_label_set_text_fmt(label, "Clicked BG: %d", cnt2);
}

void btn_event_cb(lv_event_t * e)
{
	lv_event_code_t code = lv_event_get_code(e);
	lv_obj_t * btn = lv_event_get_target(e);

	if (LV_EVENT_CLICKED == code) {
		cnt += 1U;
	}

	// Get the first child of the button which is the label and change it text
	lv_obj_t * label = lv_obj_get_child(btn, 0);
	lv_label_set_text_fmt(label, "Clicked Button: %d", cnt);
}

  lv_obj_t * bg_obj = lv_obj_create(lv_scr_act());
  lv_obj_add_event(bg_obj, bg_event_cb, LV_EVENT_ALL, NULL);
  lv_obj_t * bg_lbl = lv_label_create(bg_obj);

  lv_obj_t * btn = lv_btn_create(bg_obj);

  lv_obj_set_pos(btn, 10, 10);
  lv_obj_set_size(btn, 120, 50);
  lv_obj_align(btn, LV_ALIGN_CENTER, 0, 0);
  lv_obj_add_event(btn, btn_event_cb, LV_EVENT_ALL, NULL);

  lv_obj_t * label = lv_label_create(btn);
  lv_label_set_text(label, "Button");
  lv_obj_center(label);

  lv_obj_add_state(btn, LV_STATE_DISABLED);
  lv_obj_has_state(btn, LV_STATE_DISABLED);
```

Should we add a check for disabled objects and return early at the bottom of this snippet?
```c
static void indev_encoder_proc(lv_indev_t * i, lv_indev_data_t * data)
{
    if(data->state == LV_INDEV_STATE_PRESSED && i->wait_until_release) return;

    if(i->wait_until_release) {
        i->wait_until_release      = 0;
        i->pr_timestamp            = 0;
        i->long_pr_sent            = 0;
        i->keypad.last_state = LV_INDEV_STATE_RELEASED; /*To skip the processing of release*/
    }

    /*Save the last keys before anything else.
     *They need to be already saved if the function returns for any reason*/
    lv_indev_state_t last_state     = i->keypad.last_state;
    i->keypad.last_state = data->state;
    i->keypad.last_key   = data->key;

    lv_group_t * g = i->group;
    if(g == NULL) return;

    indev_obj_act = lv_group_get_focused(g);
    if(indev_obj_act == NULL) return;

    /*Process the steps they are valid only with released button*/
    if(data->state != LV_INDEV_STATE_RELEASED) {
        data->enc_diff = 0;
    }
```

kisvegabor, 2023-08-02:
I haven't checked what would be the best way to implement it, but an early return  for all indev types seems like a good idea.

C47D, 2023-08-03:
> I  haven't checked what would be the best way to implement it, but an early return for all indev types seems like a good idea.

Will try to implement it in the simulator, I don't know if it's possible to add a keyboard key as a button indev? to test the solution in other indevs

C47D, 2023-08-31:
I've updated my test code to master to try replicate this issue again and I'm having this error.
```console
[Error]	(0.072, +72)	 _lv_inv_area: Asserted at expression: !disp->rendering_in_progress (Invalidate area is not allowed during rendering.) 	(in lv_refr.c line #257)

```

kisvegabor, 2023-08-31:
Are you on the latest `master`?

C47D, 2023-08-31:
I noticed an error in the main.c file of the port, I've opened a issue about it.

C47D, 2023-09-07:
I've found one question, in the original code snippet the btn object is children of bg_obj, is that expected?

kisvegabor, 2023-09-08:
I think it could be a little bit simpler. All we need to do is adding an `if` before  `lv_obj_event_send()` calls. E.g. [here](https://github.com/lvgl/lvgl/blob/master/src/indev/lv_indev.c#L1168):
```c
if (!disabled) lv_obj_send_event(indev_obj_act, LV_EVENT_CLICKED, indev_act);
```

Of course, where 2-3 events are sent after each other a larger `if` can be added as well.

C47D, 2023-09-12:
If the `btn` (child of `bg_obj`) is disabled and we click on it, `indev_obj_act` points to the `bg_obj` in that particular line, which from the original discussion should not be happening, correct me if I'm wrong.

PR #4547 (2023-09-21): [lv_indev] Handle click on disabled objects
### Description of the feature or fix

Fixes #3860 

### Checkpoints
- [x] Run `code-format.py` from the scripts folder. [astyle](http://astyle.sourceforge.net/install.html) needs to be installed.
- [ ] Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed
- [ ] Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant. 
- [ ] Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- [ ] If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https:/ [... cut]
closed 2023-09-21 by kisvegabor