lvgl/lvgl #3861: Compiler error ff.h not found when FATFS is enabled in lv_conf.h

sukesh-ak, 2022-12-09:
### Environment
LVGL version: 8.3 (released branch)
Controller: ESP32-S3
Framework: ESP-IDF 5.0

### Describe the bug
When FATFS is enabled in the config, compiler error shows up `ff.h: No such file or directory`

`ff.h` is referenced here
https://github.com/lvgl/lvgl/blob/d2d886aae572a29818de322002a6393a9145ebda/src/extra/libs/fsdrv/lv_fs_fatfs.c#L12

### To Reproduce
Just enable FATFS
```
#define LV_USE_FS_FATFS 1
#if LV_USE_FS_FATFS
    #define LV_FS_FATFS_LETTER 'F'     
    #define LV_FS_FATFS_CACHE_SIZE 0   
#endif
```

### Expected behavior
Compile and fatfs should work with the drive letter provided in lv_conf.h

closed 2022-12-31 by lvgl-bot

KenVanHoeylandt, 2023-12-30:
I have the same issue. I've got an ESP32 (WROOM) with LVGL v8.3.11  via https://components.espressif.com/components/lvgl/lvgl
When I enable FatFS via `idf.py menuconfig` then the project cannot find the `ff.h` header because the library isn't available.

If I look at the root `CMakeLists.txt` file then I find this:
```c
include(${CMAKE_CURRENT_LIST_DIR}/env_support/cmake/esp.cmake)
```

`esp.cmake` Contains this:
```c
  idf_component_register(SRCS ${SOURCES} ${EXAMPLE_SOURCES} ${DEMO_SOURCES}
      INCLUDE_DIRS ${LVGL_ROOT_DIR} ${LVGL_ROOT_DIR}/src ${LVGL_ROOT_DIR}/../
                   ${LVGL_ROOT_DIR}/examples ${LVGL_ROOT_DIR}/demos
      REQUIRES esp_timer)
```

For the relevant library to be included from the `esp-idf` SDK, one has to put have `REQUIRES fatfs`.

sobieh, 2024-06-07:
Hi, is it going to be ever merged/fixed or do i have to patch lvgl component every time ?

kisvegabor, 2024-06-10:
Already fixed by https://github.com/lvgl/lvgl/pull/6204

IGR2014, 2025-02-06:
@kisvegabor happening again to me with ESP-IDF 5.4.0 and LVGL 9.2.2. What am I doing wrong?! The file @KenVanHoeylandt mentioned stil does not contain this:

```cmake
  if(CONFIG_LV_USE_FS_FATFS)
     set(REQUIRES_FATFS "fatfs")
  elif
     set(REQUIRES_FATFS "")
  endif()

...

  idf_component_register(SRCS ${SOURCES} ${EXAMPLE_SOURCES} ${DEMO_SOURCES}
      INCLUDE_DIRS ${LVGL_ROOT_DIR} ${LVGL_ROOT_DIR}/src ${LVGL_ROOT_DIR}/../
                   ${LVGL_ROOT_DIR}/examples ${LVGL_ROOT_DIR}/demos
      REQUIRES esp_timer ${REQUIRES_FATFS})
```