lvgl/lvgl #3871: Remove and replace lottie causes crash

elgerg, 2022-12-17:
### Describe the bug
I am using lottie for displaying open weathermap animations (or rather trying to).

When the payload comes through I am trying to remove the existing lottie animation and create a new one. However part way through the replacements LVGL crashes. I assume it's due to memory fragmentation or something similar.

Debug output from my print statements:

Dt::::: feel: Sat Dec 17 18:00 Cal: -0.47 Min: 2.41 Max: 2.96  Main: Rain Description: Light rain Rain: Rain: 0.38mm Wind: Wind: 8.34mph Icon: 10n
2022-12-17 15:49:31::: Total Size: 1048576, Free Size: 61424, Current 98715
Dt::::: feel: Sat Dec 17 21:00 Cal: -0.63 Min: 2.04 Max: 2.44  Main: Rain Description: Light rain Rain: Rain: 2.52mm Wind: Wind: 6.87mph Icon: 10n
2022-12-17 15:49:31::: Total Size: 1048576, Free Size: 61424, Current 98715
Dt::::: feel: Sun Dec 18 00:00 Cal: -2.19 Min: 0.83 Max: 0.83  Main: Clouds Description: Overcast clouds Rain: Rain: 0mm Wind: Wind: 5.93mph Icon: 04n
2022-12-17 15:49:31::: Total Size: 1048576, Free Size: 61424, Current 98715
Dt::::: feel: Sun Dec 18 03:00 Cal: -2.75 Min: 0.31 Max: 0.31  Main: Clouds Description: Overcast clouds Rain: Rain: 0mm Wind: Wind: 5.79mph Icon: 04n
2022-12-17 15:49:31::: Total Size: 1048576, Free Size: 61424, Current 98715
Dt::::: feel: Sun Dec 18 06:00 Cal: -2.46 Min: 0.66 Max: 0.66  Main: Clouds Description: Overcast clouds Rain: Rain: 0mm Wind: Wind: 6.11mph Icon: 04n
2022-12-17 15:49:31::: Total Size: 1048576, Free Size: 61424, Current 98715
Dt::::: feel: Sun Dec 18 09:00 Cal: -2.04 Min: 1.5 Max: 1.5  Main: Clouds Description: Overcast clouds Rain: Rain: 0mm Wind: Wind: 7.67mph Icon: 04d
2022-12-17 15:49:31::: Total Size: 1048576, Free Size: 61424, Current 98715
Dt::::: feel: Sun Dec 18 12:00 Cal: -2.64 Min: 1.7 Max: 1.7  Main: Rain Description: Light rain Rain: Rain: 0.95mm Wind: Wind: 10.58mph Icon: 10d
2022-12-17 15:49:31::: Total Size: 1048576, Free Size: 61424, Current 98715
Dt::::: feel: Sun Dec 18 15:00 Cal: -0.62 Min: 3.58 Max: 3.58  Main: Rain Description: Light rain Rain: Rain: 1.87mm Wind: Wind: 12.01mph Icon: 10d
2022-12-17 15:49:31::: Total Size: 1048576, Free Size: 61424, Current 98715
Dt::::: feel: Sun Dec 18 18:00 Cal: 1.97 Min: 5.66 Max: 5.66  Main: Clouds Description: Overcast clouds Rain: Rain: 0mm Wind: Wind: 12.17mph Icon: 04n
2022-12-17 15:49:31::: Total Size: 1048576, Free Size: 61424, Current 98715
Dt::::: feel: Sun Dec 18 21:00 Cal: 4.52 Min: 7.81 Max: 7.81  Main: Rain Description: Light rain Rain: Rain: 0.11mm Wind: Wind: 13.04mph Icon: 10n
2022-12-17 15:49:31::: Total Size: 1048576, Free Size: 61424, Current 98715
Dt::::: feel: Mon Dec 19 00:00 Cal: 5.76 Min: 8.73 Max: 8.73  Main: Rain Description: Light rain Rain: Rain: 0.52mm Wind: Wind: 12.62mph Icon: 10n
2022-12-17 15:49:31::: Total Size: 1048576, Free Size: 61424, Current 98715
Dt::::: feel: Mon Dec 19 03:00 Cal: 6.57 Min: 9.36 Max: 9.36  Main: Rain Description: Light rain Rain: Rain: 0.22mm Wind: Wind: 12.53mph Icon: 10n
2022-12-17 15:49:31::: Total Size: 1048576, Free Size: 61424, Current 98715
Dt::::: feel: Mon Dec 19 06:00 Cal: 10.43 Min: 10.81 Max: 10.81  Main: Rain Description: Light rain Rain: Rain: 0.14mm Wind: Wind: 12.48mph Icon: 10n
2022-12-17 15:49:31::: Total Size: 1048576, Free Size: 61424, Current 98715
Dt::::: feel: Mon Dec 19 09:00 Cal: 11.39 Min: 11.66 Max: 11.66  Main: Rain Description: Light rain Rain: Rain: 0.24mm Wind: Wind: 14.96mph Icon: 10d
2022-12-17 15:49:31::: Total Size: 1048576, Free Size: 61424, Current 98715
Dt::::: feel: Mon Dec 19 12:00 Cal: 11.71 Min: 12.02 Max: 12.02  Main: Rain Description: Light rain Rain: Rain: 0.18mm Wind: Wind: 16.40mph Icon: 10d
2022-12-17 15:49:31::: Total Size: 1048576, Free Size: 61424, Current 98715
Dt::::: feel: Mon Dec 19 15:00 Cal: 12.11 Min: 12.46 Max: 12.46  Main: Rain Description: Light rain Rain: Rain: 0.66mm Wind: Wind: 19.19mph Icon: 10d
2022-12-17 15:49:31::: Total Size: 1048576, Free Size: 61424, Current 98715
Dt::::: feel: Mon Dec 19 18:00 Cal: 11.51 Min: 11.79 Max: 11.79  Main: Rain Description: Light rain Rain: Rain: 2.55mm Wind: Wind: 14.32mph Icon: 10n
2022-12-17 15:49:31::: Total Size: 1048576, Free Size: 61424, Current 98715
Dt::::: feel: Mon Dec 19 21:00 Cal: 11.57 Min: 11.94 Max: 11.94  Main: Rain Description: Moderate rain Rain: Rain: 4.42mm Wind: Wind: 14.63mph Icon: 10n
2022-12-17 15:49:31::: Total Size: 1048576, Free Size: 61424, Current 98715
Dt::::: feel: Tue Dec 20 00:00 Cal: 10.9 Min: 11.43 Max: 11.43  Main: Rain Description: Light rain Rain: Rain: 1.23mm Wind: Wind: 14.03mph Icon: 10n
2022-12-17 15:49:31::: Total Size: 1048576, Free Size: 61424, Current 98715
Dt::::: feel: Tue Dec 20 03:00 Cal: 5.06 Min: 8.05 Max: 8.05  Main: Clouds Description: Broken clouds Rain: Rain: 0mm Wind: Wind: 11.70mph Icon: 04n
2022-12-17 15:49:31::: Total Size: 1048576, Free Size: 61424, Current 98715

### To Reproduce
```c
    lv_mem_monitor_t mon;
    lv_mem_monitor(&mon);
    for (int i = 0; i < json_object_array_length( weatherObj ) && i < 20; i++ )
    {
		tmpWeather = json_object_array_get_idx( weatherObj, i );

		struct json_object *dt_txt,
							*feels_like,
							*temp_min,
							*temp_max,
							*wmain,
							*description,
							*rain,
							*icon,
							*wind;
		json_object_object_get_ex(tmpWeather, "dt_txt", &dt_txt);
		json_object_object_get_ex(tmpWeather, "feels_like_raw", &feels_like);
		json_object_object_get_ex(tmpWeather, "temp_min_raw", &temp_min);
		json_object_object_get_ex(tmpWeather, "temp_max_raw", &temp_max);
		json_object_object_get_ex(tmpWeather, "main", &wmain);
		json_object_object_get_ex(tmpWeather, "description", &description);
		json_object_object_get_ex(tmpWeather, "rain", &rain);
		json_object_object_get_ex(tmpWeather, "wind", &wind);
		json_object_object_get_ex(tmpWeather, "icon", &icon);
		printf("Dt::::: feel: %s Cal: %s Min: %s Max: %s  Main: %s Description: %s Rain: %s Wind: %s Icon: %s\n", json_object_get_string(dt_txt)
											, json_object_get_string(feels_like)
											, json_object_get_string(temp_min)
											, json_object_get_string(temp_max)
											, json_object_get_string(wmain)
											, json_object_get_string(description)
											, json_object_get_string(rain)
											, json_object_get_string(wind)
											, json_object_get_string(icon)
											);

		lv_label_set_text(temps.weatherItems[i].dt, json_object_get_string(dt_txt));

		lv_obj_del(temps.weatherItems[i].icon);
		temps.weatherItems[i].icon = lv_rlottie_create_from_file(weatherPage, LOTTIE_IMAGE_SIZE, LOTTIE_IMAGE_SIZE, resolve_owm_icon_lottie( json_object_get_string(icon) ));

		int ramCurrent = (int)mon.total_size - mon.free_size;
		printDateTime();
		printf("Total Size: %d, Free Size: %d, Current %d\n", mon.total_size, mon.free_size, ramCurrent/10);

	    lv_obj_align_to(temps.weatherItems[i].icon, temps.weatherItems[i].dt, LV_ALIGN_BOTTOM_MID, 10, 90);
		//lv_img_set_src(temps.weatherItems[i].icon, (lv_obj_t*)resolve_owm_icon( json_object_get_string(icon) ));

		char strColour[7];
		getColourForTemp(strColour, json_object_get_double(feels_like));
        lv_label_set_text_fmt(temps.weatherItems[i].feels, "Feels: #%s %0.2f°c#", strColour, json_object_get_double(feels_like));

		//lv_label_set_text(temps.weatherItems[i].feels, json_object_get_string(feels_like));

		getColourForTemp(strColour, json_object_get_double(temp_min));
        lv_label_set_text_fmt(temps.weatherItems[i].min, "Min: #%s %0.2f°c#", strColour, json_object_get_double(temp_min));

		//lv_label_set_text(temps.weatherItems[i].min, json_object_get_string(temp_min));

		getColourForTemp(strColour, json_object_get_double(temp_max));
        lv_label_set_text_fmt(temps.weatherItems[i].max, "Max: #%s %0.2f°c#", strColour, json_object_get_double(temp_max));

		//lv_label_set_text(temps.weatherItems[i].max, json_object_get_string(temp_max));

		lv_label_set_text(temps.weatherItems[i].rain, json_object_get_string(rain));

		lv_label_set_text(temps.weatherItems[i].wind, json_object_get_string(wind));

		lv_label_set_text(temps.weatherItems[i].bmain, json_object_get_string(wmain));

		lv_label_set_text(temps.weatherItems[i].desc, json_object_get_string(description));

		json_object_put(dt_txt);
		json_object_put(feels_like);
		json_object_put(temp_min);
		json_object_put(temp_max);
		json_object_put(rain);
		json_object_put(wind);
		json_object_put(wmain);
		json_object_put(description);
		json_object_put(icon);

		json_object_put(tmpWeather);
	}
	json_object_put(weatherObj);

```

### Expected behavior
I was hoping that the animation would be removed, memory freed and a new animation added as expected.

### Screenshots or video
[image]

Any help would be appreciated.

Thanks!

elgerg, 2022-12-30:
Hi @kisvegabor 

Any ideas on this one?

Thanks

kisvegabor, 2023-01-04:
I simpler example would be very useful to reproduce the issue. E.g. 
```c
	lv_obj_t * icon;
        icon = lv_rlottie_create_from_file(lv_scr_act(), 100, 100, A:/lottie1.json);
	lv_obj_del(icon);
        icon = lv_rlottie_create_from_file(lv_scr_act(), 100, 100, A:/lottie2.json);
```
And please also attach the lottie files you have used.

@X-Ryl669 was working a lot on the Lottie animation so I cc him.

closed 2023-01-27 by lvgl-bot