lvgl/lvgl #3873: Improve display refresh mode

FASTSHIFT, 2022-12-19:
### Introduce the problem
- In the double buffer mode, the current [direct_mode](https://docs.lvgl.io/master/porting/display.html#direct-mode) still requires the user to be responsible for synchronizing the redrawing area of two frames, which requires the user to have a certain understanding of the dirty area management mode of lvgl, and is not conducive to the simplification of the porting layer of the display.

- The [full_refresh](https://docs.lvgl.io/master/porting/display.html#direct-mode) refresh mode takes a long time to render because the full screen is redrawn every time. The occupied buffer is the same as `direct_mode`, but the performance is not as good as `direct_mode`.

### Examples and cases

### Suggested solution
- Consider moving the dirty area to `lv_refr` synchronously, you can compare the dirty area of the two frames before and after, calculate the smallest dirty area synchronization area, and reduce copying as much as possible.

- Removing the `full_refresh` mode, because `direct_mode` has better performance and can fully achieve the desired effect of `full_refresh` (exchange framebuffer pointers in `flush_cb`), and can also simplify the logic in `lv_refr`.

kisvegabor, 2022-12-20:
I've already implemented in `arch/drivers`. See https://github.com/lvgl/lvgl/blob/arch/drivers/docs/porting/display.md#draw-buffers

We are still working on MicroPython compatibility in this new branch. Once it's fixed, it'll be merged to `master`.

kisvegabor, 2022-12-20:
It work with `LV_DISP_RENDER_MODE_DIRECT`:

> `LV_DISP_RENDER_MODE_DIRECT` The buffer(s) has to be screen sized and LVGL will render into the correct location of the buffer. This way the buffer always contain the whole image. **If two buffer are used the rendered ares are automatically copied to the other buffer after flushing. Due to this in flush_cb typically only a frame buffer address needs to be changed and always the changed areas will be redrawn.**

FASTSHIFT, 2022-12-20:
I mean if `LV_DISP_RENDER_MODE_DIRECT` can meet the needs, `LV_DISP_RENDER_MODE_FULL` can be removed.

kisvegabor, 2022-12-21:
Ah I see.

I'd keep `LV_DISP_RENDER_MODE_FULL` just because it's a very standard approach. However I agree, in most of the cases DIRECT mode with 2 buffers should do the job.

FASTSHIFT, 2022-12-22:
> Ah I see.
> 
> I'd keep `LV_DISP_RENDER_MODE_FULL` just because it's a very standard approach. However I agree, in most of the cases DIRECT mode with 2 buffers should do the job.

If `LV_DISP_RENDER_MODE_FULL` is removed, it can simplify the code in lv_refr, because there is currently a lot of special processing flow for this mode in lv_refr.

kisvegabor, 2023-01-03:
I can't decide now if we can safely remove `LV_DISP_RENDER_MODE_FULL`. Let's get back to it later when we see more about the updates.

FASTSHIFT, 2023-01-03:
OK

closed 2023-01-03 by FASTSHIFT