lvgl/lvgl #3878: fps drop problem of titleview sliding

espressif2022, 2022-12-21:
I notice a previous issue [‘Snapshot widgets to buffer them for drawing‘](https://github.com/lvgl/lvgl/issues/3269)

Is this method applicable to the scene where the entire page is moved as title_view，If I'm willing to consume a little more memory？
Currently the frame rate of my UI drops very quickly during the move（33fps ->10fps），do you have any good suggestions to make title_view switching smoother?

kisvegabor, 2022-12-21:
Do you know why is it slow?
1. The draw buffers are in a slow memory so it takes a lot of time to fill the memory?
2. The screen is to complex and slow to draw?
3. Is it slow to send the data to the display controller?
4. The CPU is simply too slow even for a simple UI?
5. Anything else?

espressif2022, 2022-12-22:
> 

**1：The draw buffers are in a slow memory so it takes a lot of time to fill the memory?**
PSRAM: 120M CLK, 8Line (_Framebuf malloc here_)
SRAM: 512K
Memory speed is not too bad.

**2：The screen is to complex and slow to draw?**
The interface should not be very complicated, there are 4 to 5 blocks of different colors, and some simple elements.

**3：Is it slow to send the data to the display controller?**
pixels 480*480,RGB565
Lines:16
PCLK:18M 
one fame:(480*480*2*8)/16/18M == 12.8ms
I think the interface rate is enough.

**4：The CPU is simply too slow even for a simple UI?**
Core: Xtensa® single-dual 32-bit LX7 CPU, frequency up to 240MHz

I think it's because of the CPU's computing performance limitation, 
the screen is always redrawing all the page when sliding, and DiretMode is enabled.

Is there some existing way or suggestions to reduce this rapid frame rate drop by reducing this redrawing?

**5：Anything else?**
[image]

FASTSHIFT, 2022-12-22:
Have you tried measuring the actual speed of the PSRAM? You can use this program to measure: https://github.com/FASTSHIFT/MemorySpeed/blob/main/MemorySpeed.c

espressif2022, 2022-12-26:
> Have you tried measuring the actual speed of the PSRAM? You can use this program to measure: https://github.com/FASTSHIFT/MemorySpeed/blob/main/MemorySpeed.c

Hello, later for this test, the result is as follows (size: 512kb):
`memcpy(): rate = 29.96 MB/s [cost: 3338 ms]`
`memset(): rate = 46.25 MB/s [cost: 2162 ms]`

what do you think of this speed, it's normal or a little low?

espressif2022, 2022-12-26:
> You can try:
> 
> 1. Set the draw buffer in SRAM (because the performance of SRAM is better than PSRAM), use the double buffer mode and set the buffer as large as possible, and use DMA to copy the draw buffer to the framebuffer
> 2. Set the code optimization level of the compiler to the highest
> 3. Disable all LV_ASSERT_XXX
> 4. Do not scale images and widgets
> 5. Increase the FLASH clock frequency

> * Set the draw buffer in SRAM (because the performance of SRAM is better than PSRAM), use the double buffer mode and set the buffer as large as possible, and use DMA to copy the draw buffer to the framebuffer
> * Set the code optimization level of the compiler to the highest
> * Disable all LV_ASSERT_XXX
> * Do not scale images and widgets
> * Increase the FLASH clock frequency

- **Set the draw buffer in SRAM (because the performance of SRAM is better than PSRAM), use the double buffer mode and set the buffer as large as possible, and use DMA to copy the draw buffer to the framebuffer**
    SRAM is 512Kb total. It's RGB panel(480*480*2),not enough.
 
- **Set the code optimization level of the compiler to the highest**
- **Disable all LV_ASSERT_XXX**
    I feel that it is because the CPU calculation is too slow when sliding. 
    I don’t know if LVGL has a mechanism to put the drawn picture into the cache to reduce the next rendering work.

- **Do not scale images and widgets**
    Without this operation, there are only three titleview slides

- **Increase the FLASH clock frequency**
    Highest speed has been set, 120M;

FASTSHIFT, 2022-12-26:
> what do you think of this speed, it's normal or a little low?

a little slow.

> SRAM is 512Kb total. It's RGB panel(480 * 480 * 2),not enough.

lvgl supports non-full-screen draw buffer. What I mean is that lvgl does not draw directly on the framebuffer, because PSRAM is slow, and lvgl needs to read and write frequently during the drawing process. You can try to allocate two quarter-screen buffers on SRAM, call DMA in flush_cb to copy the drawn buffer to the framebuffer of the corresponding area, and at the same time, lvgl can continue on another buffer during DMA copying Draw without waiting for DMA to complete.

> I don’t know if LVGL has a mechanism to put the drawn picture into the cache to reduce the next rendering work.

LVGL currently does not support buffering textures after drawing to speed up scrolling, because the RAM overhead is relatively large. But will support in the future.

Can you try to test the lv_demo_benchmark score? I use a 320*240 screen on the ESP32S3, and the highest score measured is 160FPS.

espressif2022, 2022-12-26:
you try the example on the esp-box,  SPI interface? 
what is your spi clock freq setting?

FASTSHIFT, 2022-12-26:
> you try the example on the esp-box, SPI interface? what is your spi clock freq setting?

I removed the SPI sending code, 160FPS is the result of pure software rendering. If you don't consider the overhead of sending data to the screen, what is your pure software rendering score?

espressif2022, 2022-12-28:
> > you try the example on the esp-box, SPI interface? what is your spi clock freq setting?
> 
> I removed the SPI sending code, 160FPS is the result of pure software rendering. If you don't consider the overhead of sending data to the screen, what is your pure software rendering score?

I did some tests and found that drawing on SRAM did not improve the fps. 
If double framebuf is enabled, fps can be clearly improved.

I ported the same GUI to an RGB screen, and the double framebuf cannot effectively improve the frame rate. 
After removing the RGB sending code, do draw action only, fps will change. 
It can be seen that the time here is mainly spent on the transmission process.

[image]

espressif2022, 2022-12-30:
> > you try the example on the esp-box, SPI interface? what is your spi clock freq setting?
> 
> I removed the SPI sending code, 160FPS is the result of pure software rendering. If you don't consider the overhead of sending data to the screen, what is your pure software rendering score?

It is because in direct mode, Framebuf copying each other takes time, I made some judgment optimizations.
But the frame rate of CPU rendering, I have to spend some effort to look again.

closed 2022-12-30 by espressif2022