lvgl/lvgl #3890: how to add new symbol ↻ in font

bacshoreline, 2022-12-28:
Hi all,
I want to use symbol ↻ for my font "HelveticaLTPro" which i generated from HelveticaLTPro-Roman.ttf and HelveticaMonospacedPro-Rg.ttf files.Unfortunately i am not able to add this symbol in my font style.Can someone suggest how to add support for this symbol in font which i am using or i need to use another font for this?
TIA

bacshoreline, 2023-01-03:
Thanks but unfortunately its still  not working.I am not able to change the colour of inbuilt  symbols yet or its not possible?My screen background colour is black and i want to symbol in white colour.Instead of symbol,if i use string,it is working.But when i use symbol,its not showing up.
Please see my code and correct me if i am wrong.

    static lv_obj_t* repeat_schedule_symbol;
    repeat_schedule_symbol = lv_label_create(lv_scr_act(), NULL);
	lv_label_set_align(repeat_schedule_symbol, LV_LABEL_ALIGN_CENTER);       /*Center aligned lines*/
	lv_obj_set_style_local_text_color(repeat_schedule_symbol, LV_LABEL_PART_MAIN, LV_STATE_DEFAULT, LV_COLOR_WHITE);
	lv_obj_set_width(repeat_schedule_symbol, 100);
	lv_obj_set_height(repeat_schedule_symbol, 100);
	lv_obj_align(repeat_schedule_symbol, NULL, LV_LABEL_ALIGN_CENTER, 40, 40);
	// lv_obj_add_style(repeat_schedule_symbol, 0, &label_style);
	lv_label_set_text(repeat_schedule_symbol, **LV_SYMBOL_LOOP**);

kisvegabor, 2023-01-05:
Where did you set the font of the label (the font which contains the repeat symbol)? A line link this seems missing:
```
lv_obj_set_style_local_text_font(repeat_schedule_symbol, LV_LABEL_PART_MAIN, LV_STATE_DEFAULT, &my_font);
```

closed 2023-01-28 by lvgl-bot