lvgl/lvgl #3892: LVGL 8.0.2 -> 8.3.3 Performance degredation for redraw

andrewrt, 2022-12-29:
### Perform all steps below and tick them with [x]
- [x ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x] Update lvgl to the latest version
- [  ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Observing 2x slowdown in full screen redraw when updating LVGL 8.0.2 to 8.3.3.
No changes to my application code / lvgl_conf.h, but full screen redraw (320x240) is visibly ~2x slower.

Using double buffering, with LV_HOR_RES_MAX set to 320:
```c
static lv_color_t buf_1[LV_HOR_RES_MAX * 40];
static lv_color_t buf_2[LV_HOR_RES_MAX * 40];
static uint8_t swap_buf[LV_HOR_RES_MAX * 2 ];
``` 

I would start an open discussion in the LVGL forums, but the site seems to be having an issue w/ my IP address - emailed the admins directly there.

### To Reproduce

Updating from LVGL8.0.2 -> 8.3.3

### Expected behavior
Identical performance

### Screenshots or video
will try and upload if that helps

kisvegabor, 2023-01-02:
Hi,

For this kind of issues GitHub is better than the Forum, but I'll help to find out the Forum issue too.

- What hardware do you use?
- Can you use a profiler? 
- Is there anything special about the UI?
   - large shadows?
   - many widgets?
   - lot of opacity?
   - transparent screen?  
- What `swap_buf` is?

andrewrt, 2023-01-06:
Thanks for the response - sorry for my delay in getting back on this, I got back to the forum and had posted in there, including more details (https://forum.lvgl.io/t/lvgl-8-0-2-8-3-3-performance-degredation-for-redraw/10601) and forgot to check back here.

- HW: NXP RT1052, but not using the NXP PXP pipeline or hardware specific acceleration
- memory restrictions are making profiling a challenge...
- the UI has 13 screens,
-   not doing any shadows
-   i create these screens @ startup and then load a screen with something along the lines of:
```c

void show_screen(ui_screen_t *screen){
	if (current_screen != NULL){
		last_screen = current_screen;
		if (last_screen->screen_hidden != NULL){
			last_screen->screen_hidden();  //individual screen specific "cleanup" when navigating away
		}
	  if (last_screen->status_bar.ble_icon_task != NULL){
			lv_timer_del(last_screen->status_bar.ble_icon_task);
			last_screen->status_bar.ble_icon_task = NULL;
		}
	}
	current_screen = screen;
	lv_scr_load(screen->screen);
	refresh_status_bar(screen);
	screen->screen_displayed();
}```
where `screen->screen` is a plain lv_obj_t pointer.
`swap_buf` is what i'm using to rotate the screen 180˚ currently. As noted, was doing the same thing w/ LVGL 8.0.2 too. 

Specific performance impact mentioned in the forums, is: refresh display goes from 0.15s -> 0.4s.

andrewrt, 2023-01-06:
One other piece of info I just noticed with the 8.0.2->8.3.3 update: 
All my pngs and regular widgets work / look just fine as expected (aside from the refresh speed), but I have 1 screen where I display a preview of an camera capture (read from a buffer from local memory) via the following code:
```c
  lv_obj_clean(img_container);
  img_canvas = lv_canvas_create(img_container);
  lv_canvas_set_buffer(img_canvas, (void*) &(lcd_frame_buf[0][0]), LV_HOR_RES_MAX, LV_VER_RES_MAX, LV_IMG_CF_TRUE_COLOR);
  lv_obj_set_size(img_canvas, LV_HOR_RES_MAX, LV_VER_RES_MAX);
  lv_obj_align(img_canvas, LV_ALIGN_CENTER, 0, 0);
```

In 8.0.2 is looks as expected:
[image]

in 8.3.3 is looks like it took the last buf and just repeated it across the image:
[image]

kisvegabor, 2023-01-09:
> swap_buf is what i'm using to rotate the screen 180˚ currently. As noted, was doing the same thing w/ LVGL 8.0.2 too.

FYI, LVGL has a built in feature for it. See [here](https://docs.lvgl.io/master/porting/display.html#rotation). The size of the buffer can be configures with `LV_DISP_ROT_MAX_BUF` in `lv_conf.h`.

>  lv_canvas_set_buffer(img_canvas, (void*) &(lcd_frame_buf[0][0]), LV_HOR_RES_MAX, LV_VER_RES_MAX, LV_IMG_CF_TRUE_COLOR);
  lv_obj_set_size(img_canvas, LV_HOR_RES_MAX, LV_VER_RES_MAX);

Are you sure `LV_HOR/VER_RES_MAX` has correct value in v8.3.3 too?
`lcd_frame_buf[0][0]` looks a little bit suspicious. You shouldn't use the same buffer the LVGL drawing and camera buffer else things will get messy.  

What is the value of `LV_COLOR_SCREEN_TRANSP` and `disp_drv.scr_transp`?

andrewrt, 2023-01-09:
>  FYI, LVGL has a built in feature for it.
I think I wrote that a # of versions ago when that feature wasn't there. That sounds very helpful. 

> Are you sure LV_HOR/VER_RES_MAX has correct value in v8.3.3 too?
Yes, as noted, I didn't change this code at all between the LVGL update, so I have a 320 x 240 display. So the unchanged lv_conf.h includes: 
```c
#define LV_HOR_RES_MAX          (320)
#define LV_VER_RES_MAX          (240)
```

`lcd_frame_buf[0][0]` is a dedicated low res version of the camera image, so it is not the camera buffer, but rather a dedicated buffer that is a downsample of the camera buffer that I only populate when needed for LCD display. 
The full resolution camera buffer I have verified to have the correct image, and if i merely revert to 8.0.2, the LCD displays the picture as noted in the previous post.

andrewrt, 2023-01-09:
I just realized, I actaully was using hardware accel in LVGL, in lvgl_conf.h: 
```c

/*Use NXP's PXP GPU iMX RTxxx platforms*/
#define LV_USE_GPU_NXP_PXP      1

```
The reason I skipped from 8.0.3 to 8.3.3 was because of NXP PXP not being available for 8.1.x / 8.2.x....

So the slow down and break for image preview may be related to something specific to how the PXP use change for LVGL 8.3? 

Also, further information, the `swap_buf` I was using was also to swap bytes manually as I transferred data over SPI. 
I just tried to update to use LVGL's LV_COLOR_16_SWAP and that isn't supported with LVLG 8.3.3 when PXP is enabled.

andrewrt, 2023-01-10:
Test my screen refresh with on an "info screen" with 6 lines of text, 2 buttons, a systray at top w/ battery icon, and 2 image button icons at bottom:
LVGL 8.0.2 w/ NXP PXP enabled: 0.19s
LVGL 8.0.2 w/ NXP PXP disabled: 0.20s
LVGL 8.3.3 w/ NXP PXP enabled: 0.33s
LVGL 8.3.3 w/ NXP PXP disabled: 0.28s

This is all with optimization turned off. 
NXP PXP doesn't seem to be the main reason for performance slowdown. 
I suppose the performance difference doesn't seem huge number-wise,  but it's visually noticeable. 

Moving forward I turned off NXP PXP to take advantage of the byte swap available in LVGL fwiw. 
With optimization turned on, LVGL looks OK.

kisvegabor, 2023-01-12:
It's hard to tell what could be the problem.

> memory restrictions are making profiling a challenge...

Does it need extra memory? What I mean is using the tools in the Analytics menu:
[image]

andrewrt, 2023-01-21:
> What I mean is using the tools

Sorry for the delayed response, I've had time to try and make progress, but unfortunately time "spent" != progress, as our board doesn't have the ARM TRACE SWO pin available for debug - it's routed to another pin needed for comms. 

Because I'll forget it in the future, here's the docs for SWO Debug that I can't use: 
https://www.nxp.com/docs/en/application-note/AN13234.pdf
Section 3.3 seems to tell us exactly what to do. Even when following this - SWO profile features don't enable for me.
Also:
https://www.nxp.com/docs/en/quick-reference-guide/MCUXpresso_IDE_SWO_Trace.pdf
Section12 - Appendix F 

I'm guessing that you've not seen the same performance degredation in your testing... have you managed to do any testing where data goes thru SPI via the inefficient bit banging?.. (tho in this case I'm using the NXP's `LPSPI_MasterTransferNonBlocking`)

andrewrt, 2023-01-21:
a little more useful information - i went thru and optimized LVGL 1 section @ a time. 
if i optimize everything but the misc folder, I don't see much speed up - still see the new slower redraw performance, BUT if i set /misc to -o1 along with the /core folder, then speed drops significantly. 

One more piece of info that's outside of standard LVGL: 
in both LVGL 8.0.2 and LVGL 8.3.3, I moved the work_mem_int from RAM to RAM4: 
```c
// up in the include section, include CR section macros:
#include <cr_section_macros.h>

// then...:
void lv_mem_init(void)
{
#if LV_MEM_CUSTOM == 0

#if LV_MEM_ADR == 0
#ifdef LV_MEM_POOL_ALLOC
    tlsf = lv_tlsf_create_with_pool((void *)LV_MEM_POOL_ALLOC(LV_MEM_SIZE), LV_MEM_SIZE);
#else
    /*Allocate a large array to store the dynamically allocated data*/
    __NOINIT(RAM4) static LV_ATTRIBUTE_LARGE_RAM_ARRAY MEM_UNIT work_mem_int[LV_MEM_SIZE / sizeof(MEM_UNIT)];
```

kisvegabor, 2023-01-23:
I've just checked the history and can't see anything relevant in lv_mem.* or lv_tlsf.* since v8.0.2.

Usually I test LVGL's performance with O3 optimization and not really care about the non-optimized performance. Anyway, if adding -O1 to tlsf makes v8.3.3 even faster then v8.0.2 than we can be happy I guess :slightly_smiling_face:

Just to be sure: All `LV_USE_ASSERT_...`s are disabled in `lv_conf.h` right?

andrewrt, 2023-01-24:
> LV_USE_ASSERT

no, they're not disabled... I never even thought about those - aren't they good to have enabled in case something breaks (eg - using an un-init'ed style, etc?)

[image]

andrewrt, 2023-01-24:
yes, removing the mem / obj asserts sped lvgl up significantly - thanks!

kisvegabor, 2023-01-25:
Great! :slightly_smiling_face: