lvgl/lvgl #3905: LV_IMG_SIZE_MODE_REAL excessive invalidation

kjbracey, 2023-01-04:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

When a zoomed image widget using `LV_IMG_SIZE_MODE_REAL` is invalidated, it refreshes too great an area, as it unnecessarily transforms the object size.

### To Reproduce

```c
    #define LV_USE_REFR_DEBUG 1
    
    imageObj = lv_img_create(lv_scr_act());
    imageDesc.data = (const uint8_t *)lvgl_image;
    imageDesc.data_size = sizeof lvgl_image;
    imageDesc.header.always_zero = 0;
    imageDesc.header.reserved = 0;
    imageDesc.header.cf = LV_IMG_CF_TRUE_COLOR;
    imageDesc.header.w = 224;
    imageDesc.header.h = 224;
    lv_obj_align(imageObj, LV_ALIGN_TOP_MID, 0, 16);
    lv_img_set_size_mode(imageObj, LV_IMG_SIZE_MODE_REAL);
    lv_img_set_zoom(imageObj, 2*256);
    lv_img_set_antialias(imageObj, false);
    lv_img_set_src(imageObj, &imageDesc); // object auto-sizes to 452x452, incorporating 2-pixel margin from size transformation
    ...
    while(1) {
        lv_obj_invalidate(imageObj);
        lv_timer_handler();
    }
```

### Expected behavior

With the above 2x zoomed image, a far greater area of screen can be seen to being refreshed. Only the image should be refreshed.

The issue appears to be solved by adding an extra test for `img->obj_size_mode == LV_IMG_SIZE_MODE_VIRTUAL` to the following code:

https://github.com/lvgl/lvgl/blob/26767e7f64ba1fc7d705ef859ce8a063ff184d3a/src/widgets/img/lv_img.c#L467-L476

(Without this change, `s` ends up as 343, so invalidating that much extra around the image - not quite sure how it gets that big, even with the transformation).

But I'm wondering if any similar changes need to be made elsewhere in the file.

### Screenshots or video

None at present.

kisvegabor, 2023-01-06:
Hi,

Could you send a PR?

kjbracey, 2023-01-10:
I'd prefer someone more familiar with the layout system has a closer look to review the mode as a whole - I'm a bit nervous that the `REAL` mode is not fully worked through - I can see other things that don't look right.

(For now I've switched my app code back to using `VIRTUAL` mode, achieving layout by centering the image object in a holder object.)

kisvegabor, 2023-01-17:
Sorry for the late reply. In v9 I will rework this part to be more versatile. So unless it's not critical I wouldn't touch it, just to overwrite it a little bit later.

closed 2023-02-08 by lvgl-bot