lvgl/lvgl #3908: Slow Drawing using RA8875 and ESP32C3-12F

osugiw, 2023-01-09:
### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

kisvegabor, 2023-01-10:
It seems like the interface that send the pixels to the display (in `tft.drawPixels`) is slow. What is the clock speed of the SPI bus?

osugiw, 2023-01-10:
> It seems like the interface that send the pixels to the display (in `tft.drawPixels`) is slow. What is the clock speed of the SPI bus?

I already tried to increase from 4 MHz to 25 MHz the behavior still the same, more than 25 MHz the screen become blank. Is it possible using SPI DMA rather than conventional SPI?

osugiw, 2023-01-10:
> It seems like the interface that send the pixels to the display (in `tft.drawPixels`) is slow. What is the clock speed of the SPI bus?

Below is my code for driving RA8875 and I only permit the authorization of the file using your email sir:
https://drive.google.com/drive/folders/1299f0fjN2eQ-LwqEI7n1qsleCo3XnjVi?usp=sharing

Thanks!

kisvegabor, 2023-01-10:
> I already tried to increase from 4 MHz to 25 MHz the behavior still the same, more than 25 MHz the screen become blank. Is it possible using SPI DMA rather than conventional SPI?

It's very strange. Please try it with an empty screen to be sure rendring will be very fast. 

Can you use an oscilloscope the measure the clock speed on the SPI bus?

osugiw, 2023-01-11:
> It's very strange. Please try it with an empty screen to be sure rendring will be very fast.
>
> Can you use an oscilloscope the measure the clock speed on the SPI bus?

I'll try your solution, then will be back soon.

osugiw, 2023-01-11:
> It's very strange. Please try it with an empty screen to be sure rendring will be very fast.
>
> Can you use an oscilloscope the measure the clock speed on the SPI bus?

I have just realized after reading again the RA8875 datasheet, the SPI clock rate has maximum configuration as below:
[image]

Where my RA8875 system clock is 60 MHz (Refers to datasheet for screen resolution: 800x480)
https://www.waveshare.com/w/upload/a/a5/[RA8875-Application-Note.pdf](https://www.waveshare.com/w/upload/a/a5/RA8875-Application-Note.pdf)

Therefore, I couldn't raise the clock rate further.

osugiw, 2023-01-12:
Update: Turns out we could setting the system clock as below:
[image]

I configured the divider into 2 and the result is 240 MHz, therefore I could increase the SPI write frequency. I already tried increase until 60 MHz but no significant at all

kisvegabor, 2023-01-13:
Just to be sure, could check with an oscilloscope or logic analyzer that the clock speed is really what you have calculated?

osugiw, 2023-01-13:
> Just to be sure, could check with an oscilloscope or logic analyzer that the SPI clock speed is really what you have calculated?

Yes, I already checked with oscilloscope on the SCLK pin. It generates frequency up to 60MHz after display initialization.

kisvegabor, 2023-01-16:
Just a quick calculation: 800x480x16bit/60Mhz = ~100ms -> 10 FPS, If you don't use SPI DMA, let's rendering  also takes 100 ms which is 200 ms in total -> 5FPS.

However what you see is < 1 FPS. Can you send a image abut the oscilloscope's screen when a empty screen is loading? Please include the whole ~1 sec process.

osugiw, 2023-01-17:
> Just a quick calculation: 800x480x16bit/60Mhz = ~100ms -> 10 FPS, If you don't use SPI DMA, let's rendering also takes 100 ms which is 200 ms in total -> 5FPS.
>
> However what you see is < 1 FPS. Can [...]

Here is the oscilloscope measurement for the frequency of SLCK Pin. I set my frequency speed in 40 MHz.
[image]

osugiw, 2023-01-17:
> Here is the oscilloscope measurement for the frequency of SLCK Pin. I set my frequency speed in 40 MHz. [image]
 
More brief measurement:
[image]

osugiw, 2023-01-17:
Hi Sir, I found recently it could display chart properly, but when scrolling screen tearing happened.

Here is the video:
https://drive.google.com/file/d/1rPd-GPKOku3_XrYqKfrFcMKHYEXnM36z/view?usp=sharing

kisvegabor, 2023-01-18:
On this image time/div is 250 ns. The period time of 40MHz is 25ns. So it seemsone burst is 16bit. It makes sense but what is that huge gap between two bursts?
[image]

osugiw, 2023-01-18:
> On this image time/div is 250 ns. The period time of 40MHz is 25ns. So it seemsone burst is 16bit. It makes sense but what is that huge gap between two bursts? [image]
> 
> I pixel is 1 div data + 4 div gap = 1.25us, meaning ~0.5sec on a 800x480.

I have no idea, I just followed the example from lv_port_esp32. Is it possible because of delay?

kisvegabor, 2023-01-18:
I don't know, you should look into the SPI driver to see what's going on.

osugiw, 2023-01-18:
> I don't know, you should look into the SPI driver to see what's going on.

Okay, so you suspect about the slow drawing is because the GAP right? Well, I'll try to fix the culprit then.

kisvegabor, 2023-01-20:
Yes, if you can eliminate the gap it should be ~4 times faster.

osugiw, 2023-01-20:
> Yes, if you can eliminate the gap it should be ~4 times faster.

Alright, at least now I know the culprit. Thankyou sir for helping me out, if I successfully revise it I will share through this issue 👌

kisvegabor, 2023-01-23:
Great, looking forward to hearing your feedback. :slightly_smiling_face:

closed 2023-02-15 by lvgl-bot