lvgl/lvgl #3915: use about lv_obj_align_x and lv_obj_align_y

seadog-www, 2023-01-10:
### Introduce the problem

There is some code as follow in V7:
```
lv_obj_align_x(target_obj,  obj_one,  LV_ALIGN_IN_TOP_LEFT, 20);
lv_obj_align_y(target_obj,  obj_two,  LV_ALIGN_IN_TOP_LEFT, 30);
```
In V8, is there any easy way to make one obj align to different objs(such as obj_one and obj_two) in x and y directions?

### Examples and cases

### Suggested solution

pete-pjb, 2023-01-10:
Hi @seadog-www ,

Normally these type of questions should be asked on the [forum](https://forum.lvgl.io/) as you will likely get greater exposure and a quicker answer. 🙂 (You can login to it with your Github account.)

Anyway in answer to your question, I would suggest taking a look at the function:

 ```c
/**
 * Align an object to an other object.
 * @param obj       pointer to an object to align
 * @param base      pointer to an other object (if NULL `obj`s parent is used). 'obj' will be aligned to it.
 * @param align     type of alignment (see 'lv_align_t' enum)
 * @param x_ofs     x coordinate offset after alignment
 * @param y_ofs     y coordinate offset after alignment
 * @note            if the position or size of `base` changes `obj` needs to be aligned manually again
 */
void lv_obj_align_to(struct _lv_obj_t * obj, const struct _lv_obj_t * base, lv_align_t align, lv_coord_t x_ofs,
                     lv_coord_t y_ofs);
```

I hope that is helpful.

Kind Regards,

Pete.

seadog-www, 2023-01-11:
> Hi @seadog-www ,
> 
> Normally these type of questions should be asked on the [forum](https://forum.lvgl.io/) as you will likely get greater exposure and a quicker answer. 🙂 (You can login to it with your Github account.)
> 
> Anyway in answer to your question, I would suggest taking a look at the function:
> 
> ```c
> /**
> * Align an object to an other object.
> * @param obj       pointer to an object to align
> * @param base      pointer to an other object (if NULL `obj`s parent is used). 'obj' will be aligned to it.
> * @param align     type of alignment (see 'lv_align_t' enum)
> * @param x_ofs     x coordinate offset after alignment
> * @param y_ofs     y coordinate offset after alignment
> * @note            if the position or size of `base` changes `obj` needs to be aligned manually again
> */
> void lv_obj_align_to(struct _lv_obj_t * obj, const struct _lv_obj_t * base, lv_align_t align, lv_coord_t x_ofs,
>                     lv_coord_t y_ofs);
> ```
> 
> The docs for this function can be found on [this](https://docs.lvgl.io/master/overview/coords.html?highlight=lv_obj_align_to) page. I hope that is helpful.
> 
> Kind Regards,
> 
> Pete.

Thank you for your reply，the function you mentioned may be able to achieve my need. However, it need to calculate the relative position of base and parent of obj, which is not more simple than V7.

pete-pjb, 2023-01-11:
Hi @seadog-www ,

Thank you for opening your question on the forum, hopefully someone will respond there to help, in the mean-time I will close this issue.

Kind Regards,

Pete

closed 2023-01-11 by pete-pjb