lvgl/lvgl #3921: How to speed up drawing with lvgl 8?

lanistor, 2023-01-12:
### Introduce the problem

Drawing is really slow, switch page will cost `1.5s`. How to speed up?

I touch tabs on the screen to switch page, `flush_cb` will be called several times. In the below log we can see: There is nealy `1 second` between two `flush_cb`, and i don't know what lvgl is doing.

I have empty the `flush_cb`, `set_px_cb` code to avoid the influence from hardware.

```
I (58340) ####: tabs clicked time: 56730
I (58480) ####: flush_cb start time: 56870
I (58480) EDDIY: flush_cb area rect: 34 x:1 y:31 w:68 h:85;
I (58480) ####: flush_cb end time: 56870
I (58490) ####: flush_cb start time: 56880
I (58490) EDDIY: flush_cb area rect: 35 x:1 y:116 w:68 h:85;
I (58500) ####: flush_cb end time: 56890
I (58510) ####: flush_cb start time: 56900
I (58510) EDDIY: flush_cb area rect: 36 x:1 y:201 w:68 h:85;
I (58520) ####: flush_cb end time: 56910
I (58530) ####: flush_cb start time: 56920
I (58530) EDDIY: flush_cb area rect: 37 x:1 y:286 w:68 h:85;
I (58530) ####: flush_cb end time: 56920
I (58540) ####: flush_cb start time: 56930
I (58540) EDDIY: flush_cb area rect: 38 x:1 y:371 w:68 h:85;
I (58550) ####: flush_cb end time: 56940
I (58560) ####: flush_cb start time: 56950
I (58560) EDDIY: flush_cb area rect: 39 x:1 y:672 w:68 h:85;
I (58570) ####: flush_cb end time: 56960
I (59920) ####: flush_cb start time: 58310
I (59920) EDDIY: flush_cb area rect: 40 x:70 y:30 w:954 h:728;
I (59920) ####: flush_cb end time: 58310
```

### Examples and cases

```c++
int display_pixel_size = display_width * display_height;

  lv_color_t* buf1 = (lv_color_t*)heap_caps_malloc(
    display_pixel_size * sizeof(lv_color_t) / 2, MALLOC_CAP_SPIRAM);
  assert(buf1 != NULL);

  static lv_color_t* buf2 = NULL;

  static lv_disp_draw_buf_t disp_buf;

  /* Initialize the working buffer depending on the selected display.
   * NOTE: buf2 == NULL when using monochrome displays. */
  lv_disp_draw_buf_init(&disp_buf, buf1, buf2, display_pixel_size);

  lv_disp_drv_init(&disp_drv);
  disp_drv.hor_res  = display_width;
  disp_drv.ver_res  = display_height;
  disp_drv.flush_cb = disp_driver_flush;

  ESP_LOGI(TAG, "address of disp_drv: %p", &disp_drv);

  /* When using a monochrome display we need to register the callbacks:
   * - rounder_cb
   * - set_px_cb */
#ifdef CONFIG_LV_TFT_DISPLAY_MONOCHROME
  disp_drv.rounder_cb = disp_driver_rounder;
  disp_drv.set_px_cb  = disp_driver_set_px;
#elif CONFIG_LV_EPAPER_EPDIY_DISPLAY_CONTROLLER
  disp_drv.set_px_cb = disp_driver_set_px;
#endif

  disp_drv.draw_buf = &disp_buf;
  lv_disp_drv_register(&disp_drv);

  /* Register an input device when enabled on the menuconfig */
#if CONFIG_LV_TOUCH_CONTROLLER != TOUCH_CONTROLLER_NONE
  lv_indev_drv_t indev_drv;
  lv_indev_drv_init(&indev_drv);
  indev_drv.read_cb = touch_driver_read;
  indev_drv.type    = LV_INDEV_TYPE_POINTER;
  lv_indev_drv_register(&indev_drv);
#endif

  /* Create and start a periodic timer interrupt to call lv_tick_inc */
  const esp_timer_create_args_t periodic_timer_args = {
    .callback = &lv_tick_task, .name = "lvgl_tick_periodic_timer"};

  ESP_ERROR_CHECK(
    esp_timer_create(&periodic_timer_args, &lvgl_tick_periodic_timer));
  ESP_ERROR_CHECK(esp_timer_start_periodic(lvgl_tick_periodic_timer,
                                           LV_TICK_PERIOD_MS * 1000));
```

### Suggested solution

I think may be too many calculations, but i don't know how to resolve it.

### Current 
- MCU: `ESP32` (8M PSRAM & 16M Flash)
- COLOR_DEPTH: `8`, also tested `1`
- Display: 1024*758
- Screen: EPaper (Because i empty the `flush_cb`, so this will not effect)

baldhead69, 2023-01-15:
These display have many pixels to ESP32 render(without gpu).

Most likely this is the problem.

robekras, 2023-01-15:
I think the right place for your question would be at the lvgl forum here: https://forum.lvgl.io/

I have not very much experience with the ESP32, but here it goes:
1) It seems you use the external SPI RAM. I don't think this is an good idea, as this kind of RAM is much slower than the internal RAM.
2) You set up the lvgl working buffer the size of the full frame / 2. I think most of the time 40 rows would be enough. So it would easily fit into internal RAM.
3) The first six areas which are transferred to display are relatively small, but according the times stamps, only two flushes are
within the same tick. The other four are starting on one tick and ends at the next tick. 
I would have expected that the start and end of flush are within the same tick.

Do you have any other task running (maybe with higher priority?) and with some busy loop (polling something)?

I don't think the timing problems are caused by lvgl. It's something around your code.

lanistor, 2023-01-18:
Thanks a lot, we testing it.

lanistor, 2023-01-18:
> I think the right place for your question would be at the lvgl forum here: https://forum.lvgl.io/
>
> I have not very much experience with the ESP32, but here it goes:
>
> 1. It seems you use the external [...]

Thanks for the advice, we changed code as you advice, the performance has been a little optimized, full render time is in range 800~1500ms, but still to slowly.

# Code
```c++
void app_main(void* arg) {
  xGuiSemaphore = xSemaphoreCreateMutex();

  int display_height = epd_rotated_display_height();

  ESP_LOGI(TAG, "epd_display_width: %d ; epd_display_height: %d ",
           display_width, display_height);

  int display_pixel_size = display_width * 40;

  lv_color_t* buf1 = (lv_color_t*)heap_caps_malloc(
    display_pixel_size * sizeof(lv_color_t) / 2, MALLOC_CAP_DEFAULT);
  assert(buf1 != NULL);

  static lv_color_t* buf2 = NULL;

  static lv_disp_draw_buf_t disp_buf;

  /* Initialize the working buffer depending on the selected display.
   * NOTE: buf2 == NULL when using monochrome displays. */
  lv_disp_draw_buf_init(&disp_buf, buf1, buf2, display_pixel_size);

  lv_disp_drv_init(&disp_drv);
  disp_drv.hor_res  = display_width;
  disp_drv.ver_res  = display_height;
  disp_drv.flush_cb = disp_driver_flush;

  ESP_LOGI(TAG, "address of disp_drv: %p", &disp_drv);

  /* When using a monochrome display we need to register the callbacks:
   * - rounder_cb
   * - set_px_cb */
#ifdef CONFIG_LV_TFT_DISPLAY_MONOCHROME
  disp_drv.rounder_cb = disp_driver_rounder;
  disp_drv.set_px_cb  = disp_driver_set_px;
#elif CONFIG_LV_EPAPER_EPDIY_DISPLAY_CONTROLLER
  disp_drv.set_px_cb = disp_driver_set_px;
#endif

  disp_drv.draw_buf = &disp_buf;
  lv_disp_drv_register(&disp_drv);

  /* Register an input device when enabled on the menuconfig */
#if CONFIG_LV_TOUCH_CONTROLLER != TOUCH_CONTROLLER_NONE
  lv_indev_drv_t indev_drv;
  lv_indev_drv_init(&indev_drv);
  indev_drv.read_cb = touch_driver_read;
  indev_drv.type    = LV_INDEV_TYPE_POINTER;
  lv_indev_drv_register(&indev_drv);
#endif

  /* Create and start a periodic timer interrupt to call lv_tick_inc */
  const esp_timer_create_args_t periodic_timer_args = {
    .callback = &lv_tick_task, .name = "lvgl_tick_periodic_timer"};

  ESP_ERROR_CHECK(
    esp_timer_create(&periodic_timer_args, &lvgl_tick_periodic_timer));
  ESP_ERROR_CHECK(esp_timer_start_periodic(lvgl_tick_periodic_timer,
                                           LV_TICK_PERIOD_MS * 1000));

[... 80 lines cut ...]
I (25991) ####: flush_cb start time: 24400
I (25991) EDDIY: flush_cb 56 x:65 y:445 w:959 h:42; use time: 0 ms;
I (25991) ####: flush_cb end time: 24400
I (26051) ####: flush_cb start time: 24460
I (26051) EDDIY: flush_cb 57 x:65 y:487 w:959 h:42; use time: 0 ms;
I (26061) ####: flush_cb end time: 24470
I (26121) ####: flush_cb start time: 24530
I (26121) EDDIY: flush_cb 58 x:65 y:529 w:959 h:42; use time: 0 ms;
I (26121) ####: flush_cb end time: 24530
I (26191) ####: flush_cb start time: 24600
I (26191) EDDIY: flush_cb 59 x:65 y:571 w:959 h:42; use time: 0 ms;
I (26191) ####: flush_cb end time: 24600
I (26261) ####: flush_cb start time: 24670
I (26261) EDDIY: flush_cb 60 x:65 y:613 w:959 h:42; use time: 0 ms;
I (26261) ####: flush_cb end time: 24670
I (26331) ####: flush_cb start time: 24740
I (26331) EDDIY: flush_cb 61 x:65 y:655 w:959 h:42; use time: 0 ms;
I (26331) ####: flush_cb end time: 24740
I (26391) ####: flush_cb start time: 24800
I (26391) EDDIY: flush_cb 62 x:65 y:697 w:959 h:42; use time: 0 ms;
I (26391) ####: flush_cb end time: 24800
I (26411) ####: flush_cb start time: 24820
I (26411) EDDIY: flush_cb 63 x:65 y:739 w:959 h:19; use time: 0 ms;
I (26411) ####: flush_cb end time: 24820
```

pete-pjb, 2023-01-18:
Hi @lanistor ,

Just a thought...

I expect the overhead of generating your logging output is also quite high relative to rendering process and will also impact your rendering performance...  If the rendering output is correct you don't need to see the logging... have you tried it without logging to see if the performance is good 'to the eye'?

Kind Regards

Pete

cc @robekras @baldhead69 @kisvegabor

baldhead69, 2023-01-18:
Yes @pete-pjb 

I had seen this and forgot to comment here.

I already did this in practice and it impacted a lot on the performance.

lanistor, 2023-01-18:
> Hi @lanistor ,
>
> Just a thought...
>
> I expect the overhead of generating your logging output is also quite high relative to rendering process and will also impact your rendering performance... If the [...]

Thanks a lot. I just tested removed most logs only keep the first and the last log, the time cost is similar:
```
I (31136) ####: epdiy_flush start time: 29550
I (31136) EDDIY: epdiy_flush 6 x:0 y:26 w:74 h:435; use time: 0 ms;
I (31136) ####: epdiy_flush end time: 29550
I (32406) ####: epdiy_flush start time: 30820
I (32406) EDDIY: epdiy_flush 7 x:65 y:739 w:959 h:19; use time: 0 ms;
I (32406) ####: epdiy_flush end time: 30820
```

pete-pjb, 2023-01-18:
Hi @lanistor ,

I would remove all logs and try also, depending on what you are accessing a flash disk or a UART or any other piece of hardware the overhead is likely to be large to log this information.  The only way you could make such logs with out causing a large impact on rendering performance would be to put the data in a small circular buffer in the fastest RAM on the system which holds a snapshot of data that can be read by another process which runs at a low priority.

Kind Regards,

Pete

lanistor, 2023-01-18:
> 

Hi @pete-pjb , Here is my code, i think this can vvoid the influence of log:
```c++
clock_t start_time = 0;
void flush_cb(lv_disp_drv_t*   drv,
                 const lv_area_t* area,
                 lv_color_t*      color_map) {
  if (area->x1 == 0 && area->y1 < 30) {
    // first `flush_cb` call
    start_time = clock();
  } else if (area->x1 == 65 && area->y1 == 739) {
    // last `flush_cb` call
    clock_t time_period = clock() - start_time;
    ESP_LOGE("####", "flush_cb time period: %ld", time_period);
  }
  lv_disp_flush_ready(drv);
}
```

lanistor, 2023-01-18:
> Hi @lanistor ,
> 
> I would remove all logs and try also, depending on what you are accessing a flash disk or a UART or any other piece of hardware the overhead is likely to be large to log this information. The only way you could make such logs with out causing a large impact on rendering performance would be to put the data in a small circular buffer in the fastest RAM on the system which holds a snapshot of data that can be read by another process which runs at a low priority.
> 
> Although you don't need to be logging this stuff if all is well.
> 
> Kind Regards,
> 
> Pete

Hi @pete-pjb  , Here is my code, can this avoid the influence of log? Beside i have set the log level to error, so seldom log output.
```c++
clock_t start_time = 0;
void flush_cb(lv_disp_drv_t*   drv,
                 const lv_area_t* area,
                 lv_color_t*      color_map) {
  if (area->x1 == 0 && area->y1 < 30) {
    // first `flush_cb` call
    start_time = clock();
  } else if (area->x1 == 65 && area->y1 == 739) {
    // last `flush_cb` call
    clock_t time_period = clock() - start_time;
    ESP_LOGE("####", "flush_cb time period: %ld", time_period);
  }
  lv_disp_flush_ready(drv);
}
```

pete-pjb, 2023-01-18:
Hi @lanistor,

I'm not sure you are understanding the significance of how much delay the logging could potentially introduce.

I don't have any experience with ESP platforms directly and things will vary depending on how the underlying drivers are written etc. but hopefully this will explain my concerns and how it can be a source of problems.  If I am telling you what you already know please just ignore me. 🙂 

Firstly the output of your string ```I (36786) ####: flush_cb end time: 35200``` for example, if your ESP UART is configured for 115200 Baud with 8 bits of data, no parity and 1 stop bit it means for each character the UART is sending 9-bits, your string is ~40 characters long so 40 * 9 = 360 bits transmitted at 115200 bits per second takes 3.125 milliseconds.  This is the time to physically send the characters.  There is also the overhead of the call to the ```sprintf()``` function used by the ```ESP_LOGE()``` macro to format the data output. If any other processes are also calling ```ESP_LOGE()``` you may have to wait for those to complete adding more uncertainty and potential delays.  Depending on the implementation of the ESP UART driver the function may or may not wait for the characters to be transmitted.  If it is an interrupt driven driver with a hardware FIFO, straight away the transmit process will kick off some interrupts which will impact on your timings measured.  There is also system latency, code latency etc. This will all vary greatly depending on the implementation of the UART driver and if other things are going on in the system.   Say you are getting a total logging induced delay of 4 milliseconds on each call to ```flush_cb()``` and your expected frame rate is 33 frames per second this would be reduced to 27 frames per second.  Some of your statistics show times much higher than this.

#### The moral of the story is I would avoid the output of ANY DATA to UARTs or slow file systems in time critical code like display rendering.  

#### Without visibility of your entire code base, this also applies to anything going on within the LVGL execution path, lots of logging activity, ```vTaskDelay()``` calls, blocking semaphores with timeouts and any code which blocks in anyway in an event handler, is going to break the system in this way.   So it is worth checking your LVGL event handlers also.

The only true test for performance is to have no data being printed or logged and the lvgl debugging disabled during normal execution of your application.

I would also consider @robekras point (I quote below) this is also a real possibility:

> Do you have any other task running (maybe with higher priority?) and with some busy loop (polling something)?

There might also be something going on internally in the ESP libraries as well as your own code.

I hope that makes sense, and is helpful. 🙂 

Kind Regards,

Pete

lanistor, 2023-01-19:
> Hi @lanistor,
>
> I'm not sure you are understanding the significance of how much delay the logging could potentially introduce.
>
> I don't have any experience with ESP platforms directly and things will [...]

Thanks for advice, we are testing about this.

lanistor, 2023-01-21:
> Hi @lanistor,
>
> I'm not sure you are understanding the significance of how much delay the logging could potentially introduce.
>
> I don't have any experience with ESP platforms directly and things will [...]

Hi, @pete-pjb , @robekras:

Now i'm very sure this is not the problem of `log`. If the time is `10ms`, maybe log effects a lot, but it's `1000ms`, log can't effect so much.

I reproduced the problem step by step by the demo `keypad_encoder`:
1. Condition 1: official code, switch tab will cost `300~400ms`.
2. Condition 2: clean the inactive tab content, switch tab will cost `1100~1300ms`
3. Condition 3: set background(white or black) to tab content, switch tab will cost `600~700ms`

So i think the problem is `lv_obj_clean`, this slow the project down, beside setting background to container will also slow project down.

The test code:
```c++
static void tab_changed(lv_event_t* e) {
  lv_obj_t* tv          = (lv_obj_t*)e->user_data;
  uint16_t  current_tab = lv_tabview_get_tab_act(tv);
  if (current_tab == 0) {
    selectors_create(t1_cont);
    lv_obj_clean(t2_cont);
  } else {
    text_input_create(t2_cont);
    lv_obj_clean(t1_cont);
  }
  LV_LOG_ERROR("#### tab_changed code: %d", current_tab);
}

void lv_demo_keypad_encoder(void) {
  g = lv_group_create();
  lv_group_set_default(g);

  lv_indev_t* cur_drv = NULL;
  for (;;) {
    cur_drv = lv_indev_get_next(cur_drv);
    if (!cur_drv) {
      break;
    }

    if (cur_drv->driver->type == LV_INDEV_TYPE_KEYPAD) {
      lv_indev_set_group(cur_drv, g);
    }

    if (cur_drv->driver->type == LV_INDEV_TYPE_ENCODER) {
      lv_indev_set_group(cur_drv, g);
    }
  }

  tv = lv_tabview_create(lv_scr_act(), LV_DIR_TOP, LV_DPI_DEF / 3);
  lv_obj_add_event_cb(lv_tabview_get_content(tv), scroll_begin_event,
                      LV_EVENT_SCROLL_BEGIN, NULL);

  lv_obj_add_event_cb(tv, tab_changed, LV_EVENT_VALUE_CHANGED, tv);

  t1 = lv_tabview_add_tab(tv, "Selectors");
  t2 = lv_tabview_add_tab(tv, "Text input");

  t1_cont = lv_obj_create(t1);
  lv_obj_set_size(t1_cont, lv_pct(100), lv_pct(100));
  //   lv_obj_set_style_bg_color(t1_cont, lv_color_white(), LV_PART_MAIN);
  //   selectors_create(t1_cont);

  t2_cont = lv_obj_create(t2);
  lv_obj_set_size(t2_cont, lv_pct(100), lv_pct(100));
  //   lv_obj_set_style_bg_color(t2_cont, lv_color_white(), LV_PART_MAIN);
  //   text_input_create(t2_cont);

  //   msgbox_create();
}
```

lanistor, 2023-01-22:
Hi, @pete-pjb , @robekras:
I think i had found the reasons, it's `set_px_cb`, it effect so much, i removed it then the time cost reduce to `400ms`.
Beside: the priority of task truly effects a lot too.

Thanks a lot.

kisvegabor, 2023-01-23:
> I think i had found the reasons, it's set_px_cb, it effect so much, i removed it then the time cost reduce to 400ms.

Oh yes, it's quite slow and recommended mainly for small displays. 

If the problem seems to be fixed, please close this issue.

weilian1977, 2023-01-27:
关注一下

lanistor, 2023-01-29:
> 

Yeah, we found the problem is `set_px_cb`, it effect so much, i removed it then the time cost reduce to `400ms`.

Beside, here is the result data we tested:
1. `set_px_cb` has a great impact on performance, which will increase the time consumption by 2.0~2.5 times
2. Task priority has a great impact on performance, set the priority as high as possible
3. The number of Buffer rows has a little effect on performance. It can balance high performance and low memory usage
4. Business logic has an impact on rendering performance, but not much
5. Memory location has little effect on performance
6. Double Buffer or single Buffer have no effect on performance at all
7. `lv_tick_task` has no effect on performance at all
8. `ESP_LOGX` has little impact on performance and can be ignored

closed 2023-01-29 by lanistor

kisvegabor, 2023-01-31:
Thanks for the summary! :+1: