lvgl/lvgl #3930: LV_PRId32 and LV_PRIu32 compile error in lv_demo_benchmark

baldhead69, 2023-01-15:
### Describe the bug

 

### To Reproduce

### Expected behavior

baldhead69, 2023-01-15:
Some suggestion.

I am using lvgl version 8.0.0 i think dev.

Thank's.

kisvegabor, 2023-01-17:
Can you update to `release/v8.3`? I believe these are already fixed there.

baldhead69, 2023-01-17:
I updated to v8.3 and worked.

First i tried to update to v9.0.0 dev, but problems with makefile apoear.

https://www.microchip.com/forums/m/tm.aspx?m=1224283&fp=1

baldhead69, 2023-01-17:
@kisvegabor 

I think that i found the problem:

'''
c:\microchip\xc32\v4.21\bin\bin\gcc\pic32mx\8.3.1\..\..\..\..\bin/pic32m-ld.exe: build/default/production/_ext/443351157/animimg001.o:(.rodata.animimg001+0x0): multiple definition of `animimg001'; build/default/production/_ext/1010940615/animimg001.o:(.rodata.animimg001+0x0): first defined here
c:\microchip\xc32\v4.21\bin\bin\gcc\pic32mx\8.3.1\..\..\..\..\bin/pic32m-ld.exe: build/default/production/_ext/443351157/animimg001.o:(.rodata.animimg001_map+0x0): multiple definition of `animimg001_map'; build/default/production/_ext/1010940615/animimg001.o:(.rodata.animimg001_map+0x0): first defined here
c:\microchip\xc32\v4.21\bin\bin\gcc\pic32mx\8.3.1\..\..\..\..\bin/pic32m-ld.exe: build/default/production/_ext/443351157/animimg002.o:(.rodata.animimg002+0x0): multiple definition of `animimg002'; build/default/production/_ext/1010940615/animimg002.o:(.rodata.animimg002+0x0): first defined here
c:\microchip\xc32\v4.21\bin\bin\gcc\pic32mx\8.3.1\..\..\..\..\bin/pic32m-ld.exe: build/default/production/_ext/443351157/animimg002.o:(.rodata.animimg002_map+0x0): multiple definition of `animimg002_map'; build/default/production/_ext/1010940615/animimg002.o:(.rodata.animimg002_map+0x0): first defined here
c:\microchip\xc32\v4.21\bin\bin\gcc\pic32mx\8.3.1\..\..\..\..\bin/pic32m-ld.exe: build/default/production/_ext/443351157/animimg003.o:(.rodata.animimg003+0x0): multiple definition of `animimg003'; build/default/production/_ext/1010940615/animimg003.o:(.rodata.animimg003+0x0): first defined here
c:\microchip\xc32\v4.21\bin\bin\gcc\pic32mx\8.3.1\..\..\..\..\bin/pic32m-ld.exe: build/default/production/_ext/443351157/animimg003.o:(.rodata.animimg003_map+0x0): multiple definition of `animimg003_map'; build/default/production/_ext/1010940615/animimg003.o:(.rodata.animimg003_map+0x0): first defined here
c:\microchip\xc32\v4.21\bin\bin\gcc\pic32mx\8.3.1\..\..\..\..\bin/pic32m-ld.exe: build/default/production/_ext/443351157/img_caret_down.o:(.rodata.img_caret_down+0x0): multiple definition of `img_caret_down'; build/default/production/_ext/1010940615/img_caret_down.o:(.rodata.img_caret_down+0x0): first defined here
c:\microchip\xc32\v4.21\bin\bin\gcc\pic32mx\8.3.1\..\..\..\..\bin/pic32m-ld.exe: build/default/production/_ext/443351157/img_caret_down.o:(.rodata.img_caret_down_map+0x0): multiple definition of `img_caret_down_map'; build/default/production/_ext/1010940615/img_caret_down.o:(.rodata.img_caret_down_map+0x0): first defined here
c:\microchip\xc32\v4.21\bin\bin\gcc\pic32mx\8.3.1\..\..\..\..\bin/pic32m-ld.exe: Link terminated due to previous error(s).
'''c

These files are with same name in two different folder:

'''
C:\Users\Rangel\Desktop\Mplab-Progs\pic32mz-teste\firmware\src\my_Files\graphic_Interface\lvgl\tests\src\test_assets\animimg001.c
C:\Users\Rangel\Desktop\Mplab-Progs\pic32mz-teste\firmware\src\my_Files\graphic_Interface\lvgl\examples\assets\animimg001.c
'''c

The same is true for other files:
* animimg001
* animimg002
* animimg003
* img_caret_down

kisvegabor, 2023-01-18:
I've pushed a fix [ad8eed9](https://github.com/lvgl/lvgl/commit/ad8eed9eb58f6d6cc50720447f05921f1bc738d0).
Could you try it out?

closed 2023-02-09 by lvgl-bot