lvgl/lvgl #3933: lv_demo_benchmark - incorrect FPS results (8.3)

tdjastrzebski, 2023-01-16:
`lv_demo_benchmark()` shows all the FPS results being either 33 or 34k.
LVGL 8.3 the latest, GCC compiler, STM32F769NI-DISCO
[image]

baldhead69, 2023-01-16:
I think it's right.

The fps is not calculate to full screen render + flush.

https://github.com/lvgl/lvgl/tree/release/v8.3/demos/benchmark

"Note that it can result in very high FPS results for simple cases. E.g. if some simple rectangles are drawn in 5 ms, the benchmark will tell it's 200 FPS. So it ignores LV_DISP_REFR_PERIOD which tells LVGL how often it should refresh the screen. In other words, the benchmark shows the FPS from the pure rendering time."

baldhead69, 2023-01-17:
The chip STM32F769NI has hardware graphics accelerators, but it looks kinda high.

"Graphics
Chrom-ART Accelerator (DMA2D), graphical hardware accelerator enabling enhanced graphical user interface
Hardware JPEG codec
LCD-TFT controller supporting up to XGA resolution
MIPI® DSI host controller supporting up to 720p 30 Hz resolution"

tdjastrzebski, 2023-01-17:
Thanks, maybe those values are correct then, but they seem to be a bit odd. Why only two values?
DMA2D is not used by LVGL, unless one explicitly enables it in **lv_conf.h** - `LV_USE_GPU_STM32_DMA2D`.
DMA2D was not well supported, so I just made a significant rework - see PR #3904

GorgonMeducer, 2023-01-17:
Those issue has been fixed in v9

kisvegabor, 2023-01-17:
Usually it happens if you don't call `lv_tick_inc()` from an interrupt. Could you check it?

kisvegabor, 2023-01-17:
Exactly this configuration causes this kind of issues. This way `lv_tick_inc()` cannot tick-ing while `lv_task_handler` is working. So anything happens in `lv_task_handler` can not know how long an operation lasted. 

On STM all you need to do is this:

https://github.com/lvgl/lv_port_stm32f769_disco/blob/c567b238d629b4dbb1164a43392db64e69e48cc9/src/stm32f7xx_it.c#L39-L46

And call `lv_task_handler()` somewhere in a lower priority interrupt or main function.

tdjastrzebski, 2023-01-17:
I see, so the problem is `lv_tick_inc()` and `lv_task_handler()` are called from within the interrupt context?
The above does not mean the same interrupt handler. Probably the same handler can be used, as long as it is a different interrupt context. Is that right?

closed 2023-01-17 by tdjastrzebski

tdjastrzebski, 2023-01-18:
Interesting that fixing this issue noticeably speeded up lv_demo_music intro rays rendering and lv_demo_widgets responsiveness.

kisvegabor, 2023-01-18:
It's because let's rendering took 40 ms in `lv_task_handler` but you only added 5 in `lv_tick_inc`. So the animations thought that on 5ms elapsed while it was 40ms.