lvgl/lvgl #3934: LVGL 7.11 Test Minimal Features fails to build

enribird, 2023-01-16:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

I ticked some steps even when they are not related to the issue (for example simulator). Sorry for that.

### Describe the bug
Version 7.11 test fails on Minimal Features test because a build error

### To Reproduce
```shell
cd lvgl/tests
./build.py
```

### Expected behavior
I expected the test to run and be successful. Instead the second round of tests (the one called All objects, minimal features) fails to build because of an error in **lv_tileview.h**

The problem is that **lv_tileview** style involving animation is not protected with macro **#if LV_USE_ANIMATION** as the other components. Neither the themes than clean their style

### Screenshots or video
Here is the unprotected use of **LV_PAGE_PART_EDGE_FLASH** on **lv_tileview.h**
[image]

That is unprotected unlike in other widgets as for example **lv_list.h** 
[image]

Also, there are several themes (lv_theme.c, lv_theme_mono.c and lv_theme_template.c ) that clean this style without checking the definition of **#if LV_USE_ANIMATION**

[image]
[image]

I have checked this is the source of the error by modifying the test script and adding animation completes the tests without build errors.

kisvegabor, 2023-01-17:
Hi,

v7.11 is no longer supported. See [here](https://docs.lvgl.io/master/intro/index.html#version-support).

Despite of that, if you can send a patch for the `release/v7` branch I can merge it.

spanish-worker, 2023-01-23:
[I sent a pull request](https://github.com/lvgl/lvgl/pull/3942).

kisvegabor, 2023-01-24:
Merged, thank you!

closed 2023-01-24 by kisvegabor