lvgl/lvgl #3935: Where is LV_COLOR_16_SWAP in 9.0.0 dev version ?

baldhead69, 2023-01-16:
Hi,

I did not find LV_COLOR_16_SWAP in lv_conf.h.

Need i set this in driver init ?
if yes, how ?

I want 16 bits color and not swapped.

Thank`s.

kisvegabor, 2023-01-17:
You can set `disp_drv.color_format = LV_COLOR_FORMAT_NATIVE_REVERSE`.

Note that `master` is still heavily under development and it's recommended to use `release/v8.3`.

closed 2023-01-17 by baldhead69

hyansuper, 2023-07-02:
Hi,
I am using GC9A01 TFT display.
Setting `LV_COLOR_FORMAT_NATIVE_REVERSE` or `LV_COLOR_FORMAT_NATIVE_ALPHA_REVERSED` has no effect.

I understand v9.0 is still in dev but willing to try because it fixed some bugs in v8.3

kisvegabor, 2023-07-03:
Hi,

It's working for me. Did you set LV_COLOR_DEPTH to 16?

yunshanghen, 2025-10-02:
It is possible to enable     lv_draw_sw_rgb565_swap(px_map, lv_area_get_size(&offset_area)); to solve problems such as color inversion.