lvgl/lvgl #3936: no member named 'clip_area'?

cuishuang413, 2023-01-17:
Hello,

I have been trying to compile the live widgt demo, but I got this error and don't know where to start. Please help!

error: 'lv_obj_draw_part_dsc_t' {aka 'struct <anonymous>'} has no member named 'clip_area'; did you mean 'draw_area'?

Thanks,

kisvegabor, 2023-01-18:
Do you use the `release/v8.3` branch? If not, please use it as `master` is under development and it might be unstable.

cuishuang413, 2023-01-24:
That fixed it. Thanks!

closed 2023-01-24 by cuishuang413

wolfgangwayland7, 2023-02-12:
For your information the "lvgl-release-v8.3" {down load as a zip only or cloned} still dose not have 
"clip_area" defined inside the {lv_obj_draw_part_dsc_t  [struct]}     of the  as of  2-11-2023

The struct is int the "lv_obj_draw.h" header inside the "lvgl-release-v8.3/src/core" folder...

I don't know which which file  user "cuishuang413" downloaded? I'd like to know.. Thanks

kisvegabor, 2023-02-14:
@wolfgangwayland7 Exactly what is not working? We have CI running on each commit testing the demos so they should work on the release//v8.3 branch.

kisvegabor, 2023-02-14:
Let's continue in #3982