lvgl/lvgl #3940: Timer creation order vs or Timer calling order

thomasm-scle, 2023-01-19:
### Perform all steps below and tick them with [x]
- [X] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [X] Update lvgl to the latest version
- [X] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Hello,
I allow myself to write to you because I have a problem in the timers calling order which is different from the order of creation (which leads to a segfault in my application).

Thanks in advance.

### To Reproduce

On WSL2, with Debian10 and X11 server (and SDL).

Explanation :
I have 2 items:
- a parent (OBJECT_A)
- a child (OBJECT_B) which is a child of OBJECT_A.

In my code, in the same function I have the following code:
- lv_obj_del_async(OBJECT_B) which creates a timer TIMER_A.
- lv_obj_del_async(OBJECT_A) which creates a timer TIMER_B.

In the '**lv_timer_handler**' function of **lv_timer.c**, the '**lv_timer_exec**' function executes **first the TIMER_B** **then the TIMER_A** (which is the **reverse order** of the creation of the timers).

In my application, this leads:
   - via the TIMER_B when releasing OBJECT_A (and therefore the child OBJECT_B)
   - via TIMER_A when OBJECT_B is freed, except that being already freed I have a segfault.

### Expected behavior
After analyzing the **LVGL v8.1** code, the '**lv_timer_create**' function creates a timer and places it at the top of the list of timers via the '**_lv_ll_ins_head**' function.
If I replace the '**_lv_ll_ins_head**' function by '**_lv_ll_ins_tail**', I have the TIMER_A which is executed first, and then the TIMER_B. In this case my application no longer has a segfault.

What do you think about the '**lv_timer_create**' function placing the new timer at the top of the list, instead of placing it next?

N.B. : The code is same in the master branch for the function '**lv_timer_create**'

kisvegabor, 2023-01-20:
Hi,

Thanks for reporting this. I see the problem and it's really counter intuitive. Ideally LVGL should be smart enough to solve this kind of issues internally because we can't expect that all the user will know these complex internal behaviors. 

Could you try if adding this
```c
    /*Delete pending lv_obj_del_async calls on obj  */
    lv_timer_t * timer = lv_timer_get_next(NULL);
    while(timer) {
    	/*If the timer will be deleted, we can't get the next timer from it later*/
    	lv_timer_t * timer_next = lv_timer_get_next(timer);

    	if(timer->timer_cb == lv_obj_del_async_cb && timer->user_data == obj) {
    		lv_timer_del(timer);
    	}
    	timer = timer_next;
    }
```

[here](https://github.com/lvgl/lvgl/blob/ac31c7c1b12cf3e437b2d04614893dd17b9a6748/src/core/lv_obj_tree.c#L380) solves the issue?

thomasm-scle, 2023-01-23:
Hello,

Glad to report the problem. The LVGL is a very good library, so if I can (a little) help so much the better ;)

I tried your code, unfortunately successful. The problem is still present (segfault of my application).
I also get a compiler warning:

_/lvgl/src/core/lv_obj_tree.c:390:25: warning: comparison of distinct pointer types lacks a cast
      if(timer->timer_cb == lv_obj_del_async_cb && timer->user_data == obj) {_

kisvegabor, 2023-01-25:
Huh, it was really more complicated then it seemed at the first look :slightly_smiling_face: 

I pushed a patch and now it's working well for me. I've tested with this code:

```c
  lv_obj_t * btn = lv_btn_create(lv_scr_act());
  lv_obj_t * label = lv_label_create(btn);

  lv_obj_del_async(label);
  lv_obj_del_async(btn);
```

commit 37bed322 (2023-01-25): fix(timer): fix working with nested lv_async_calls

fixes: #3940
closed 2023-01-25 by kisvegabor