lvgl/lvgl #3941: [lv_chart] Handling zoom factor being less than 256

C47D, 2023-01-22:
### Introduce the problem
Docs states that `factor` can't be less than 256, what should `lv_chart_set_zoom_y` and `lv_chart_set_zoom_x` functions do when trying to do `factor < 256`? Return without doing anything?

[image]

### Examples and cases
Found about it when doing the minimal test case for both functions, I was expecting not to be able to set `factor` to 128.
[image]

### Suggested solution
Check `factor` and return if it's less than 256.

kisvegabor, 2023-01-22:
I'd set to the zoom to 256 if the user tries to set < 256 value.

C47D, 2023-01-22:
> I'd set to the zoom to 256 if the user tries to set < 256 value.

Ok, I'll do that. What about `LV_ZOOM_NONE` where can I declare it? `lv_chart.h` looks ok for me.

kisvegabor, 2023-01-23:
> Ok, I'll do that. What about LV_ZOOM_NONE where can I declare it? lv_chart.h looks ok for me.

In `master` we have `LV_IMG_ZOOM_NONE` now which is not a good fit for the chart. Could you refactor `LV_IMG_ZOOM_NONE` back to `LV_ZOOM_NONE` so that we can use it for both the images and the charts. It's in `misc/lv_style.h`.

> Done it in https://github.com/lvgl/lvgl/commit/b544990b122d7c2f01d3d800ec4e5c59189c154c and https://github.com/lvgl/lvgl/commit/41eaf3dc3691ebef23945c3fc8c3e80f26534881.

It seems something is wrong with these commits.

[image]

C47D, 2023-01-23:
> Could you refactor LV_IMG_ZOOM_NONE back to LV_ZOOM_NONE so that we can use it for both the images and the charts. It's in misc/lv_style.h.

Sure, I can update it in a separated pull request? I don't know how long it will take me to finish the chart tests

> It seems something is wrong with these commits.

Maybe github are searching in this repo and not mine, here's the link to the branch where I'm doing the updates https://github.com/C47D/lvgl/tree/test/chart

kisvegabor, 2023-01-24:
> Sure, I can update it in a separated pull request? I don't know how long it will take me to finish the chart tests

A separate PR would be welcome. Thank you!

kisvegabor, 2023-02-09:
Not stale

kisvegabor, 2023-02-27:
Fixed in #4005

closed 2023-02-27 by kisvegabor