lvgl/lvgl #3944: How to implement "home" gesture in a scrollable menu?

952736147, 2023-01-25:
### Introduce the problem

### Examples and cases

### Suggested solution

kisvegabor, 2023-01-25:
I'd do something like this:
```c
static void home_event_cb(lv_event_t * e)
{
    lv_indev_t * indev = lv_event_get_indev(e);
    if(lv_indev_get_gesture_dir(indev) == LV_DIR_TOP) {
        printf("Home\n");
    }
}

void lv_example_home(void)
{
    /*Create a list*/
    lv_obj_t * list1 = lv_list_create(lv_scr_act());
    lv_obj_set_size(list1, lv_pct(100), lv_pct(100));
    lv_obj_center(list1);

    /*Add buttons to the list*/
    for(int i = 0; i < 50; i++) {
        lv_list_add_btn(list1, LV_SYMBOL_CLOSE, "Close");
    }

    lv_obj_t * bottom_obj = lv_obj_create(lv_scr_act());
    lv_obj_set_size(bottom_obj, lv_pct(100), 50);
    lv_obj_align(bottom_obj, LV_ALIGN_BOTTOM_MID, 0, 0);
    lv_obj_clear_flag(bottom_obj, LV_OBJ_FLAG_GESTURE_BUBBLE);
    lv_obj_add_event_cb(bottom_obj, home_event_cb, LV_EVENT_GESTURE, NULL);
    lv_obj_set_style_bg_opa(bottom_obj, LV_OPA_50, 0);
}
```
Of course you can make `bottom_obj` fully transparent. 

[image]

952736147, 2023-01-25:
> I'd do something like this:
>
> ```c
> static void home_event_cb(lv_event_t * e)
> {
>     lv_indev_t * indev = lv_event_get_indev(e);
>     if(lv_indev_get_gesture_dir(indev) == LV_DIR_TOP) { [...]

Thank you for your good advise. I will work on it and try to support more gestures. Thanks again

kisvegabor, 2023-01-26:
You're welcome. :slightly_smiling_face: 

Not that we are using GitHub for development related discussions and how-to questions should be asked in the Forum.

So I close this issue now.

closed 2023-01-26 by kisvegabor