lvgl/lvgl #3945: lv_example_chart_6  Attribute error in chart example.

kdschlosser, 2023-01-25:
### Perform all steps below and tick them with [x]
- [x ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x ] Update lvgl to the latest version
- [ x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
/examples/widgets/chart/lv_example_chart_6.py has an error in it.

```
Traceback (most recent call last):
  File "<stdin>", line 88, in event_cb
AttributeError: no such attribute
```

line88: `                lv.draw_rect(a, dsc.clip_area, draw_rect_dsc)```

The attribute error is 2 fold. 
` lv.draw_rect` and also `dsc.clip_area` 

This problem exists in both 8.3 and 9.0. I believe it is a build issue. In the lv_mp.c file for my compiled version draw_rect is present.

there is no clip_area field for `lv.obj_draw_part_dsc_t`.
The order of the parameters for `lv.draw_rect` is also wrong and the types being passed to it are incorrect as well.
`void lv_draw_rect(lv_draw_ctx_t * draw_ctx, const lv_draw_rect_dsc_t * dsc, const lv_area_t * coords)`

I do not believe that `clip_area` could actually be used because of the parameters that need to be passed to `lv.draw_rect` does not allow for passing 2 lv.area_t objects. Maybe the call is supposed to be `lv.draw_rect(dsc.draw_ctx, draw_rect_dsc, a)` Then the only thing that would need to be corrected after that would be the build in order to get `lv_draw_rect` exposed to MicroPython 

### To Reproduce
https://sim.lvgl.io/v8.3/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/1853cc5143f82842aa7b1f9f80bde64ce65845fe/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/1853cc5143f82842aa7b1f9f80bde64ce65845fe/examples/widgets/chart/lv_example_chart_6.py

https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/1853cc5143f82842aa7b1f9f80bde64ce65845fe/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/1853cc5143f82842aa7b1f9f80bde64ce65845fe/examples/widgets/chart/lv_example_chart_6.py

you have to click on one of the dots on the chart in the example for the error to occur. 

### Expected behavior
not have the attribute error

I would fix the issue but I am not sure of how to go about doing it because of the need to change the compiled MicroPython used in the simulator.

kisvegabor, 2023-01-26:
Hi,

I've just checked it and this part is really out dated and it really should work as you have described. I don't know why `lv.draw_rect` and `lv.draw_label()` are missing from the binding. Probably @amirgon knows the reason. 

In an [other example](https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/347d16b4217cef7a67566394d4b468a1307464d9/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/347d16b4217cef7a67566394d4b468a1307464d9/examples/widgets/chart/lv_example_chart_2.py) the draw function are called from the `draw_ctx` directly. It works, but a little bit hacky.

I [updated the script](https://sim.lvgl.io/v8.3/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/1853cc5143f82842aa7b1f9f80bde64ce65845fe/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/1853cc5143f82842aa7b1f9f80bde64ce65845fe/examples/widgets/chart/lv_example_chart_6.py&script_direct=2fa809b63cea79f715ed4fc2c744e6caabbfc2de) accordingly and now it works well.

amirgon, 2023-01-27:
> I don't know why `lv.draw_rect` and `lv.draw_label()` are missing from the binding. Probably @amirgon knows the reason.

They are not missing.  

`lv_draw_rect` first argument is `lv_draw_ctx_t`, therefore it is considered as a member of `lv_draw_ctx_t`.  
If `c` is an object of type `lv.draw_ctx_t`, then the Python method `c.rect` calls `lv_draw_rect`.  
Same for draw_label - `c.label`.

> I [updated the [...]

if `lv_draw_rect` expects argument of type `lv_draw_ctx_t *`, why do you pass it an object of type `lv_draw_rect_dsc_t *`?
The binding script cannot guess this kind of type casting.

kdschlosser, 2023-01-28:
There are a lot of examples that have issues.

Ones that I have reported scroll bar issues may or may not be an issue. I think it is an issue but it may be an internal thing to LVGL instead of the script needing to be updated.

A scroll bar when clicked on if dragged up should scroll the screen up with the bar moving in the same direction and down should be down for scrolling and the bar as well. The bar highlights like it has been clicked on but it moves in the opposite direction it should

The mouse wheel does not work in any of the examples running in the simulators. On the help webpage if an example is in focus the mouse wheel changes the focus of the items inside of the example. Mouse wheel is generally used for scrolling purposes not as a tab key would be used to change item focus. I think there is an issue there.

You can ignore the examples where I have noted scroll bar problems if the design of LVGL is to move in the wrong direction when a scroll bar is clicked and dragged.

The image is not rotated
https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/styles/lv_example_style_6.py

Just doesn't work at all. The code is not even loaded.
https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/styles/lv_example_style_15.py

issues with the scroll bar
https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/event/lv_example_event_3.py

issues with the scroll bars
https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/layouts/flex/lv_example_flex_1.py

issues with scroll bars
https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/layouts/flex/lv_example_flex_4.py

issues with scroll bar
https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/layouts/flex/lv_example_flex_6.py

scroll bar is not supposed to be on the one item
https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/layouts/grid/lv_example_grid_2.py

scroll bars are not supposed to be on the items
https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/layouts/grid/lv_example_grid_3.py

scroll bars are not supposed to be on the items
https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/layouts/grid/lv_example_grid_4.py

scroll bars are not supposed to be on the items
https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/layouts/grid/lv_example_grid_5.py

scroll bars are not supposed to be on the items
https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/layouts/grid/lv_example_grid_6.py

There is no nesting of scroll-able items and no colors either
https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/scroll/lv_example_scroll_1.py

scroll bar does not work properly
[... cut ...]
https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/widgets/tileview/lv_example_tileview_1.py

scroll bars not working properly
https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/a21241d00cd8aa9879290a8290848264c23870b3/examples/widgets/win/lv_example_win_1.py

kdschlosser, 2023-01-28:
> `lv_draw_rect` first argument is `lv_draw_ctx_t`, therefore it is considered as a member of `lv_draw_ctx_t`.
> If `c` is an object of type `lv.draw_ctx_t`, then the Python method `c.rect` calls `lv_draw_rect`.
> Same for draw_label - `c.label`.

This is a bit of information that is exceedingly useful to know. I have not seen any documentation on this anywhere.

So what you are saying is that the first parameter passed is what determines the class it should be placed in?? Even if it should be placed in a different class as possibly a static method?? IDK if that situation exists at all in c code and I also do not know the internal working of c code to that extent to begin to understand how it decides what goes where. I only know enough to get into trouble with it.

I had thought initially that the file name was used to determine the class layout. It almost would make more sense that way. While the lv binding for MicroPython does work using it requires a huge amount of code to do what should be fairly simple tasks. an example is the `obj` class and its subclasses. The ability to pass things like size and position along with the parent to the constructor would reduce the code footprint quite a bit. In C code it would only need to be written a single time to set the size and the position and it would get populated to every subclass of `obj`.  The need to continually add additional lines for setting position and size not only takes up RAM it also takes up storage space. The python code file size consumes memory when loaded. Files that have less code in them consume less ram even if the same work is being done. It's just how it is.

 When working with this MCU's it is absolutely incredible how important memory management is, well resource management as a whole. The folks that work on Chromium and Firefox and other web browsers need to be given a machine with extremely tight resources and say make the program work on this without loosing any functionality. LOL. I never understood how a browser that has 2 maybe 3 websites open ends up consuming > 6gb of memory and there is really nothing that fantastic on the website, maybe a couple of small JPG images and some text. 

I am also not sure why things like borders and outlines are defaulted to an on state. It is  a lot of extra code to turn that stuff off and it also screws with the positioning of the objects. well the outline and shadow does. some classes like the lv.label class has them defaulted to off. The color should also be defaulted to transparent for the background.

If you run this code in the simulator you will see what I am taking about.

```python
import lvgl as lv

import display_driver

screen = lv.scr_act()
screen.set_style_bg_color(lv.color_hex(0x000000), 0)

obj = lv.obj(screen)
obj.set_style_bg_color(lv.color_hex(0xFFFFFF), 0)
obj.set_style_bg_opa(255, 0)
obj.set_x(0)
obj.set_y(0)
obj.set_size(2, 2)

obj1 = lv.obj(screen)
obj1.set_style_bg_opa(0, 0)
obj1.set_style_border_opa(0, 0)

obj1.set_x(0)
obj1.set_y(0)

obj2 = lv.obj(obj1)
obj2.set_style_bg_color(lv.color_hex(0xFFFFFF), 0)
obj2.set_style_bg_opa(255, 0)
obj2.set_x(0)
obj2.set_y(0)
obj2.set_size(2, 2)

print(obj1.get_style_border_width(0))
print(obj1.get_style_outline_width(0))
print(obj1.get_style_outline_pad(0))
print(obj1.get_style_shadow_width(0))
```

Now I have the white dots set to a size of 2, 2. The ONLY thing that reports a width is the border and it is reporting a width of 2. The space between the dots is a hell of a lot bigger that 2 pixels.

in order to get things to align properly when setting the position I have to call these functions of each and every single object that is created. 

```python
obj.set_style_bg_opa(255, 0)
obj.set_style_border_opa(0, 0)
obj.set_style_outline_opa(0, 0)
obj.set_style_clip_corner(True, 0)
obj.set_scrollbar_mode(lv.SCROLLBAR_MODE.OFF)
obj.set_style_border_width(0, 0)
obj.set_style_outline_width(0, 0)
obj.set_style_outline_pad(0, 0)
obj.set_style_shadow_width(0, 0)
obj.set_style_shadow_spread(0, 0)
obj.set_style_shadow_ofs_x(0, 0)
obj.set_style_shadow_ofs_y(0, 0)
obj.set_style_pad_left(0, 0)
obj.set_style_pad_right(0, 0)
obj.set_style_pad_top(0, 0)
obj.set_style_pad_bottom(0, 0)
```

and that just makes for a lot of additional code when a UI has 200+ elements in it.

kisvegabor, 2023-01-28:
@amirgon 
Thank, it makes sense now.

 > if lv_draw_rect expects argument of type lv_draw_ctx_t *, why do you pass it an object of type lv_draw_rect_dsc_t *?
The binding script cannot guess this kind of type casting.

I can't see what's the problem here:
[image]

[image]

@kdschlosser 

> You can ignore the examples where I have noted scroll bar problems if the design of LVGL is to move in the wrong direction when a scroll bar is clicked and dragged.

The scrollbar is just an indicator as on smart phones. However, it already tricked enough people to add support for desktop-like scrollbars and scrolling too. :slightly_smiling_face: 

Anyway, thank you for listing all these issues. @uraich, do you have some time take look at these?

> I am also not sure why things like borders and outlines are defaulted to an on state. It is a lot of extra code to turn that stuff off and it also screws with the positioning of the objects. well [...]

You can do a couple of things:
1. Use the basic theme. See [here](https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L572)
2. You can create you own theme or extend the current. See [here](https://docs.lvgl.io/master/overview/style.html#extending-themes)
3. Create a style instead of adding local style properties. [Example](https://docs.lvgl.io/master/overview/style.html#extending-themes)
4. Just create a function or class to create a widget with custom initial settings. Like `dot_create(parent, x, y, w, h)`

amirgon, 2023-01-28:
> I can't see what's the problem here: [image]
> 
> [image]

You are right, there is no problem:

```python
>>> help(lv.draw_ctx_t())
object struct lv_draw_ctx_t is of type lv_draw_ctx_t
  __SIZE__ -- 176
  rect -- <function>
  label -- <function>
  letter -- <function>
  img -- <function>
  img_decoded -- <function>
  line -- <function>
  polygon -- <function>
  triangle -- <function>
  arc -- <function>
  transform -- <function>
  layer_create -- <function>
  layer_adjust -- <function>
  layer_blend -- <function>
  layer_destroy -- <function>
  wait_for_finish -- <function>
```

So given an object `c` of type `draw_ctx_t`, we can simply call: `c.rect(draw_rect_dsc, a)`

kisvegabor, 2023-02-14:
@uraich could you comment if you have some spare time and interest to update the examples pointed out above?

uraich, 2023-02-15:
I am currently on travel. Will be back next week anf have a look at in then.Sent from my Galaxy

kisvegabor, 2023-02-15:
Thank you very much and have a nice trip!

kdschlosser, 2023-02-21:
sorry for not being active on this. I missed seeing the notification.

I don't think there is too much that I can do to help sort things out. I know its a lot.

@kisvegabor 
Funny about the scroll bars. I thought it worth mentioning and as it turns out the "trick" may have already been corrected??

uraich, 2023-02-21:
I had a look at example_chart_6 and compared it with my local version. The local version uses dsc.draw_ctx_rect and works perfectly fine, while the version on lvgl github uses lv.draw_rect. Has this been modified later?
I will go through the examples none by one now.
Already found why example_style_15 does not work: The Python version is simply missing!
More later.

kdschlosser, 2023-02-22:
TY for doing that @uraich I appreciate it. I am one of those people that learns by looking at the code examples. It's easier for me that way.

uraich, 2023-02-25:
@amirgon :
Going through the problems, I get this when running the C code of lv_example_canvas_1:
[image]
and this:
[image]
when doing the same with the Python equivalent.
The problem seems to come from the assignment of the grad stops (e.g.):
 rect_dsc.bg_grad.stops[0].color = lv.palette_main(lv.PALETTE.RED)
When printing the value of the color during assignment, I get something that looks ok, however, when reading back after assignment, I get 0xffffffff:

print("lv.palette_main(lv.PALETTE.RED: 0x{:08x}".format(lv.palette_main(lv.PALETTE.RED).full))
rect_dsc.bg_grad.stops[0].color = lv.palette_main(lv.PALETTE.RED)
print("grad_stops[0].color: 0x{:08x}".format(rect_dsc.bg_grad.stops[0].color.full))
gives me:
[image]
Something is fishy there but I cannot see what the problem is. Can you help?

amirgon, 2023-02-25:
Hi Uli,
It's a known problem, but there's a workaround.
See https://github.com/lvgl/lv_binding_micropython/issues/253

uraich, 2023-02-25:
Thanks Amir, I suspected that it was a problem that had to do with C arrays vs Python lists.
Unfortunately I still have a (very strange) problem:
When I try to print rect_dsc.bg_grad.stops I get:
[image]
which looks ok to me.
However when I try to assign the stops, as you propose, I get:
[image]

In fact, if I do a copy/paste of the online program, which works fine, I get the same error on my local lv_micropython.

amirgon, 2023-02-25:
> Unfortunately I still have a (very strange) problem:

You are right - there was a bug where some fields were considered "const" although they are not (const fields cannot be assigned to). `stops` is not const and should be assignable.

I've [fixed this](https://github.com/lvgl/lv_binding_micropython/commit/f960502a3260f795f21bfa90b2e7d7199430cf88) now on latest lv_micropython master branch, please try again.

kdschlosser, 2023-02-26:
This is where the problem is with the gradient stops.

#define mp_read_lv_gradient_stop_t(field) mp_read_ptr_lv_gradient_stop_t(copy_buffer(&field, sizeof(lv_gradient_stop_t)))

what is in the array is being copied and not directly returned. so any updates to the stop that are taking place in MicroPython are taking place on a copy and not the actual object in the array.

kdschlosser, 2023-02-26:
Here is the code that uses that macro

```c
    /*
     * Array convertors for lv_gradient_stop_t [2]
     */

    GENMPY_UNUSED STATIC lv_gradient_stop_t *mp_arr_to_lv_gradient_stop_t___2__(mp_obj_t mp_arr)
    {
        mp_obj_t mp_len = mp_obj_len_maybe(mp_arr);
        if (mp_len == MP_OBJ_NULL) return mp_to_ptr(mp_arr);
        mp_int_t len = mp_obj_get_int(mp_len);
        //TODO check dim!
        lv_gradient_stop_t *lv_arr = (lv_gradient_stop_t*)m_malloc(len * sizeof(lv_gradient_stop_t));
        mp_obj_t iter = mp_getiter(mp_arr, NULL);
        mp_obj_t item;
        size_t i = 0;
        while ((item = mp_iternext(iter)) != MP_OBJ_STOP_ITERATION) {
            lv_arr[i++] = mp_write_lv_gradient_stop_t(item);
        }
        return (lv_gradient_stop_t *)lv_arr;
    }
    
    GENMPY_UNUSED STATIC mp_obj_t mp_arr_from_lv_gradient_stop_t___2__(lv_gradient_stop_t *arr)
    {
        mp_obj_t obj_arr[2];
        for (size_t i=0; i<2; i++){
            obj_arr[i] = mp_read_lv_gradient_stop_t(arr[i]);
        }
        return mp_obj_new_list(2, obj_arr); // TODO: return custom iterable object!
    }
```

what needs to be done in gen_mpy is a check needs to be made to see if an array is an array of structures and if it is an array of structures the items in the array should not be copied. Python instances of those objects should be created and set into the python list that is to be returned. This way the actual objects in the array get updated.

uraich, 2023-02-26:
@amirgon: Your bug fix worked. lv_example_canvas_1 is ok now.
Many thanks!

kdschlosser, 2023-02-26:
I actually know where the issue is.

The wrong macro is being called.

`mp_read_byref_lv_gradient_stop_t` needs to be called not `mp_read_lv_gradient_stop_t`

this

```c
    GENMPY_UNUSED STATIC mp_obj_t mp_arr_from_lv_gradient_stop_t___2__(lv_gradient_stop_t *arr)
    {
        mp_obj_t obj_arr[2];
        for (size_t i=0; i<2; i++){
            obj_arr[i] = mp_read_lv_gradient_stop_t(arr[i]);
        }
        return mp_obj_new_list(2, obj_arr); // TODO: return custom iterable object!
    }
```

should read

```c
    GENMPY_UNUSED STATIC mp_obj_t mp_arr_from_lv_gradient_stop_t___2__(lv_gradient_stop_t *arr)
    {
        mp_obj_t obj_arr[2];
        for (size_t i=0; i<2; i++){
            obj_arr[i] = mp_read_byref_lv_gradient_stop_t(arr[i]);
        }
        return mp_obj_new_list(2, obj_arr); // TODO: return custom iterable object!
    }
```

uraich, 2023-02-27:
@amirgon: I have another problem with the meters. I get:
[image]
meter.add_scale() worked before.
I checked in lv_mpy.c and could not find it anymore.

uraich, 2023-02-27:
@kisvegabor : I am slowly going through all the above problems and I am through them to ~ 75%. I guess, that once finished, I should create a PR against lvgl master. However, at the moment I cannot build lv_micropython with the latest updates to the master repository because of changes to the SDL driver and lv_obj_add_event, which has been renamed.
Should I wait until these things have stabilized?

amirgon, 2023-02-27:
> However, at the moment I cannot build lv_micropython with the latest updates to the master repository because of changes to the SDL driver and lv_obj_add_event, which has been renamed.
> Should I wait until these things have stabilized?

@uraich you can work with arch/drivers branch of lv_micropython for now, it should be aligned with LVGL master.  
I still didn't merge it to lv_micropython master because there are some fixes to do in examples and drivers, but I think they shouldn't block you.

> meter.add_scale() worked before.

Looking at LVGL API I can see `lv_meter_add_scale_lines`, did you mean that?

https://github.com/lvgl/lvgl/blob/426739d06081928c7086c589b07c7ee21a02b724/src/widgets/meter/lv_meter.h#L204

uraich, 2023-02-27:
In fact there should be both: lv_meter_add scale and lv_meter_add_scale_lines:
[image]

but I only find lv_meter_add_scale_lines in lv_mpy.c

uraich, 2023-02-27:
Hmmm, I see this:
[image]

amirgon, 2023-02-27:
> Hmmm, I see this:

Are you aligned to latest LVGL?
Please try `git log -1` on lvgl dir
Also, please try to clean and rebuild.

uraich, 2023-02-27:
You are right, I was not on the latest lvgl master. lv_sim_eclipse_sdl also cannot be built with the latest master at the moment. So... things are a little tricky.
I now rebuilt lv_micropython using the arch/drivers branch and the latest lvgl master and there the C code of the meter examples has changed. 
... and that helps! The Python meter examples need quite a few changes but I got the first one to work.

uraich, 2023-02-27:
... but changing to the latest lvgl breaks plenty of other things. Renaming "add_event_cb" to "add_event" alone means changes in many examples.

amirgon, 2023-02-27:
> ... but changing to the latest lvgl breaks plenty of other things.

This surprises me a little because most LVGL python examples should work.  
There's an automated GitHub action that runs all Python examples and demos on every LVGL commit on the master branch. ([link for example](https://github.com/lvgl/lvgl/actions/runs/4286945692/jobs/7467179421), expand the "Run test" stage)
So to my understanding, most Python examples should already be fixed on the master branch, except from some event handlers which are hard to invoke in automated tests.

> Then, evt.get_target now returns a blob, which breaks many of the callbacks.

You should replace it by `evt.get_target_obj` and it should work the same as before.
but it was already replaced in many examples: [See this search results page](https://github.com/search?q=repo%3Alvgl%2Flvgl+get_target_obj+path%3A*.py&type=code)

> Maybe it is better for me to wait until things have stabilized a bit?

With the automated tests in place - I had the feeling we are already quite stable, but maybe I'm wrong?  
Where do you experience the instability?

uraich, 2023-02-28:
Sorry, I was on the wrong example folder. Things look a bit better now. 
However, the problem not being able to assign stops has re-appeared since I use the arch/drivers branch. Did you do the modification on the master branch only?

amirgon, 2023-02-28:
> However, the problem not being able to assign stops has re-appeared since I use the arch/drivers branch. Did you do the modification on the master branch only?

You are right.  
Could you run `git cherry-pick f960502a3260f795f21bfa90b2e7d7199430cf88` locally on arch/drivers branch?  
That should allow you to continue.  
I'll merge this change to `arch/drivers` later.

uraich, 2023-02-28:
It tells me: 
[image]
but I can continue with other problems and there are still quite a few! Just tell me when you did the merge.
Thanks Uli

amirgon, 2023-02-28:
> It tells me:
> [image]

That's because you need to run it in lv_binding_micropython dir

uraich, 2023-02-28:
Ah... It works now.

uraich, 2023-03-02:
Just created a PR with many changes to the examples.
lv_example_style_15 was missing and has now been added.
lv_examples_label_4 is still missing. I still have problems with it.
Quite a few of the problems described by @kdschlosser exist already in the C code and those problems have not been addressed, because all I did was providing the Python equivalents to the C codes.
I had problems comparing my Python codes to the C codes because I was unable to build 
lv_port_pc_eclipse with the newest lvgl master.
lv_example_canvas_1.c does not work correctly any longer (see screen dump taken from lvgl doc:
[image]

kisvegabor, 2023-03-03:
@uraich I've fixed the canvas transformation issue.

uraich, 2023-03-03:
In this case you may wait a little with the merge. I will try to correct the Python equivalent as well.

uraich, 2023-03-03:
Strange: I tried the canvas transformation an hour ago and things looked ok, but now the same problem is back?

kisvegabor, 2023-03-06:
It's still working in C for me.

uraich, 2023-03-06:
@kisvegabor: It seems that your modification was not yet registered in master, when I downloaded and rebuilt lv_micropython. I tried again this morning and saw a modification to canvas. After the rebuild of lv_micropython my Python version of lv_example_canvas_1 also worked. I will add it to the PR.

uraich, 2023-03-06:
lv_example_span_1 uses the fonts montserrat-20 and 24. I see that in lv_conf.h of lv_micropython only montserrat-14 and 16 are enabled by default. In such I case I try to dynamically load the fonts. However, in the standard lvgl distribution only the font files for montserrat-16,18,22 are available. 
If I do not find the font file, I use the default font.
If we want the C and the Python span examples to look the same, we will either have to use font 18 and 22 instead of 20 and 24 or we should make the font files for 20 and 24 available.

kisvegabor, 2023-03-07:
We can use font 18 and 22 in C as well. It's equally good.

closed 2023-03-29 by lvgl-bot