lvgl/lvgl #3949: Cannot generate a QR code

mircopz, 2023-01-27:
### Introduce the problem
With LVGL 8.2 on ESP32, I'm trying to replicate the QR code example, but the canvas show me "No Data". 
This is my log:

> [Info]	(13.264, +8)	 lv_qrcode_create: begin 	(in lv_qrcode.c line #61)
> [Info]	(13.291, +27)	 lv_obj_update_layout: Layout update begin 	(in lv_obj_pos.c line #311)
> [Info]	(13.292, +1)	 lv_obj_update_layout: Layout update begin 	(in lv_obj_pos.c line #311)
> [Warn]	(13.350, +58)	 decode_and_draw: Image draw can't read the line 	(in lv_draw_img.c line #313)
> Warn]	(13.350, +0)	 lv_draw_img: Image draw error 	(in lv_draw_img.c line #81)

### Examples and cases

This is my piece of code, I tried a small qr code (50 pixel)
```
    ...
    lv_color_t bg_color = lv_palette_lighten(LV_PALETTE_LIGHT_BLUE, 5);
    lv_color_t fg_color = lv_palette_darken(LV_PALETTE_BLUE, 4);

    lv_obj_t * qr = lv_qrcode_create(lv_scr_act(), 50, fg_color, bg_color);

    /*Set data*/
    const char * data = "https://lvgl.io";
    lv_qrcode_update(qr, data, strlen(data));
    lv_obj_center(qr);

    /*Add a border with bg_color*/
    lv_obj_set_style_border_color(qr, bg_color, 0);
    lv_obj_set_style_border_width(qr, 5, 0);
    ...
```

### Suggested solution

kisvegabor, 2023-01-28:
Could you try with the latest LVGL version? (v8.3.4)?

mircopz, 2023-01-30:
I just tried using the v8.3.4, but the problem is still present. How can I debug it more in depth?

PS: migrating to that version I probably found a bug, now fixed with this PR https://github.com/lvgl/lvgl/pull/3951. Please check if it's reasonable.

kisvegabor, 2023-02-01:
I've just copy pasted the code you have added above and it worked just fine for me.

Please try again with the last commit from `release/v8.3` and send the log again. 

> [Warn] (13.350, +58) decode_and_draw: Image draw can't read the line (in lv_draw_img.c line https://github.com/lvgl/lvgl/issues/313)

would be useful, but in the latest `release/v8.3` it points to a not related line.

> PS: migrating to that version I probably found a bug, now fixed with this PR https://github.com/lvgl/lvgl/pull/3951. Please check if it's reasonable.

Merged, thank you :slightly_smiling_face:

mircopz, 2023-02-01:
@kisvegabor I found my mistake!
In my project I'm using a custom decoder to load image files from file system. Since I read on the documentation that `decoder_read` is optional, I didn't set a read line callback.
Simply adding the definition of the this callback (the built in one), it works properly
```
lv_img_decoder_t *dec = lv_img_decoder_create();
lv_img_decoder_set_info_cb(dec, decoder_info); // my function
lv_img_decoder_set_open_cb(dec, decoder_open); // my function
lv_img_decoder_set_close_cb(dec, decoder_close); // my function
lv_img_decoder_set_read_line_cb(dec, lv_img_decoder_built_in_read_line); // built-in function
```

I suggest to specify it in the documentation that for custom decoder is mandatory to specify at least the built in callback. Otherwise, if the image is loaded with a custom decoder, the lv_img_decoder_read_line function doesn' find a `read_line_cb` and returns LV_RES_INV.

```
lv_res_t lv_img_decoder_read_line(lv_img_decoder_dsc_t * dsc, lv_coord_t x, lv_coord_t y, lv_coord_t len, uint8_t * buf)
{
    lv_res_t res = LV_RES_INV;
    if(dsc->decoder->read_line_cb) res = dsc->decoder->read_line_cb(dsc->decoder, dsc, x, y, len, buf);

    return res;
}
```

Thank you for your support

closed 2023-02-01 by mircopz

kisvegabor, 2023-02-04:
Normally, your custom `decoder_open` should return `LV_RES_INV` if it can't handle the image, so LVGL will try an other decoder. 

Does your `decoder_open` callback do this?

mircopz, 2023-02-06:
No my custom decoder is 

```
static lv_res_t decoder_open(lv_img_decoder_t *decoder, lv_img_decoder_dsc_t *dsc) {
  if (dsc->src_type != LV_IMG_SRC_FILE)
    return lv_img_decoder_built_in_open(decoder, dsc);

  // my custom code to load image from file system

  return LV_RES_OK;
}
```

Maybe I should replace `lv_img_decoder_built_in_open(decoder, dsc)` with `LV_RES_INV`, right?

kisvegabor, 2023-02-07:
You should do something like this:
```c
static lv_res_t decoder_open(lv_img_decoder_t *decoder, lv_img_decoder_dsc_t *dsc) {
   //Ignore the non-files, leave them to an other decoder
   if (dsc->src_type != LV_IMG_SRC_FILE) return LV_RES_INV;      
  
  //If the extension doesn't match return
  const char * fn = src;
  if(strcmp(lv_fs_get_ext(fn), "abc")) return LV_RES_INV;  

  // my custom code to load image from file system

  return LV_RES_OK;     
}
```

mircopz, 2023-02-07:
Ok thank you, now that I've read the documentation again it's clearer!