lvgl/lvgl #3950: lv_roller does not honor lv_label_set_recolor

kdschlosser, 2023-01-28:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I am running the MicroPython binding and so the code example is written in Python

I have looked at the c code for lv_roller and because it bypasses the code in lv_label that calls lv_draw_label and uses its own code to call lv_draw_label it does not set LV_TEXT_FLAG_RECOLOR in lv_draw_label_dsc_t.flag so the renderer never takes into account the color commands in the text. 

### To Reproduce
<details>
  <summary>Code Example (click to expand)</summary>

```python
##### startup script #####

#!/opt/bin/lv_micropython -i

import lvgl as lv

import display_driver

# find a better more memory efficient way to get the colors
def _hue2rgb(v1, v2, vH):
    while vH < 0:
        vH += 1
    while vH > 1:
        vH -= 1

    if 6 * vH < 1:
        return v1 + (v2 - v1) * 6 * vH
    if 2 * vH < 1:
        return v2
    if 3 * vH < 2:
        return v1 + (v2 - v1) * ((2.0 / 3) - vH) * 6

    return v1

def hsl2rgb(h, s, l):
    if s == 0:
        return l, l, l

    if l < 0.5:
        v2 = l * (1.0 + s)
    else:
        v2 = (l + s) - (s * l)

    v1 = 2.0 * l - v2

    r = _hue2rgb(v1, v2, h + (1.0 / 3))
    g = _hue2rgb(v1, v2, h)
    b = _hue2rgb(v1, v2, h - (1.0 / 3))

    return int(r * 255), int(g * 255), int(b * 255)

def rgb2hsl(r, g, b):
    r /= 255.0
    g /= 255.0
    b /= 255.0

    vmin = min(r, g, b)
    vmax = max(r, g, b)
    diff = vmax - vmin

    vsum = vmin + vmax

    l = vsum / 2

    if diff < 0.0000005:
        return 0.0, 0.0, l

    if l < 0.5:
        s = diff / vsum
    else:
        s = diff / (2.0 - vsum)

    dr = (((vmax - r) / 6) + (diff / 2)) / diff
    dg = (((vmax - g) / 6) + (diff / 2)) / diff
    db = (((vmax - b) / 6) + (diff / 2)) / diff

    if r == vmax:
        h = db - dg
    elif g == vmax:
        h = (1.0 / 3) + dr - db
    elif b == vmax:
        h = (2.0 / 3) + dg - dr
    else:
        raise RuntimeError  # sanity check

    if h < 0:
        h += 1
    if h > 1:
        h -= 1

    return h, s, l

def color_scale(begin_hsl, end_hsl, nb):
    step = tuple([float(end_hsl[i] - begin_hsl[i]) / nb for i in range(0, 3)]) if nb > 0 else (0, 0, 0)

    def mul(step, value):
        return tuple([v * value for v in step])

    def add_v(step, step2):
        return tuple([v + step2[i] for i, v in enumerate(step)])

    def rgb2hex(r, g, b):
        r = hex(r)[2:].upper()
        g = hex(g)[2:].upper()
        b = hex(b)[2:].upper()

        if len(r) < 2:
            r = '0' + r

        if len(g) < 2:
            g = '0' + g
        
        if len(b) < 2:
            b = '0' + b

        return '#' + r + g + b

    return [rgb2hex(*hsl2rgb(*add_v(begin_hsl, mul(step, r)))) for r in range(0, nb + 1)]
##### main script #####

import fs_driver
import sys

class Lv_Roller_3():

    def __init__(self):
        self.mask_top_id = -1
        self.mask_bottom_id = -1

        #
        # Add a fade mask to roller.
        #
        style = lv.style_t()
        style.init()
        style.set_bg_color(lv.color_black())
        style.set_text_color(lv.color_white())

        lv.scr_act().add_style(style, 0)

        roller1 = lv.roller(lv.scr_act())
        roller1.add_style(style, 0)
        roller1.set_style_border_width(0, 0)
        roller1.set_style_pad_all(0, 0)
        roller1.set_style_bg_opa(lv.OPA.TRANSP, lv.PART.SELECTED)

        #if LV_FONT_MONTSERRAT_22
        #    lv_obj_set_style_text_font(roller1, &lv_font_montserrat_22, LV_PART_SELECTED);
        #endif
        try:
            roller1.set_style_text_font(lv.font_montserrat_22,lv.PART.SELECTED)
        except:
            fs_drv = lv.fs_drv_t()
            fs_driver.fs_register(fs_drv, 'S')
            print("montserrat-22 not enabled in lv_conf.h, dynamically loading the font")
            font_montserrat_22 = lv.font_load("S:" + "../../assets/font/montserrat-22.fnt")
            roller1.set_style_text_font(font_montserrat_22,lv.PART.SELECTED)

        colors = color_scale(
            rgb2hsl(255, 0, 0),
            rgb2hsl(255, 142, 10),
            101
        )

        colors.extend(
            color_scale(rgb2hsl(255, 142, 10), rgb2hsl(0, 0, 255), 240)
        )

        items = list(str(i) + '°F' for i in range(300, -41, -1))
        items = '\n'.join(item[0] + ' ' + item[1] + '# ' for item in zip(colors, items))

        lv.label.__cast__(roller1.get_child(0)).set_recolor(True)

        roller1.set_options(items, lv.roller.MODE.NORMAL)

        # lv.label.__cast__(roller1.get_child(0)).set_text(items)

        roller1.center()
        roller1.set_visible_row_count(9)
        roller1.add_event_cb(self.mask_event_cb, lv.EVENT.ALL, None)
        # lv_roller_set_selected

    def mask_event_cb(self,e):

        code = e.get_code()
        obj = e.get_target()

        if code == lv.EVENT.COVER_CHECK:
            e.set_cover_res(lv.COVER_RES.MASKED)

        elif code == lv.EVENT.DRAW_MAIN_BEGIN:
            # add mask
            font = obj.get_style_text_font(lv.PART.MAIN)
            line_space = obj.get_style_text_line_space(lv.PART.MAIN)
            font_h = font.get_line_height()

            roller_coords = lv.area_t()
            obj.get_coords(roller_coords)

            rect_area = lv.area_t()
            rect_area.x1 = roller_coords.x1
            rect_area.x2 = roller_coords.x2
            rect_area.y1 = roller_coords.y1
            rect_area.y2 = roller_coords.y1 + (obj.get_height() - font_h - line_space) // 2

            fade_mask_top = lv.draw_mask_fade_param_t()
            fade_mask_top.init(rect_area, lv.OPA.TRANSP, rect_area.y1, lv.OPA.COVER, rect_area.y2)
            self.mask_top_id = lv.draw_mask_add(fade_mask_top,None)

            rect_area.y1 = rect_area.y2 + font_h + line_space - 1
            rect_area.y2 = roller_coords.y2

            fade_mask_bottom = lv.draw_mask_fade_param_t()
            fade_mask_bottom.init(rect_area, lv.OPA.COVER, rect_area.y1, lv.OPA.TRANSP, rect_area.y2)
            self.mask_bottom_id = lv.draw_mask_add(fade_mask_bottom, None)

        elif code == lv.EVENT.DRAW_POST_END:
            fade_mask_top = lv.draw_mask_remove_id(self.mask_top_id)
            fade_mask_bottom = lv.draw_mask_remove_id(self.mask_bottom_id)
            # Remove the masks
            lv.draw_mask_remove_id(self.mask_top_id)
            lv.draw_mask_remove_id(self.mask_bottom_id)
            self.mask_top_id = -1
            self.mask_bottom_id = -1

roller3 = Lv_Roller_3()

```

</details>

### Expected behavior
The rendered text to be colored properly

kisvegabor, 2023-01-28:
I've just fixed it both `master` and `release/v8.3`.

closed 2023-01-28 by kisvegabor

kdschlosser, 2023-01-30:
TY for fixing this so fast I appreciate it. One question tho.

How long before the fix gets populated into the simulator?

kisvegabor, 2023-01-31:
> How long before the fix gets populated into the simulator?

@amirgon @embeddedt could you comment on this? I've pushed a fix to lvgl `master` and `release/v8.3`.

kdschlosser, 2023-01-31:
It's just that the simulator makes it easier to tinker about. It would be pretty cool if there was the ability to upload files instead of having to have a single file with everything dumped into it. I am talking about the MicroPython simulator.

IDK if there is a C simulator. I wouldn't imagine there is because of compile times but you never know.

amirgon, 2023-01-31:
> @amirgon @embeddedt could you comment on this? I've pushed a fix to lvgl `master` and `release/v8.3`

I've updated LVGL on lv_micropython "master" and "release/v8" branches.
This should be reflected now on the online simulator (you may need to clear the browser cache)

kdschlosser, 2023-02-01:
OK thanks a heap for taking care of this so fast. I really appreciate it. I did my best in describing what I thought was causing the problem hoping that it would make it easier fixing it.

kdschlosser, 2023-02-01:
OoOOo it's not working properly..

[image]

see the temperatures?. 188, 187, 148?!?? 185 184....

Here is the code I am using
<details>
  <summary>Code (click here to see)</summary>

```python
##### startup script #####

#!/opt/bin/lv_micropython -i

import lvgl as lv

import display_driver

# find a better more memory efficient way to get the colors
def _hue2rgb(v1, v2, vH):
    while vH < 0:
        vH += 1
    while vH > 1:
        vH -= 1

    if 6 * vH < 1:
        return v1 + (v2 - v1) * 6 * vH
    if 2 * vH < 1:
        return v2
    if 3 * vH < 2:
        return v1 + (v2 - v1) * ((2.0 / 3) - vH) * 6

    return v1

def hsl2rgb(h, s, l):
    if s == 0:
        return l, l, l

    if l < 0.5:
        v2 = l * (1.0 + s)
    else:
        v2 = (l + s) - (s * l)

    v1 = 2.0 * l - v2

    r = _hue2rgb(v1, v2, h + (1.0 / 3))
    g = _hue2rgb(v1, v2, h)
    b = _hue2rgb(v1, v2, h - (1.0 / 3))

    return int(r * 255), int(g * 255), int(b * 255)

def rgb2hsl(r, g, b):
    r /= 255.0
    g /= 255.0
    b /= 255.0

    vmin = min(r, g, b)
    vmax = max(r, g, b)
    diff = vmax - vmin

    vsum = vmin + vmax

    l = vsum / 2

    if diff < 0.0000005:
        return 0.0, 0.0, l

    if l < 0.5:
        s = diff / vsum
[... 141 lines cut ...]
            fade_mask_top = lv.draw_mask_fade_param_t()
            fade_mask_top.init(rect_area, lv.OPA.TRANSP, rect_area.y1, lv.OPA.COVER, rect_area.y2)
            self.mask_top_id = lv.draw_mask_add(fade_mask_top,None)

            rect_area.y1 = rect_area.y2 + font_h + line_space - 1
            rect_area.y2 = roller_coords.y2

            fade_mask_bottom = lv.draw_mask_fade_param_t()
            fade_mask_bottom.init(rect_area, lv.OPA.COVER, rect_area.y1, lv.OPA.TRANSP, rect_area.y2)
            self.mask_bottom_id = lv.draw_mask_add(fade_mask_bottom, None)

        elif code == lv.EVENT.DRAW_POST_END:
            fade_mask_top = lv.draw_mask_remove_id(self.mask_top_id)
            fade_mask_bottom = lv.draw_mask_remove_id(self.mask_bottom_id)
            # Remove the masks
            lv.draw_mask_remove_id(self.mask_top_id)
            lv.draw_mask_remove_id(self.mask_bottom_id)
            self.mask_top_id = -1
            self.mask_bottom_id = -1

roller3 = Lv_Roller_3()
```
</details>

I haven't looked at the roller code yet but I would imagine it is calculating the location based on the length of the string before it is parsed and the color commands removed.  probably need to calculate the position by counting the newline characters instead of using string length.

kisvegabor, 2023-02-04:
I think the issue is the MicroPython is was not built with `LV_USE_LARGE_COORD  0` (`lv_conf.h`). Could you try how it works if you set it to `1`?

kdschlosser, 2023-02-04:
I will give that a go. I have only been messing about with it in the simulator and I have not moved to to an actual part. It is going to be a while before I do that.

commit 8cbd718e (2023-01-28): fix(roller): consider the recolor setting of the label

fixes #3950
commit 39f42476 (2023-01-28): fix(roller): consider the recolor setting of the label

fixes #3950