lvgl/lvgl #3958: Discussion of whether `lvgl` border needs to shrink the `clip area` for children

W-Mai, 2023-01-31:
### Introduce the problem

Note that lvgl adds the width of the border to the inward offset, but does not subtract the area of the border when drawing the children. This results in an unexpected overlap between the border and the children when the border is translucent.
### Examples and cases

After making the border transparent

The unexpected:
[image]

The expected:
[image]

### Suggested solution

https://github.com/lvgl/lvgl/blob/837f5035923e9ee782fb3663cbef379c78f201c2/src/core/lv_refr.c#L162-L173

make changes like this (Not the final code)

```cpp
    lv_area_t clip_coords_for_children;
    bool refr_children = true;
    if(lv_obj_has_flag(obj, LV_OBJ_FLAG_OVERFLOW_VISIBLE)) {
        clip_coords_for_children  = *clip_area_ori;
    }
    else {
        lv_area_t chi_coords = obj->coords;
        lv_coord_t border_width = lv_obj_get_style_border_width(obj, LV_PART_MAIN);
        if (border_width) {
            chi_coords.x1 += border_width;
            chi_coords.x2 -= border_width;
            chi_coords.y1 += border_width;
            chi_coords.y2 -= border_width;
        }

        if(!_lv_area_intersect(&clip_coords_for_children, clip_area_ori, &chi_coords)) {
            refr_children = false;
        }
    }

```

kisvegabor, 2023-01-31:
It should work as you have described:
```c
  lv_obj_t * cont = lv_obj_create(lv_scr_act());
  lv_obj_set_style_border_width(cont, 30, 0);
  lv_obj_set_style_pad_all(cont, 0, 0);
  lv_obj_set_size(cont, 300, 300);

  LV_IMG_DECLARE(img_cogwheel_argb);
  lv_obj_t * img = lv_img_create(cont);
  lv_img_set_src(img, &img_cogwheel_argb);
```
[image]

With:
```c
  lv_obj_set_style_border_width(cont, 5, 0);
```
[image]

W-Mai, 2023-01-31:
What I mean is that when the child is larger than the parent component, scroll the child component under the parent's border, and the child under the border still renders

W-Mai, 2023-01-31:
The expected:

[image]

This screenshot might clarify the problem better

kisvegabor, 2023-01-31:
There is a `border_post` style property to control when the border should be drawn:
```c
lv_obj_set_style_border_post(cont, true, 0);
```
[image]

```c
lv_obj_set_style_border_post(cont, false, 0);
```
[image]

Can it solve the issue?

Actually I'm thinking about removing the `border_post` property to make it just work as what the user would expect. But I still need to understand more what would  the "expected way of working".

W-Mai, 2023-01-31:
> here is a border_post style property to control when the border should be drawn:
> lv_obj_set_style_border_post(cont, true, 0);

Yes, this will hide the children, but if the border is translucent, the border and children will overlap
```cpp
lv_obj_set_style_border_opa(obj0, LV_OPA_30, LV_PART_MAIN);
```

But what I'm expecting is that children won't overlap the border, even though the border is transparent

W-Mai, 2023-01-31:
The way to fix this is to shrink the children's clip region so that it renders only the area within the subtracted border

So, the border drawing post or not will not to be a problem😁

kisvegabor, 2023-01-31:
It doesn't work well if the border has an inner radius. Like this:
[image]

W-Mai, 2023-01-31:
Exactly, so can we fix this problem?

kisvegabor, 2023-02-01:
This is how it works on web:
[image]
https://codepen.io/kisvegabor/pen/gOjQaNq

We there is a mask on the inner part of the border to clip the content.

It might be a little overkill for most of the embedded systems.

I think `border_post:true` is a good compromise. What do you think?

W-Mai, 2023-02-01:
Agree

I also think clip causes more computation, which is a bit of a pants drop in an embedded development case

Indeed, `border_post: True` is a good solution

kisvegabor, 2023-02-04:
Okay. So can we close this issue? :slightly_smiling_face:

W-Mai, 2023-02-04:
yes😁

kisvegabor, 2023-02-04:
Anyway, now we see it clearer.

closed 2023-02-04 by kisvegabor