lvgl/lvgl #3963: Memory leak (lv_disp_drv_register) draw_ctx

simplejack-src, 2023-02-02:
Valgrind is of the opinion (and I concur) that the memory malloc'd for the `draw_ctx` in `lv_disp_drv_register` is never freed*. I don't have enough context in this area to submit a clear fix (maybe an API addition of the `lv_disp_drv_unregister` variety)?

*`lvgl/src/hal/lv_hal_disp:lv_disp_drv_register:182`

kisvegabor, 2023-02-04:
We will soon merge some rework on display/indev architecture to `master` and I'd like to fix it in the new version. 

BTW, these changes are part of a conceptual change in which we will move all the drivers to the main `lvgl` repo. Of course it'd be great to have the Wayland driver there as well. 

Can I tag you to help with it once we get there?

simplejack-src, 2023-02-05:
That makes sense (I did see some re-arch issues around and the commit did look somewhat temporary).

On the other front, I'd actually recommend deprecating the Wayland driver (over in lv_drivers). I think the SDL driver is a much better fit (it supports more Wayland protocols, libdecor and HW-acceleration). The only thing the Wayland driver has going for it is size (somewhat moot, given for Wayland you still have to be running an entire Linux OS) and shared-memory-only buffers (which I may look into adding on the SDL backend).

kisvegabor, 2023-02-05:
So you say that, if you can use Wayland, you can use SDL too which is better choice, right?

simplejack-src, 2023-02-05:
Full disclosure, I haven't actually used the SDL driver in LVGL, however, a cursory look over it seems it's just using the standard SDL functions to render the LVGL frame (`cb_flush`)*. Which in turn get passed off to an actual SDL video backend (be it Windows, X11, Wayland, UIKit, etc.).

Seeing as SDL also supports a SW-raster (and mesa for that matter), I can't recall a system where you would be able to run Linux but not SDL (as the only thing lacking from the SDL Wayland video backend is using shared-memory as opposed to EGL). But in the event that you had a Linux system, with no EGL support (perhaps mesa SW-rast wasn't cutting it for you) and your using a Wayland compositor that doesn't need EGL, you would then have to (currently) use a different LVGL driver (perhaps the Linux FB one). So yes, in this sort of specific scenario SDL is missing that support (but I may look into adding it upstream).

*This is just referencing the SDL driver (in lv_drivers repo) not the potentially HW-accelerated use (in lvgl repo) used to actually perform draw operations when actually composing the frame (i.e. before `cb_flush`).

simplejack-src, 2023-02-06:
The DRM driver looks similar (in functionality) to the Linux FB driver. Arguably, the FB driver should also be deprecated, as last I heard those driver were deprecated by the upstream kernel and were being phased out/dropped (especially given that a shim driver exists that emulates the FB interface, over-top of the current DRM interface).

kisvegabor, 2023-02-07:
I think it will take a long time for the embedded Linux distros to catch up with this change the kernel. But anyway, we should focus more on DRM. 
Thanks!

closed 2023-03-01 by lvgl-bot