lvgl/lvgl #3968: tiny_ttf sometimes crashes with "stbtt_FindGlyphIndex: Assertion `0' failed."

amirgon, 2023-02-03:
This happens on `arch/drivers` branch, but I think I've seen it also on the master branch.

Sometimes (non deterministic), `lv_example_tiny_ttf_2.py` test fails with
```
lvgl/src/libs/tiny_ttf/stb_truetype_htcw.h:1781: stbtt_FindGlyphIndex: Assertion `0' failed.
```

Example: https://github.com/lvgl/lv_micropython/actions/runs/4088205161/jobs/7049622649

Test log:
```
Running the SDL lvgl version
micropython-dev: ../../lib/lv_bindings/lvgl/src/libs/tiny_ttf/stb_truetype_htcw.h:1781: stbtt_FindGlyphIndex: Assertion `0' failed.
Aborted (core dumped)
Running lib/lv_bindings/tests/../lvgl/examples/libs/tiny_ttf/lv_example_tiny_ttf_2.py ...
parallel: This job failed:
timeout 5m catchsegv lib/lv_bindings/tests/../../../ports/unix/micropython-dev lib/lv_bindings/tests/run_test.py lib/lv_bindings/tests/../lvgl/examples/libs/tiny_ttf/lv_example_tiny_ttf_2.py
Error: Process completed with exit code 134.
```

This seems like an internal issue with `tiny_ttf` because this is a very simple example that only calls `lv_tiny_ttf_create_file` and displays some text.

kisvegabor, 2023-02-05:
I've also saw that, but I have no idea. I've tried with ASan, Valgrind, -O0, -O3 but can't reproduce the issue. Interestingly it happens only the MicroPython build. Why could it be different from the others?

Is there a way print the call stack on assert/crash?

amirgon, 2023-02-05:
> Interestingly it happens only with the MicroPython build. Why could it be different from the others?

One important difference is memory allocation. If we missed a gc root for example.
It's also interesting that it only happens with `lv_tiny_ttf_create_file`. So the problem could be related to:
- tiny_ttf cache. missing gc root?
- lv_fs, there's also some cache there right?

> Is there a way print the call stack on assert/crash?

Unfortunately we don't have a stack trace, but I'm not sure a stack trace would be helpful here if this is really a memory corruption.
Looking at the assert location:

https://github.com/lvgl/lvgl/blob/70321fb102ce34fba2e3828e4744631a7ed34004/src/libs/tiny_ttf/stb_truetype_htcw.h#L1780-L1782

Looks like `format` has some value that `stbtt_FindGlyphIndex` can't handle. If this is a memory corruption then format has some invalid value here and a stack trace won't help much to find out how it was corrupted.

But most of the memory corruptions we had in the past (where Micropython was involved) were eventually related to a missing gc root, so that is the direction I would look.

kisvegabor, 2023-02-05:
I've tried to review the related code but it's a quite large and a little bit messy on some points. Anyway, I've see only `malloc`s and no static or global variables. In `lv_fs` the cache is also `malloc`ed.

For tiny_ttf the `malloc` are used like this:
```c
#define TTF_CACHE_MALLOC(x)  (lv_malloc(x))
#define TTF_CACHE_REALLOC(x,y)  (lv_realloc(x,y))
#define TTF_CACHE_FREE(x)    (lv_free(x))
#define TTF_MALLOC(x)  (lv_malloc(x))
#define TTF_FREE(x)    (lv_free(x))
```
which seems good to me.

Do you see the error if you run the tests locally too? If so what happens if you turn off GC?

amirgon, 2023-02-05:
> Do you see the error if you run the tests locally too? If so what happens if you turn off GC?

It's not easily reproducible, but I'll try to run some stress tests later this week.

amirgon, 2023-02-06:
> Do you see the error if you run the tests locally too?

Yes, I can see this happening locally by running the example in a loop long enough.
For example:
```bash
for i in {1..50}; do ports/unix/micropython-dev lib/lv_bindings/tests/run_test.py /home/amirgon/projects/lv_micropython/lib/lv_bindings/lvgl/examples/libs/tiny_ttf/lv_example_tiny_ttf_2.py; done
```

>  If so what happens if you turn off GC?

Actually, it also happens with gc disabled!!
I had to increase the gc RAM to 100MB for this to work (`heap_size` in `ports/unix/main.c`) and run it in a loop like above, and the problem eventually happens in the same way.
So, this is not a missing gc root, but something else.

> Is there a way print the call stack on assert/crash?

I created a crash dump, to make it easier to debug the problem.
You can run it like this:
```bash 
gdb ports/unix/micropython-dev core.1898500
```

Both `ports/unix/micropython-dev` and `core.1898500` are in tgz file: https://www.transfernow.net/dl/lvgl_3968 (available for 7 days). Sources are the "arch/drivers" branch of lv_micropython, if you need them.

Could you have a look at the crashdump?

kisvegabor, 2023-02-07:
Thanks. I've looked into it but doesn't seem to work:
```shell
kisvegabor ~/Desktop/lvgl_3968 $gdb ports/unix/micropython-dev core.1898500
GNU gdb (Debian 10.1-1.7) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ports/unix/micropython-dev...

warning: Can't open file /home/amirgon/projects/lv_micropython/ports/unix/micropython-dev during file-backed mapping note processing

warning: Can't open file /usr/lib/x86_64-linux-gnu/libgpg-error.so.0.28.0 during file-backed mapping note processing

warning: Can't open file /usr/lib/x86_64-linux-gnu/libbsd.so.0.10.0 during file-backed mapping note processing

warning: Can't open file /usr/lib/x86_64-linux-gnu/libvorbisenc.so.2.0.11 during file-backed mapping note processing

warning: Can't open file /usr/lib/x86_64-linux-gnu/libvorbis.so.0.4.8 during file-backed mapping note processing

warning: Can't open file /usr/lib/x86_64-linux-gnu/libgcrypt.so.20.2.5 during file-backed mapping note processing

warning: Can't open file /usr/lib/x86_64-linux-gnu/liblz4.so.1.9.2 during file-backed mapping note processing

warning: Can't open file /usr/lib/x86_64-linux-gnu/liblzma.so.5.2.4 during file-backed mapping note processing

warning: Can't open file /usr/lib/x86_64-linux-gnu/libapparmor.so.1.6.1 during file-backed mapping note processing

warning: Can't open file /usr/lib/x86_64-linux-gnu/libsndfile.so.1.0.28 during file-backed mapping note processing

warning: Can't open file /usr/lib/x86_64-linux-gnu/libsystemd.so.0.28.0 during file-backed mapping note processing

warning: Can't open file /usr/lib/x86_64-linux-gnu/libdbus-1.so.3.19.11 during file-backed mapping note processing

warning: Can't open file /usr/lib/x86_64-linux-gnu/pulseaudio/libpulsecommon-13.99.so during file-backed mapping note processing

warning: Can't open file /usr/lib/x86_64-linux-gnu/libX11.so.6.3.0 during file-backed mapping note processing

warning: Can't open file /usr/lib/x86_64-linux-gnu/libpulse.so.0.21.2 during file-backed mapping note processing

warning: Can't open file /usr/lib/x86_64-linux-gnu/libfreetype.so.6.17.1 during file-backed mapping note processing

warning: Can't open file /usr/lib/x86_64-linux-gnu/librlottie.so.0.0.1-1 during file-backed mapping note processing

warning: Can't open file /usr/lib/x86_64-linux-gnu/libSDL2-2.0.so.0.10.0 during file-backed mapping note processing
[New LWP 1898500]

warning: .dynamic section for "/lib/x86_64-linux-gnu/libSDL2-2.0.so.0" is not at the expected address (wrong library or version mismatch?)

warning: .dynamic section for "/lib/x86_64-linux-gnu/libfreetype.so.6" is not at the expected address (wrong library or version mismatch?)

warning: .dynamic section for "/lib64/ld-linux-x86-64.so.2" is not at the expected address (wrong library or version mismatch?)

warning: .dynamic section for "/lib/x86_64-linux-gnu/libasound.so.2" is not at the expected address (wrong library or version mismatch?)

warning: .dynamic section for "/lib/x86_64-linux-gnu/libpulse.so.0" is not at the expected address (wrong library or version mismatch?)

warning: .dynamic section for "/lib/x86_64-linux-gnu/libX11.so.6" is not at the expected address (wrong library or version mismatch?)

warning: .dynamic section for "/lib/x86_64-linux-gnu/libXrandr.so.2" is not at the expected address (wrong library or version mismatch?)
[... 35 lines cut ...]
warning: File "/usr/lib/x86_64-linux-gnu/libthread_db-1.0.so" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load".
To enable execution of this file add
	add-auto-load-safe-path /usr/lib/x86_64-linux-gnu/libthread_db-1.0.so
line to your configuration file "/home/kisvegabor/.gdbinit".
To completely disable this security protection add
	set auto-load safe-path /
line to your configuration file "/home/kisvegabor/.gdbinit".
--Type <RET> for more, q to quit, c to continue without paging--c
For more information about this security protection see the
"Auto-loading safe path" section in the GDB manual.  E.g., run from the shell:
	info "(gdb)Auto-loading safe path"

warning: Unable to find libthread_db matching inferior's thread library, thread debugging will not be available.

warning: td_ta_new failed: generic error

warning: File "/usr/lib/x86_64-linux-gnu/libthread_db-1.0.so" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load".

warning: Unable to find libthread_db matching inferior's thread library, thread debugging will not be available.
Core was generated by `/home/amirgon/projects/lv_micropython/ports/unix/micropython-dev lib/lv_bindings'.
Program terminated with signal SIGABRT, Aborted.
#0  0x00007ffff79f500b in add_module (rp=0x5555560ef33c <error: Cannot access memory at address 0x5555560ef33c>, directory=<error reading variable: Cannot access memory at address 0x7ffff7b6a528>, dir_len=140737331769984, modcounter=<error reading variable: Cannot access memory at address 0x7ffff7b6a530>, nmodules=<optimized out>, modules=<optimized out>) at gconv_conf.c:305
305	gconv_conf.c: No such file or directory.

```

amirgon, 2023-02-07:
> Thanks. I've looked into it but doesn't seem to work:

Instead of using the crashdump, you can try to reproduce the problem directly like this:
```
gdb --args ports/unix/micropython-dev lib/lv_bindings/tests/run_test.py lib/lv_bindings/lvgl/examples/libs/tiny_ttf/lv_example_tiny_ttf_2.py

handle SIGUSR34 nostop noprint
run
run 
run
...
run
```

until the problem happens and it stops with an assertion.

kisvegabor, 2023-02-07:
Thank for the gdb commands.

I needed to use my "micropython-dev" because  yours was linked with and other version of rlottie. I built mine like this:
```
make -j VARIANT=dev DEBUG=1
```

And after ~50 "run"s in GDB I've seen only these messages, which look normal:
```
Starting program: /home/kisvegabor/projects/lvgl/lv_micropython/ports/unix/micropython-dev lib/lv_bindings/tests/run_test.py lib/lv_bindings/lvgl/examples/libs/tiny_ttf/lv_example_tiny_ttf_2.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffedac6700 (LWP 22482)]
[New Thread 0x7fffed170700 (LWP 22483)]
[New Thread 0x7fffec96f700 (LWP 22484)]
[New Thread 0x7fffdffff700 (LWP 22485)]
[New Thread 0x7fffdf7fe700 (LWP 22486)]
[New Thread 0x7fffdeffd700 (LWP 22487)]
[New Thread 0x7fffde7fc700 (LWP 22488)]
[New Thread 0x7fffddffb700 (LWP 22489)]
[New Thread 0x7fffdd7fa700 (LWP 22490)]
[New Thread 0x7fffdcff9700 (LWP 22491)]
[New Thread 0x7fffbffff700 (LWP 22492)]
[New Thread 0x7fffbf7fe700 (LWP 22493)]
[New Thread 0x7fffbeffd700 (LWP 22494)]
[New Thread 0x7fffbe7fc700 (LWP 22495)]
[New Thread 0x7fffbdffb700 (LWP 22496)]
[New Thread 0x7fffbd7fa700 (LWP 22497)]
[New Thread 0x7fffbcff9700 (LWP 22498)]
[New Thread 0x7fff9ffff700 (LWP 22499)]
[New Thread 0x7fff9f7fe700 (LWP 22500)]
[New Thread 0x7fff9effd700 (LWP 22501)]
[New Thread 0x7fff9e7fc700 (LWP 22502)]
[New Thread 0x7fff9dffb700 (LWP 22503)]
[New Thread 0x7fff9d7fa700 (LWP 22504)]
[Thread 0x7fff9d7fa700 (LWP 22504) exited]
[New Thread 0x7fff9d7fa700 (LWP 22505)]
[Thread 0x7fff9d7fa700 (LWP 22505) exited]
Running the SDL lvgl version
Running lib/lv_bindings/lvgl/examples/libs/tiny_ttf/lv_example_tiny_ttf_2.py ...
lvgl obj 
lvgl label  text:"Hello world
I'm a font created with Tiny TTF"
[Thread 0x7fff9dffb700 (LWP 22503) exited]
[Thread 0x7fff9e7fc700 (LWP 22502) exited]
[Thread 0x7fff9effd700 (LWP 22501) exited]
[Thread 0x7fff9f7fe700 (LWP 22500) exited]
[Thread 0x7fff9ffff700 (LWP 22499) exited]
[Thread 0x7fffdcff9700 (LWP 22491) exited]
[Thread 0x7fffbdffb700 (LWP 22496) exited]
[Thread 0x7fffbe7fc700 (LWP 22495) exited]
[Thread 0x7fffbeffd700 (LWP 22494) exited]
[Thread 0x7fffbf7fe700 (LWP 22493) exited]
[Thread 0x7fffbffff700 (LWP 22492) exited]
[Thread 0x7fffdd7fa700 (LWP 22490) exited]
[Thread 0x7fffddffb700 (LWP 22489) exited]
[Thread 0x7fffde7fc700 (LWP 22488) exited]
[Thread 0x7fffbcff9700 (LWP 22498) exited]
[Thread 0x7fffbd7fa700 (LWP 22497) exited]
[Thread 0x7fffdeffd700 (LWP 22487) exited]
[Thread 0x7fffdf7fe700 (LWP 22486) exited]
[Thread 0x7fffdffff700 (LWP 22485) exited]
[Thread 0x7fffec96f700 (LWP 22484) exited]
[Thread 0x7fffed170700 (LWP 22483) exited]
[Thread 0x7fffedac6700 (LWP 22482) exited]
--Type <RET> for more, q to quit, c to continue without paging--
```

amirgon, 2023-02-07:
Ok, so I'm making another attempt to let you debug this on your side.
I've created a Docker image with built lv_micropython and a crashdump, and uploaded it to Dockerhub.

Could you try running this on your side?
You should be able to see the assert on frame 4

```bash 
sudo docker pull amirgon/lvgl_issue_3968:latest
sudo docker run -it --rm amirgon/lvgl_issue_3968:latest gdb /lv_micropython/ports/unix/micropython-dev /lv_micropython/tiny_ttf.core
```

kisvegabor, 2023-02-08:
It's a good idea to use Docker, but seemingly it's not working:
```
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04.1) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /lv_micropython/ports/unix/micropython-dev...
[New LWP 16250]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `/lv_micropython/ports/unix/micropython-dev lib/lv_bindings/tests/run_test.py lib'.
Program terminated with signal SIGABRT, Aborted.
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) run
Starting program: /lv_micropython/ports/unix/micropython-dev 
warning: Error disabling address space randomization: Operation not permitted
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
MicroPython v1.19.1-659-g3bb7c7565 on 2023-02-07; linux [GCC 9.4.0] version
Use Ctrl-D to exit, Ctrl-E for paste mode
```

amirgon, 2023-02-08:
> It's a good idea to use Docker, but seemingly it's not working:

Not working? This is exactly what I expected to see!

```
Program terminated with signal SIGABRT, Aborted.
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
```

Don't `run`... this is a crashdump, the program already ran and crashed, and you can debug the crash now.
Try `bt` to see the stack trace for example, the assertion is on frame 4. You can examine variables and memory etc.

kisvegabor, 2023-02-08:
Oh, I see now. Can you make a build with ASan enabled?

amirgon, 2023-02-08:
> Can you make a build with ASan enabled?

Yes, I've tried enabling ASan now.  
I found an unrelated issue related to qstr, but after fixing it, tiny_ttf still crashes and ASan does not report any problem.  
I cannot create a core-dump when ASan is running (because it maps infinite virtual memory), but I don't think there's any need. Since ASan didn't report any errors related to tiny_ttf, I don't think gdb+ASan would give any more information other than we have so far without ASan.

<details>
  <summary>Diff for enabling ASan and fixing qstr issue</summary>

```diff
diff --git a/ports/unix/Makefile b/ports/unix/Makefile
index 0b05d7433..ea9e711c0 100644
--- a/ports/unix/Makefile
+++ b/ports/unix/Makefile
@@ -219,6 +219,10 @@ endif

 # Additional optional libraries

+CFLAGS_MOD += -fsanitize=address
+CXXFLAGS_MOD += -fsanitize=address
+LDFLAGS_MOD += -fsanitize=address
+
 RLOTTIE_CFLAGS_MOD :=  $(shell pkg-config --silence-errors --cflags rlottie)
 RLOTTIE_LDFLAGS_MOD := $(shell pkg-config --silence-errors --libs   rlottie)
 ifneq ($(RLOTTIE_LDFLAGS_MOD),)
diff --git a/py/qstr.c b/py/qstr.c
index ecac2cd0c..c203e2e29 100644
--- a/py/qstr.c
+++ b/py/qstr.c
@@ -82,6 +82,7 @@ const qstr_hash_t mp_qstr_const_hashes0[] = {
 #undef QDEF0
 #undef QDEF1
     #endif
+    0
 };

 const qstr_hash_t mp_qstr_const_hashes1[] = {
@@ -92,6 +93,7 @@ const qstr_hash_t mp_qstr_const_hashes1[] = {
 #undef QDEF0
 #undef QDEF1
     #endif
+    0
 };

 const qstr_len_t mp_qstr_const_lengths0[] = {
@@ -102,6 +104,7 @@ const qstr_len_t mp_qstr_const_lengths0[] = {
 #undef QDEF0
 #undef QDEF1
     #endif
+    0
 };

 const qstr_len_t mp_qstr_const_lengths1[] = {
@@ -112,6 +115,7 @@ const qstr_len_t mp_qstr_const_lengths1[] = {
 #undef QDEF0
 #undef QDEF1
     #endif
+    0
 };

 const qstr_pool_t mp_qstr_special_const_pool = {
```
</details>

<details>
  <summary>Reproducing the problem in a loop with gdb</summary>

```bash
for i in {1..20}; do gdb -ex='set confirm on' -ex 'handle SIG34 nostop noprint' -ex run -ex quit --args ports/unix/micropython-dev lib/lv_bindings/tests/run_test.py lib/lv_bindings/lvgl/examples/libs/tiny_ttf/lv_example_tiny_ttf_2.py; done
```

When the following appears:
```
Program received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
A debugging session is active.                                                             
                                                                                                                                                                                       
        Inferior 1 [process 32990] will be killed.
                                                                                           
Quit anyway? (y or n) n
```

Click <kbd>n</kbd> and debug.

</details>

Any ideas?

kisvegabor, 2023-02-09:
I have no idea... I cc @codewitch-honey-crisis who added the feature. In summary:
- [lv_example_tiny_ttf_2](https://github.com/lvgl/lvgl/blob/master/examples/libs/tiny_ttf/lv_example_tiny_ttf_2.c) sometimes crashes, but in MicroPython.
- It's not garbage collection issue, as @amirgon tryed it out without GC too
- The debugger deosn't show any useful info 
- ASan doesn't show any useful info either

@codewitch-honey-crisis  Could you try disabling parts of the code (e.g. caching) to see which part causes the problem?

codewitch-honey-crisis, 2023-02-09:
I can't even build the latest bits of LVGL on my system. Whenever I've tried to upload anything I downloaded off this site to an ESP32 I get a big fat fail.

I can modify code, but I can't run it to see if it even works.

amirgon, 2023-02-09:
> I can modify code, but I can't run it to see if it even works.

You can build and run the Unix port.
Also, to make things easier, I've prepared a docker image that reproduces the error with a built mp+lvgl. See above.

codewitch-honey-crisis, 2023-02-09:
Oh the error message. I think i remember that one. You can reproduce it by looking for a glyph that isn't present in the font. If it's the one I remember I wanted to eliminate it and make it put a blank square in place of the glyph but I never was successful at modifying it as it's not my codebase originally. Why it would be intermittent I have no idea.

codewitch-honey-crisis, 2023-02-09:
Okay hang on. I've read the whole thread now. I don't know anything about docker. I installed it once on an old machine, never used it and that was end of it. The only image I could find in the thread above was a crash dump, so my eyes must be just skipping over your docker image. Can you relink to it? 

It's curious that this is only failing with stream.

amirgon, 2023-02-09:
> Can you relink to it?

```
sudo docker pull amirgon/lvgl_issue_3968:latest
sudo docker run -it --rm amirgon/lvgl_issue_3968:latest gdb /lv_micropython/ports/unix/micropython-dev /lv_micropython/tiny_ttf.core
```

codewitch-honey-crisis, 2023-02-09:
Thank you. Forewarned all: I know nothing about docker, WSL, nor much about linux. I just had to install it all on this machine.

I've got the two commands you just posted to me working. I drop to a debug prompt in gdb which is not at all helpful to me but at least it shows something is working.

Eventually I will find the files, and find the offending code. Could be today. Could be a month. Likely at least a week since I don't know what I am doing.

amirgon, 2023-02-09:
> Forewarned all: I know nothing about docker

You have an excellent opportunity here to learn new exciting things! Docker is a great tool and definitely worth learning about it. There is lots of information on the internet so you don't have to rely on your friend. Feel free to ask us too.
Good luck!

codewitch-honey-crisis, 2023-02-09:
Okay, I just didn't want to clutter up this feed with docker help. I'm a total noob here. I got those commands working, but that's it. I don't where it put the files or anything. When I find the files I still have to build them.

If this were C or C++, I'd fire up VS Code and dev it using PIO. I've never used MicroPython before, and I only know enough python to port code away from it which tells you how i feel about it in general.

So I am completely out of my element and this is all new ground. I'm not even quite sure where to begin. I've never built anything with MP. I've never run MP. If this only reproduces under MP, I've got a massive learning curve that frankly, I will only ever use for this one thing.

What I can do to sidestep all of this is suggest changes. But that's better done in a chat format with someone with a working dev env on the other end. If anyone wants to go that route I'm on discord Honey Crisis #4038

codewitch-honey-crisis, 2023-02-09:
Okay, I found the files. They are buried under var/lib/docker/image/whatever

The problem with that is i don't have read/list permissions on any of those folders or subfiles

So I'm sitting here setting permissions just to navigate the source tree.

This is unusable. I'm finding another way to do this.

If it's only reproducible under MP, then I at least need to know which assert is being fired. I can update the code with a pull request to that effect or someone can go in here 

https://github.com/lvgl/lvgl/blob/master/src/libs/tiny_ttf/stb_truetype_htcw.h

To the routine in line 1683 (the sbtt_FindGlyIndex) and change each assert so instead of zero it says something like 0 /* assert 1 failed */ 
0 /* assert 2 failed */

and so on. I'm happy to make a pull request but that seems like a crap ton of extra churn for everyone involved, so if there's an easier way to update that code, then go for it.

amirgon, 2023-02-09:
> If anyone wants to go that route I'm on discord Honey Crisis #4038.

I'm not very familiar with discord. How do I contact you?

codewitch-honey-crisis, 2023-02-09:
i just waved at you

amirgon, 2023-02-11:
Thank you @codewitch-honey-crisis for your help debugging this! It really helped me focus where the problem could be.

So apparently something is wrong with the file-system access. We found out that some bytes that are read from a file are different from the actual contents of the file on disk.

Some additional observations:
- This happens with [fs_driver.py](https://github.com/lvgl/lv_binding_micropython/blob/master/lib/fs_driver.py) as a file-system driver. I also tried the LVGL posix driver and could not reproduce the problem.
- This only happens when caching is enabled, with at least 16 bytes of cache. With smaller cache or no cache I could not reproduce this
- As mentioned earlier - this is an intermittent problem. It only happens sometimes

So apparently, this has something to do with the fs_driver.py when caching is enabled...

Another thing I noticed was, that the lv_fs callbacks are not deterministic (multiple runs with the same inputs yield different outputs).  
When I run exactly the same script, LVGL invokes different calls to `read_cb` and `seek_cb` on each run. This makes it more difficult to debug because I can't compare "good" run to "bad" run, when every two "good" runs are different from each other.

@kisvegabor Any idea why runs are not deterministic? Could it be related to lv_fs cache somehow?  
I don't think it's related to fs_driver.py because I confirmed fs_driver.py returns the same results on different runs up until LVGL starts calling the callbacks differently (diff is the same until one run calls additional "reads" and "seeks" and then runs diverge)

codewitch-honey-crisis, 2023-02-12:
@amirgon Thanks for working with me over Discord. It really helped at least for me to keep things flowing compared to trying to do it here - unconventional as it may have been. I'm relieved that it wasn't in the font code because I was running out of ideas. The problem is well outside my wheelhouse, so I won't be of any help except to say good luck tracking down that issue.

kisvegabor, 2023-02-14:
It's great that you could narrow down the issue! 

I've also tried it with caching but it worked well here too. Anyway, I've enabled FS cache for the test we run with CI. 

LVGL's FS API is a very thin layer and I don't think the issue is there for 2 reasons:
1. When there where bugs in the e.g. the cache people have reported it and they were fixed
2. The STDIO drivers are used during CI tests too and complex tests (font load, or tiny_ttf) are running well

So the problem should be in `fs_driver.py`. The only non obvious parts are the casts like [this](https://github.com/lvgl/lv_binding_micropython/blob/master/lib/fs_driver.py#L39-L41). 

I think it'd be useful to have a simple test for `fs_driver.py`. Something like
- create a file with 256 byte data counting from 0..255
- set cache size to 32
- read data from random locations with random length

I have no idea why is it no deterministic. Hopefully a simple test like this will shed some light on it too.

amirgon, 2023-02-25:
> So the problem should be in `fs_driver.py`. The only non obvious parts are the casts like [this](https://github.com/lvgl/lv_binding_micropython/blob/master/lib/fs_driver.py#L39-L41).

@kisvegabor I did a simple test by changing `fs_driver.py`: 

```python
def fs_open_cb(drv, path, mode):
    posix_file = lv.fs_file_t()
    posix_file.open(f'P:{path}', mode)
    return {'posix_file' : posix_file}

def fs_close_cb(drv, fs_file):
    fs_file.__cast__()['posix_file'].close()
    return lv.FS_RES.OK

def fs_read_cb(drv, fs_file, buf, btr, br):
    fs_file.__cast__()['posix_file'].read(buf, btr, br)
    sleep(0.00001)
    return lv.FS_RES.OK

def fs_seek_cb(drv, fs_file, pos, whence):
    fs_file.__cast__()['posix_file'].seek(pos, whence)
    return lv.FS_RES.OK
```

On `lv_conf.h` I've defined:
```c
#define LV_USE_FS_POSIX 1
#if LV_USE_FS_POSIX
    #define LV_FS_POSIX_LETTER 'P'     /*Set an upper cased letter on which the drive will accessible (e.g. 'A')*/
    #define LV_FS_POSIX_PATH ""         /*Set the working directory. File/directory paths will be appended to it.*/
    #define LV_FS_POSIX_CACHE_SIZE 10000    /*>0 to cache this number of bytes in lv_fs_read()*/
#endif
```

This `fs_driver.py` redirects every FS call to LVGL Posix FS driver which is registered on `P:` with a large cache.   
All non-obvious casts are removed (except from the fs_file cast which is needed for extracting the `fs_file_t` instance from the opaque file handle).

When running this - the problem reproduces! (sometimes crashes with "stbtt_FindGlyphIndex: Assertion `0' failed", sometimes doesn't). This shows that the problem is not related to Micropython filesystem or to the non-obvious casts.

The most interesting thing is the `sleep(0.00001)` that I had to add on "fs_read_cb". If I remove this delay the problem disappears!

So to summarize:
- The problem only happens when caching is enabled
- The problem only happens when "read" is delayed a little
- The problem is only observed in lv_micropython so far, but is unrelated to MP filesystem or to casts, and also happens with LVGL Posix FS driver.

What do you think?

amirgon, 2023-02-27:
I get this build warning, you probably need a `default` case in the switch:
```
../../lib/lv_bindings/lvgl/src/libs/fsdrv/lv_fs_posix.c: In function ‘fs_seek’:
../../lib/lv_bindings/lvgl/src/libs/fsdrv/lv_fs_posix.c:193:20: error: ‘w’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  193 |     off_t offset = lseek((lv_uintptr_t)file_p, pos, w);
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [../../py/mkrules.mk:80: build-dev/lib/lv_bindings/lvgl/src/libs/fsdrv/lv_fs_posix.o] Error 1
```

> Could you try it with MP?

Unfortunately the bug is still there, even after applying https://github.com/lvgl/lvgl/commit/84d91adaf48db5979708865cf5f7aae0edd70705.  
Maybe there are some additional cache issues? I'm not sure what else could go wrong.  

Did you try running the tiny_ttf_2 C example with posix driver, a large cache and some delay on the "read" function? I think it would help to reproduce this without MP at all.

kisvegabor, 2023-02-28:
> Did you try running the tiny_ttf_2 C example with posix driver, a large cache and some delay on the "read" function? I think it would help to reproduce this without MP at all.

I've tried it but I can't reproduce it in C. :slightly_frowning_face: 

Could you convert [test_fs](https://github.com/lvgl/lvgl/blob/master/tests/src/test_cases/test_fs.c#L68) to MP? 

I can't imagine how the delay can cause this issue.

codewitch-honey-crisis, 2023-02-28:
Hey, I'm largely lost here, but if it helps, the reason tiny_ttf might be bringing this to the surface is it does a lot of little seeks and little reads. Basically it's grinding it's way through the file, a word at a time, often seeking between every few words. So my first place to look regarding your cache would be your seek handling. Are you clearing the read cache after every seek? Or are you partially clearing the read cache after every seek? <--- that last bit can be hard to test, but when it comes up, ouch. I'm not saying that's your problem, but if you haven't looked there, you might. In any case, good luck!

kisvegabor, 2023-02-28:
Good news! I've took a deeper look into `seek`, added some more tests, and it turned out that is some special cases it really fails.

I've pushed an other fix. I hope it solve the issue :crossed_fingers: 

Thanks for the hint @codewitch-honey-crisis

amirgon, 2023-02-28:
> I've pushed an other fix. I hope it solve the issue crossed_fingers

Unfortunately, the problem is still there with your latest fixes... :confused:
Something is still wrong.

amirgon, 2023-02-28:
**I just found out something very interesting!**

In MP we call the event loop periodically and asynchronically.
This means that when it's time to refresh the screen, before executing the next Python command, MP first calls `lv_task_handler` to render and update the screen.

The point is that since MP FS driver is written in Python, `lv_task_handler` can be called inside the read/seek functions.

I did an experiment: I disabled the event loop in `lv_example_tiny_ttf_2.py` and enabled it only as the last command in the script - and the problem disappeared!  
If I disable the event loop only around `lv_tiny_ttf_create_file` the problem still happens. It must be disabled around all operations such as setting label text, centering etc.

In C this usually doesn't happen because the event loop really starts only after everything was set up and layed down.

So I suspect that LVGL can't handle the case where `lv_task_handler` is called from FS callbacks, which themselves are called from LVGL when rendering the font...
A simple solution could be to make `lv_task_handler` return immediately in case it's called while LVGL is not ready to handle it. (in our case, when called from within FS callbacks)

kisvegabor, 2023-03-01:
Huh, it's great that you have found something. I was really out of ideas :slightly_smiling_face: 

I wonder why isn't a problem in general. Following your explanation it can easily happen that when you create an object and set its parameters (let's say  50 LVGL function calls) MP runs `lv_timer_handler` somewhere. E.g. after line 27. If it'd happen we should see a flickering with a half configured widgets, but I've never seen that. 

So can it really run `lv_timer_handler` after any line of command, or only when MP is waiting for something, like  Sleep, or a File I/O operation?

If it can happen only while reading, then it can still mess up things.  Imagine that `lv_label_set_text` reads a lot of data about glyphs to set the label's properties, but during a read LVGL starts to refresh the screen, and use this half-configured label.

amirgon, 2023-03-01:
> If it'd happen we should see a flickering with a half configured widgets, but I've never seen that.

Actually, with tiny_ttf_2 MP example I've seen the font being drawn first on the top and then jumps to the middle when text is centered, etc.
This is observable only after I added the [delay in the read function](https://github.com/lvgl/lvgl/issues/3968#issuecomment-1445220743) and everything is deliberately slower.

> So can it really run `lv_timer_handler` after any line of command, or only when MP is waiting for something, like Sleep, or a File I/O operation?

MP is an interpreter (of bytecode) so it runs each individual Python command from a main loop.  
It can really call the `lv_task_handler` after any single Python command and not only on blocking operations.  
Usually when MP calls some LVGL API function it will not call `lv_task_handler` in the middle of that LVGL API function. But in our case LVGL calls back to MP FS driver which sometimes calls `lv_task_handler`, so it's a special case where `lv_task_handler` is called "in the middle" of some LVGL function.

> Imagine that `lv_label_set_text` reads a lot of data about glyphs to set the label's properties, but during a read LVGL starts to refresh the screen, and use this half-configured label.

Maybe this can happen, but it's not what actually happens.  
What we see is that tiny_ttf receives wrong data from the FS driver... I still don't understand how this can happen but it only happens when `lv_task_handler` is called from the read callback

btw, I think that you can now try to reproduce this without MP.  
Simply initialize a display and add some random calls to `lv_task_handler` from the C FS driver when running the tiny_ttf_2 C example - my guess is that you'll see the same problem

kisvegabor, 2023-03-01:
I haven't tried reproducing it yet, but calling `lv_task_handler` (possibly) at any random places is clear against of how LVLG is designed. If we solved the FS issue now, we can just hope that MP (or its later version) won't call `lv_task_handler` at any time.

I don't know the options, but I think it should be solved in MP. If I understood correctly, the problem is that MP calls `lv_timer_handler`  independently of the main script and it doesn't know that there are times when it's not appropriate to call it.

Could we manage it the binding with something like this?
```c
void binding_to_some_lvgl_function(...)
{
   binding_lock++;
   
    //The actual binding here

   binding_lock--;
}

void lv_timer_handler_mp_loop(void)
{
   if(binding_lock == 0) {
      lv_timer_handler();
      usleep(5000);
   } else {
      usleep(10);
   }
}
```

amirgon, 2023-03-01:
> `lv_task_handler` (possibly) at any random places is clear against of how LVLG is designed.

It's not "at any random place" because it will not happen when LVGL code is running - there is only one thread.

In LVGL it's legal, for example, to create a button, call `lv_task_handler`, then later center the button and call `lv_task_handler` again right? This is what happens with MP - it can call `lv_task_handler` only between LVGL primitives.  
The exception to that is callbacks where LVGL calls MP, as in our case of MP FS driver.

> I don't know the options, but I think it should be solved in MP.

I think that if this is solved in LVGL it would be more a general solution. I can imagine cases either for some other bindings or for some C code with async loops that can reach the same situation of `lv_task_handler` call inside LVGL callback.  

> Could we manage it the binding with something like this?

The same can be done in either C or MP. LVGL could set `lv_task_handler` to return immediately if called when another LVGL function is in progress.  
Anyway, the flow on MP is different than what you describe. There is no `lv_timer_handler_mp_loop` but some event handler. 
We can probably solve this in MP if you don't want to solve this in LVGL itself, although I think LVGL support for this would be more robust.

kisvegabor, 2023-03-01:
> In LVGL it's legal, for example, to create a button, call lv_task_handler, then later center the button and call lv_task_handler again right? This is what happens with MP - it can call lv_task_handler only between LVGL primitives.

It's legal, but unexpected as the user doesn't want it explicitly. For example if there is `timer` which checks some widgets, it might find a half configured widget and user will wonder why it happens.

> The exception to that is callbacks where LVGL calls MP, as in our case of MP FS driver.

If really this is the case let's check where we have LVGL->MP calls. Basically all places where there is an `add_cb` or `set_cb`, including:
- **FS**
- **obj events**, but these are called from `lv_timer_handler` anyway. Except when the user uses `lv_obj_event_send` directly
- **indev and display callbacks**, also called from `lv_timer_handler`
- **animation callbacks**, but these are also called from `lv_timer_handler`, except when the animation is created and `early_apply` is set.
- **timers**, obviously also called from `lv_timer_handler`.

So there are really finite places if we ignore the places where the callback are called from `lv_timer_handler` and we can add the same `lock++` `lock--` mechanism in LVGL too. However I'm still not sure this it the approach.

amirgon, 2023-03-02:
> It's legal, but unexpected as the user doesn't want it explicitly.

Today MP starts the event loop automatically once the display driver is configured.  
Another option was to ask the user to start it explicitly after setting up everything, but on the other hand most users don't think of the event loop or even aware of it, they just want things to work after setting up their widgets. So it's simpler for most users that it's started automatically.  
For advanced users it's possible to start the event loop later, but I believe very few actually do that today.

> So there are really finite places if we ignore the places where the callback are called from `lv_timer_handler` and we can add the same `lock++` `lock--` mechanism in LVGL too. However I'm still not sure this it the approach.

I think the first step is to decide if we need this fix on MP side only or in LVGL core.  Both options are valid solutions.  

Do you think LVGL users are aware to the fact that they must prevent `lv_task_handler` from being called during callbacks?  
This could happen in MP or other LVGL bindings, but can also happen in pure-C programs where the event loop calls `lv_task_handler` automatically when execution is blocked by IO, or when there's some await-async mechanism in place, even when running on a single thread and no locking is necessary.

amirgon, 2023-03-02:
Not sure if we decide to fix this in LVGL or not, but [the fix in MP is pretty simple](https://github.com/lvgl/lv_binding_micropython/commit/04f2ab4e09f29d8858e0256a21c33359cd6d6f13) so I've committed it.  
If this is fixed in LVGL I'll revert this, no point in fixing this on both LVGL and MP.

With this fix we should no longer see this problem.

kisvegabor, 2023-03-03:
It's great that this simple approach worked. I suggest leaving it in MP as I think it'd be more complicated to manage in LVGL.

If there will be request about having it in LVGL, of course we can reconsider it.

amirgon, 2023-03-03:
Ok.