lvgl/lvgl #3973: Two or more indevs handling

mmar22, 2023-02-06:
Roller is controlled by two indevs touch and encoder, but if touch is used
object is switched from EDITED to FOCUS and encoder stop working,

Howto best way to handle this ???

EDITED is activated from setup screen 
`lv_group_set_editing(g1, true);`

kisvegabor, 2023-02-07:
Hi,

I also think that it'd better if it worked as you suggested. Can you try these modifications? 
```diff
diff --git a/src/core/lv_group.c b/src/core/lv_group.c
index 2c4fa93e9..63fe4c0c9 100644
--- a/src/core/lv_group.c
+++ b/src/core/lv_group.c
@@ -238,13 +238,13 @@ void lv_group_focus_obj(lv_obj_t * obj)
 
     if(g->frozen != 0) return;
 
-    /*On defocus edit mode must be leaved*/
-    lv_group_set_editing(g, false);
-
     lv_obj_t ** i;
     _LV_LL_READ(&g->obj_ll, i) {
         if(*i == obj) {
             if(g->obj_focus != NULL && obj != *g->obj_focus) {  /*Do not defocus if the same object needs to be focused again*/
+                /*On defocus edit mode must be leaved*/
+                lv_group_set_editing(g, false);
+
                 lv_res_t res = lv_event_send(*g->obj_focus, LV_EVENT_DEFOCUSED, get_indev(g));
                 if(res != LV_RES_OK) return;
                 lv_obj_invalidate(*g->obj_focus);
```

If it works well I'll add these changes to LVGL.

mmar22, 2023-02-07:
Maybe you miss copy paste need here true
+ lv_group_set_editing(g, false);

and from this code I see frozen. Cant be used for lock ? Or from my side is better based on some setup ignore false and return from lv_group_set_editing?

From: Gabor Kiss-Vamosi 
Sent: Tuesday, February 7, 2023 8:45 AM
To: lvgl/lvgl 
Cc: mmar22 ; Author 
Subject: Re: [lvgl/lvgl] Two or more indevs handling (Issue #3973)

Hi,

I also think that it'd better if it worked as you suggested. Can you try these modifications?

diff --git a/src/core/lv_group.c b/src/core/lv_group.c
index 2c4fa93e9..63fe4c0c9 100644
--- a/src/core/lv_group.c
+++ b/src/core/lv_group.c
@@ -238,13 +238,13 @@ void lv_group_focus_obj(lv_obj_t * obj)
 
     if(g->frozen != 0) return;
 
-    /*On defocus edit mode must be leaved*/
-    lv_group_set_editing(g, false);
-
     lv_obj_t ** i;
     _LV_LL_READ(&g->obj_ll, i) {
         if(*i == obj) {
             if(g->obj_focus != NULL && obj != *g->obj_focus) {  /*Do not defocus if the same object needs to be focused again*/
+                /*On defocus edit mode must be leaved*/
+                lv_group_set_editing(g, false);
+
                 lv_res_t res = lv_event_send(*g->obj_focus, LV_EVENT_DEFOCUSED, get_indev(g));
                 if(res != LV_RES_OK) return;
                 lv_obj_invalidate(*g->obj_focus);If it works well I'll add these changes to LVGL.

mmar22, 2023-02-07:
SOLVED with
`  lv_group_focus_freeze(g1, true);
`

closed 2023-02-07 by mmar22