lvgl/lvgl #3977: LVGL Terminal now available on NuttX

lupyuen, 2023-02-08:
### Introduce the problem

While porting NuttX RTOS to PINE64 PinePhone, we needed a way to run NuttX Commands through the Touchscreen.

That's why we created the LVGL Terminal App that runs commands entered with a Touchscreen Keyboard:

- [Screenshot](https://lupyuen.github.io/images/terminal-title.jpg)

- [Demo Video](https://www.youtube.com/watch?v=WdiXaMK8cNw)

- [How to run it](https://youtu.be/kGI_0yK1vws)

- [How it works](https://lupyuen.github.io/articles/terminal)

- [Source Code](https://github.com/apache/nuttx-apps/blob/master/examples/lvglterm/lvglterm.c)

### Examples and cases

The LVGL Terminal might be helpful for developers who need a Command-Line Interface on phone and tablet devices.

Without LVGL Terminal, PinePhone Developers would need a Serial Cable to run NuttX Commands.

### Suggested solution

We have implemented a simple barebones solution for LVGL Terminal, see above.

The code might be portable to other POSIX / Linux platforms. It uses `posix_spawn()` to start a shell, and POSIX Pipes to send shell commands. I hope this code will be helpful to the LVGL Community.

[(More details in the article)](https://lupyuen.github.io/articles/terminal)

Thanks everyone for the awesome LVGL code that made NuttX come alive on PinePhone! :-)

kisvegabor, 2023-02-10:
Hi Lee,

Very nice! Thank for sharing. 

It seems there isn't that much NuttX specific parts in the code. Do you think it might be useful to have it as a built-in LVGL example (with an abstract interface to stdin and stdout). No pressure at all, if you'd prefer to keep it in NuttX, it's also perfect.

BTW, you can also  share it in our [Forum's My projects](https://forum.lvgl.io/c/my-projects/10) category. :slightly_smiling_face:

lupyuen, 2023-02-10:
Thanks Gabor! I think I'll stick to NuttX for now, since I have no idea how to test the code on other platforms :-)

I'll post this to the forum. Thanks! :-)

closed 2023-02-10 by lupyuen