lvgl/lvgl #3978: get text extent function

kdschlosser, 2023-02-09:
### Introduce the problem
Not a problem but a feature that should be added.

In every GUI framework I have used there has always been the ability to pass a string of text to the rendering engine or "DC" that would return the width and height in pixels of the text that is passed to it. The returned vales are dependent on what font the rendered has been told to use.

This is an extremely handy feature to have when wanting to manually position items or to figure out how tall to make a widget. An example of the latter is the roller widget and setting the number of visible items. How many items to display is going to be based on how large the font is and the height of the display. To have a clean UI you would want to set the widget height to a multiple of the number of lines shown with a little bit of space added. The returned values from the "get_text_extent" function would also include a line spacing added into it for a single line and for multiline text it would be `line_spacing * number_of_lines` that gets added to the sum of the height of each line.

The "get_text_extent" function could also return the tallest characters height and an averaged width of all of the characters in the text.

I would make the function so that a font could be passed to it or instead of a font an object could be passed that has a font set to it. If an object is passed that is not a font then anything that would alter the appearance of the font should be taken into consideration when calculating the values.

kisvegabor, 2023-02-10:
Hi,

A function like this is required in LVGL too the render texts. It is [lv_txt_get_size](https://github.com/lvgl/lvgl/blob/master/src/misc/lv_txt.h#L83) and it does exactly what you have described.

> An example of the latter is the roller widget and setting the number of visible items.

We have a dedicated function for it: [lv_roller_set_visible_row_count](https://github.com/lvgl/lvgl/blob/master/src/widgets/roller/lv_roller.h#L92). :slightly_smiling_face:

kdschlosser, 2023-02-10:
I know you have the dedicated function for the roller. and no wonder why I couldn't find a function in the documentation for getting the text size. It's because it;s not in the documentation!!!! That is the reason why I made this issue/suggestion.

So now it is a bug report. a bug in the documentation because that function is missing.

I am wanting to maximize space so if the text height is 10 pixels and the line spacing is 5 pixels and I want to show 8 lines if I set the height of the widget to 100 pixels it's going to fall short. if I set the height to 140 pixels that's wasted space. 125 would be where it needs to be. Or if I wanted to use the widget height to select the font that is to be used.

kisvegabor, 2023-02-13:
This function is meant to be used only internally, that's why there is no docs about it. 

I think a better solution (instead of promoting this function) would be to have an [`em` unit](https://www.w3schools.com/cssref/css_units.php) (stolen from web), however it should consider the line-height too.

This way you could write `lv_obj_set_height(label, lv_em(8))`.

What do you think?

kdschlosser, 2023-02-14:
> Please share a code example so that I can reproduce the issue.

I am only going to be able to do that for MicroPython. I do not know if the problem exists in the core or in the binding.

> For v9 we will establish the practice of having *_private.h files to make it more intuitive which functions is meant to be used by the user.

OK cool. That works

kisvegabor, 2023-02-15:
> I am only going to be able to do that for MicroPython. I do not know if the problem exists in the core or in the binding.

It's fine too. You click the 
[image]

button to save the script. After that you can just send the link.

kdschlosser, 2023-03-02:
Oh yeah I forgot about this.. I gotta get that test code into the simulator but it seems as tho the simulator is broken.

closed 2023-03-26 by lvgl-bot